Koenkk/zigbee2mqtt #1231: Enhancement request : ban a second zigbee2mqtt network

khzd, 2019-03-10:
Hello @Koenkk,
this is not an issue but an enhancement request.

Context : I have 2 zigbee2mqtt networks : a production one and an experimental one.
when I pair a new device it goes randomly to the first or to the second instance ( witch i would like to keep clean)
I would like to know if it's possible to extend the current ban functionality to cross ban two networks?
the idea is to use the configuration file of the second one as a ban file for the first and vis versa.
so the second configuration file of the second network  may contain:

**banned_network : zigbee2mqtt/data/configuration.yaml**
experimental:
  livolo: true
devices:
  '0x00124b000bf07xxx':
    friendly_name: my_office_light_up
    retain: false

the advantage of this option is to not replicate id's between networks.
Hop that need will be shared by all experimental users ;)
cheers

Koenkk, 2019-03-11:
Why not just disable joining on the production network?

khzd, 2019-03-13:
Hi
thanks @Koenkk for your prompt response.

i'm in a use case where i'm adding device to prod network and the experiment one respond also to the join : result randomly the experimental or the prod respond to the join :(
if i stop the experimental one ( livolo in this case) i have to re-pair the devices...

do you have a workaround  to avoid device to join ?
thanks

Koenkk, 2019-03-13:
You can try banning it from the network: http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigban

khzd, 2019-03-17:
Hi
I setup a python script to to read prod config  and ban all devices in experimental one : thanks for your prompt response.
Cheers

pkedvessy, 2019-03-20:
Hi,
I have a similar issue:  I can't disable permit join or ban a given device to connect where I don't want it to connect to. 
In the configuration.yml permit_join flag is false for network A and for network B as well, but new device still can connect randomly to A or B (changing log level by MQTT messages result messages from both bridges with disable permit join option). 
When I'm trying to ban the new device, the result is the same: still can join to the network where it is banned (meanwhile I get a response about the ban is initiated).

closed 2019-05-27 by stale[bot]