Koenkk/zigbee2mqtt #1232: IKEA TRADFRI remote control, color temperature not working

ripperle, 2019-03-10:
Hello,

after updating to the latest coordinator firmware (stable, January 2019) I was able to connect my IKEA TRADFRI floalt panel und the Ikea 5 button remote control...

Over zigbee2mqtt I have full control to the IKEA light panel (on/off, brightness, color temperature)

With the remote control I lost the ability to change the color temperature... So I only can switch it on and off and change brightness...

Very weird behavior... 😁

Koenkk, 2019-03-10:
Did you try this: http://www.zigbee2mqtt.io/information/binding.html#ikea-tradri-remote-control-e1524 ?

ripperle, 2019-03-11:
No, not yet,

I don't want do use the remote in zigbee2mqtt so I thought that wouldn't be necessary

timstanley1985, 2019-03-12:
@koenkk I am having a similar issue. I have followed the cc sniffer guidance and bound the switch to a light fitting with three bulbs. The on/off switch and brightness work flawlessly but the change colour buttons do not change colour temperature like they used to when using the tradfri hub.

ripperle, 2019-03-12:
I noticed that the normal IKEA panel supports only 3 different color temperatures...

With zigbee2mqtt I can set more than that! Maybe the firmware of the panel gets in an unknown state or something?!

Koenkk, 2019-03-13:
I've just checked and indeed setting colour temperature is not possible using this method. Perhaps some extra setup is done, need to do some more sniffing..

timstanley1985, 2019-03-19:
I have just paired a second remote to a full colour tradfri bulb. Set up went well and it integrates well with home assistant but the arrow buttons do not change colour with this bulb too

didiht, 2019-03-20:
In my environment it works well inclusive color temperature!

What I've done:

- Complete new setup after two crashes due to tests of the reporting feature (but that's another story)

didiht, 2019-03-20:
Sorry, this is my first comment in Github

What I've done

- Complete new setup after two crashes due to tests of the reporting feature (but that's another story)
- Paired some IKEA bulbs
- Paired IKEA Remote Control
- Found out and added the group number of the Remote Control
- Added the bulbs to the group of the Remote Control (no bind!)
- every feature works well

zigbee2mqtt version 1.2.1 (commit #e5ca977)
Coordinator firmware version: '20190223'

Great Job!
Thanks to Koenkk and all contributors!

ripperle, 2019-03-21:
@didiht
So you don't paired the remote directly with the IKEA lamps?
Are the remote working without the zigbee2mqtt coordinater?

didiht, 2019-03-22:
@ripperle 
No, I did not pair the remote directly with the IKEA bulbs.
Yes, the remote control works with the IKEA bulbs, even without the zigbee2mqtt coordinator.

I made following test:
- Delete the lamp from the group of the IKEA remote control -> IKEA remote does not work with the bulb anylonger
- Add lamp again to the group of the IKEA remote -> it works again, toggle, brightness, color temperature

With the 'bind' feature I had no success to set the color_temperature with the IKEA remote previously, but I didin't test this with the current version.

bjobo, 2019-03-23:
@didiht 
I tried exactly your setup: reset everything, pair control and lights to cc2530 controller, sniff the group, add lights to group. Switching on/off and changing brightness works like a charm. However: setting the color temperature via IKEA remote control ist not working.

Can I help doing some sniffing? When I press the arrow buttons on the IKEA remote control I receive the following message in Wireshark: `ZCL Scenes: Unknown Command, Seq: xx` There ist no "response" from the "receiving" lamp.
 
```
Frame 581: 55 bytes on wire (440 bits), 55 bytes captured (440 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0xa041, Src: 0xf909
ZigBee Network Layer Data, Dst: Broadcast, Src: 0xf909
ZigBee Application Support Layer Data, Group: 0xb62e, Src Endpt: 1
    Frame Control Field: Data (0x0c)
    Group: 0xb62e
    Cluster: Scenes (0x0005)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 7
ZigBee Cluster Library Frame, Mfr: Unknown (0x117c)
    Frame Control Field: Cluster-specific (0x05)
    Manufacturer Code: Unknown (0x117c)
    Sequence Number: 88
    Command: Unknown (0x07)
    Payload
```

didiht, 2019-03-26:
@bjobo 
Maybe I have to mention that before pairing to z2m all IKEA devices were paired to the IKEA bridge to ge get the newest firmware.

I also sniffed an **arrow-left-click** and identified two frames:
- The first one with external source address of the remote (Frame 2)
- The second one with external source address of the bulb (Frame 4)

```
Frame 2: 115 bytes on wire (920 bits), 115 bytes captured (920 bits) on interface 0
Internet Protocol Version 4, Src: 192.168.1.3, Dst: 192.168.1.3
User Datagram Protocol, Src Port: 17754 (17754), Dst Port: 17754 (17754)
ZigBee Encapsulation Protocol, Channel: 11, Length: 55
IEEE 802.15.4 Data, Dst: 0x6e92, Src: 0x0409
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x0409
Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
ZigBee Application Support Layer Data, Group: 0x7a2a, Src Endpt: 1
    Frame Control Field: Data (0x0c)
    Group: 0x7a2a
    Cluster: Scenes (0x0005)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 25
ZigBee Cluster Library Frame, Mfr: Unknown (0x117c)
    Frame Control Field: Cluster-specific (0x05)
    Manufacturer Code: Unknown (0x117c)
    Sequence Number: 77
    Command: Unknown (0x07)
Data (4 bytes)

0000  01 01 0d 00                                       ....

Frame 4: 115 bytes on wire (920 bits), 115 bytes captured (920 bits) on interface 0
Internet Protocol Version 4, Src: 192.168.1.3, Dst: 192.168.1.3
User Datagram Protocol, Src Port: 17754 (17754), Dst Port: 17754 (17754)
ZigBee Encapsulation Protocol, Channel: 11, Length: 55
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x6e92
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x0409
Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
ZigBee Application Support Layer Data, Group: 0x7a2a, Src Endpt: 1
    Frame Control Field: Data (0x0c)
    Group: 0x7a2a
    Cluster: Scenes (0x0005)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 25
ZigBee Cluster Library Frame, Mfr: Unknown (0x117c)
    Frame Control Field: Cluster-specific (0x05)
    Manufacturer Code: Unknown (0x117c)
    Sequence Number: 77
    Command: Unknown (0x07)
Data (4 bytes)

0000  01 01 0d 00                                       ....
```

Maybe that helps.

If I can do any further investigation just tell me.

didiht, 2019-03-28:
@ripperle @bjobo 
Now I had the same issue: Removing, and pairing again (6 x powercycle) lost the ability of changing the color_temp with the (grouped) remote.

But following procedure leads in my environment to the expected functionality:
- remove the bulb from the zigbee2mqtt bridge
- stop zigbee2mqtt bridge
- pair the bulb to the IKEA bridge with the IKEA remote via Tradfri app
- remove the bulb from the bridge via Tradfri app
- power off the bulb
- start zigbee2mqtt bridge
- power on the bulb nearby the CC253x (no powercycle!)
- add the bulb to the group of the IKEA Remote
- **SUCCESS** color_temp is working again with the IKEA remote

It seems there is a (small?) difference of the initial states (reset with powercycle, reset by pairing and removing from IKEA bridge) that leads to the variance of functionality.
Could you test this please?

@Koenkk 
Maybe we could give a hint in the documentation after others have tested this procedure and confirm that it works?

Koenkk, 2019-03-28:
So it seems the tradfri hub does some special setup, could you sniff the zigbee traffic when pairing it to the hub? http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ripperle, 2019-03-28:
@didiht
I have no IKEA gateway or IKEA app 😁

didiht, 2019-04-01:
@Koenkk 
Finally I could sniff the traffic during pairing. Involved components are IKEA hub, IKEA App, IKEA remote and IKEA bulb LED1545G12 (white spectrum).
But I don't have the Khow How to analyze.
What next? Shall I upload the WS file?

bjobo, 2019-04-03:
@didiht
Hooray! Today I finaly got time to test your procedure. I can confirm that color changing now ist working.
I'd support your hypothesis that the key aspect is not to powercycle the bulb after removing it from the TRADFRI app.
However, this evening I was not able to filter the IKEA TRADFRI connect from the zigbee traffic due to too much 'noise' of other zigbee devices (and missing experience with wireshark and the zigbee protocol). If necessary I'v to try again at the weekend.

> But following procedure leads in my environment to the expected functionality:
>
>     * remove the bulb from the zigbee2mqtt bridge
>
>     * stop zigbee2mqtt bridge
>
>     * pair the bulb to the IKEA [...]

Koenkk, 2019-04-04:
@didiht could you upload it here?

didiht, 2019-04-04:
@bjobo 
I am glad to be able to help :smile:

@Koenkk here it is:
[Ws.zip](https://github.com/Koenkk/zigbee2mqtt/files/3042144/Ws.zip)

djusHa, 2019-05-04:
Hi,
same problem here.

Is there an Solution without Tradfri Bridge?

Devices are LED1624G9 and LED1546G12.

After reset the bulbs and tradfri remote
to factory and pairing with MQTT2Zigbee, no more ability to change colors / color temp with the remote.

After adding the bulbs to Tradfri remotes group, even no luck.

Same sniffer output like https://github.com/Koenkk/zigbee2mqtt/issues/1232#issuecomment-475870840

On/Off and brightness on Tradfri remotes working good.

Koenkk, 2019-05-17:
I've did some further investigation but didn't manage to decrypt the network.

I think the only hope to ever solve this issue is that IKEA shares the required Zigbee commands. I've asked them here: https://www.reddit.com/r/tradfri/comments/bpx9qf/what_magic_setup_does_the_e1524_ikea_tradfri/, fingers crossed..

vogler, 2019-07-17:
Does the coordinator know which bulbs the remote is paired to?
Maybe introduce a device option for the remote to emulate the original behavior by sending commands to change the color temperature on left/right button presses?

BafHS, 2019-07-17:
i have the two problems.
1. Colortemp
my steps:
- http://www.zigbee2mqtt.io/information/binding.html#ikea-tradri-remote-control-e1524
- pair the Device check
- toggle check
- brightness check
- color temp not working.
in FHEM i see, press arrow_left_click or arrow_right_click but no change of color temp.
2. lamp state
- in FHEM the power state not changing, when i switch the lamp with the remote control???
what can i do?

Koenkk, 2019-07-18:
@vogler I think the whole idea of this is that it works when zigbee2mqtt is down, note that your solution can already be achieved via e.g. a home assistant automation.

@BafHS for 2 you can use the reporting feature: http://www.zigbee2mqtt.io/information/report.html

vogler, 2019-07-18:
> that it works when zigbee2mqtt is down
Would be nice, but you wrote it's not easy to do without information from Ikea?

> be achieved via e.g. a home assistant automation
Of course everyone could fix it outside (what are the colors of the scenes?), but IMHO it's better to just make it behave as expected. Less confusion and bug reports.
Is there anything speaking against it?

Koenkk, 2019-07-20:
@vogler for that, first #1716  has to be implemented.

BafHS, 2019-07-23:
thx, it works fine...
> @BafHS for 2 you can use the reporting feature: http://www.zigbee2mqtt.io/information/report.html

groenmarsmannetje, 2019-09-26:
My assumption has always been that this will select the next/previous scene that have been setup with the IKEA app on the Tradfri bridge. The 3 color Temp scenes are just the default. Not using IKEA app and Gateway anymore, so unable to verify this assumption myself.

jd1900, 2020-05-04:
I find your experiences quite awkward, since I got this working from the very first day just using zigbee2mqtt, and I never had the official Tradfri gateway. 

As described here (https://github.com/Koenkk/zigbee2mqtt/issues/3485) I had this working, and suddenly (I can find the trigger, maybe there was one, but I can't recall which) it stopped working. However, I never had the official Tradfri gateway, so there must be something else.

didiht, 2020-05-04:
With the latest firmware update of the remote and/or the bulb something has changed. I have bulbs paired before the upgrade where color temperature still can be controlled by the remote.
With a newly paired bulb it is not possible to control color temp anylonger.

vogler, 2020-05-04:
For me it's the other way around. Changing color temp did not work.
Then I updated my remote and bulbs last week, paired the remote to the coordinator and then to the bulb, and it worked.

jd1900, 2020-05-04:
> For me it's the other way around. Changing color temp did not work.
> Then I updated my remote and bulbs last week, paired the remote to the coordinator and then to the bulb, and it worked.

I'll try to re-pair the bulb and comment back to see if that works. 
Maybe is something related to pairing. I think that in one of my tests I did a direct pairing using the Ikea motion detector which "destroyed" my pairing, and I had to re-pair. 
Maybe that's the trigger which made me loose the functionality. 

I repaired but with the coordinator a bit farther away than the first pairing, and maybe that had something to do.
Well..., and I has an outdated firmware in the first pairing, and this pairing was with the newest pairing...

ngdio, 2020-05-05:
I've set up zigbee2mqtt properly for the first time last week and it's not working for any of my bulbs. I haven't done firmware updates in weeks either.

jd1900, 2020-05-06:
I've tried repairing the bulb. Same thing. 

And I think I know whats the problem. The older firmwares had included different scenes prebuilt. This latest firmware doesn't. So if you re-pair using latest firmware, you "loose" those scenes, that have to be set again into the bulb, probably using genScenes cluster, if that makes sense. 

The thing is, how to "insert" / "add" scenes into the bulb?

vide, 2020-05-09:
After reading all the comments here and in #2069 looks like there is no solution right now :( Is there a way to downgrade the ikea bulbs firmware to get them back to its original settings with the scenes integrated?

vogler, 2020-05-09:
@vide My bulb still has all its scenes and I can change them with the remote. Both updated to the newest firmware.

ngdio, 2020-05-09:
@vogler Did you bind the remotes to the groups before or after the update? This problem only occurs in the latter scenario, as far as I can tell.

jd1900, 2020-05-09:
> @vogler Did you bind the remotes to the groups before or after the update? This problem only occurs in the latter scenario, as far as I can tell.

I'd rather say when was the pairing. Binding shouldn't affect.

vogler, 2020-05-09:
First I updated the remote which then didn't work anymore. So I paired it again: first to the coordinator (press 4 times), then to the bulb (pressed 10s). I could see messages from the remote on the coordinator and the bulb worked with it including left/right for changing scenes.
Then I updated all bulbs. Did not have to pair them. Kept working with coordinator and remote.

vide, 2020-05-09:
@vogler just tried to pair again the remote to the bulb (10 second reset button) and nothing. I would tell you if you want to risk to pair again the bulbs with the controller. The arrows should stop working (no more scenes to control)

vogler, 2020-05-09:
Ok, thanks, than I will avoid re-pairing them :)

connesc, 2020-05-11:
I'm facing the same problem with a color bulb ([IKEA LED1624G9](https://www.zigbee2mqtt.io/devices/LED1624G9.html)).

I have noticed exactly the same things:
1. Pairing the bulb directly with the remote allows to control everything, including scenes.
2. Pairing the remote and the bulb to the coordinator allows to control ON/OFF and brightness, but **not** scenes.
3. Upgrading the bulb while already paired to the coordinator magically restores scene control.
4. Re-pairing the bulb after an upgrade breaks scene control again (game over if no new firmware is available).

My guess is that factory reset (which is the only known way to trigger pairing mode) erases some kind of scene configuration from the bulb. This configuration may be initialized either by a firmware upgrade, or by the Tradfri remote itself when paired directly to the bulb.

If these assumptions are correct, there would be two options:
1. Generate the same commands as the Tradfri remote do when configuring paired bulbs.
2. Find a way to pair bulbs without factory reset.

jd1900, 2020-05-11:
> If these assumptions are correct, there would be two options:
> 
> 1. Generate the same commands as the Tradfri remote does when configuring paired bulbs.
> 2. Find a way to pair bulbs without factory reset.

I'd go for first option.
As linked above it may be related to this: https://github.com/Koenkk/zigbee2mqtt/issues/2069

We need someone with sniffer and both gateways.

Dennis14e, 2020-05-11:
> We need someone with sniffer and both gateways.

I have both gateways (Tradfri & CC26X2R1) and could use a CC2531 as sniffer.
How exactly can I help?

jd1900, 2020-05-14:
> > We need someone with sniffer and both gateways.
> 
> I have both gateways (Tradfri & CC26X2R1) and could use a CC2531 as sniffer.
> How exactly can I help?

The thing would be to sniff a pairing of the bulb with the latest firmware (reset and re-pair in the official gateway). 
But..., we would need someone with enough skills in zigbee2mqtt to implement it afterwards. Anyway, that would be useful anyway. 

Another useful sniffing would be to add / remove scenes from the bulb in the official gateway.

dmak, 2020-05-29:
Please can someone explain how scenes pay a role with regards to the issue? Is setting a color = a scene that needs to be present/initialized in a bulb (hence it is not an issue with IKEA E1524 remote control)?

sjorge, 2020-06-05:
I recently added another puck remote and after updating it, the color selection was indeed working (it wasn't before)... so the upgrade must internally have the remote rebind to the old group and do the setup.

I sadly did not capture the upgrade, but I did just capture me left,right,left,left,right,right on it. Not sure it's helpful, @Koenkk  should have the network key.

https://pkg.blackdot.be/cores/zigbee/ikea_scene_puck.pcapng

Koenkk, 2020-06-07:
@sjorge the commands still seem to be the non standard commands like it used to send before. Perhaps the remote did some setup on the bulb side so that it now responds.

sjorge, 2020-06-07:
Should it in theory be possible to pair a remote to a bulb directly and sniff the traffic?
We won't have the decryption key in that case I guess?

Koenkk, 2020-06-07:
@sjorge if it commissions that to the same network you should be able to sniff that indeed.

drzony, 2020-06-09:
Here are my findings on this issue:
1. Binding procedure
  - Create a group for required devices (LIGHT_GROUP)
  - Unbind controller from default group
  - Bind controller to LIGHT_GROUP
  - Bind bulb/bulbs to LIGHT_GROUP
2. Color temperature change
  - Not working on LED1545G12 with firmware 2.3.007, working on firmware 2.3.050
  - Working on LED1537R6 with firmware 2.0.023
3. Other functions working great

wes1993, 2020-06-15:
I have the same problem.. do you know also if we can downgrade the firmware?

drzony, 2020-06-15:
For my bulbs this is the latest firmware, I just needed to update them (the update was released during the last month). I don't think downgrade is supported by zigbee2mqtt, but there may be some weird manual method to do that.

wes1993, 2020-06-15:
Thanks a lot for your reply :)
So for now I can’t use the option to change the color is correct?

drzony, 2020-06-15:
Which bulbs do you have? What firmware do they have? My LED1545G12 and LED1537R6 are working fine

wes1993, 2020-06-15:
My lamp are LED1537R6/LED1739R5 Firmware version is 2.0.023.
Remote is

drzony, 2020-06-15:
Just use the instructions provided here: https://www.zigbee2mqtt.io/devices/E1524_E1810.html
Make sure your remote has the 2.3.014 firmware

wes1993, 2020-06-15:
I have do all in the wiki but won't work... latest firmware.. :-(
2.3.014.

wes1993, 2020-06-15:
I don' know why.. the brightness work and also onn and off but no color.. :-(

drzony, 2020-06-16:
You can try this:
1. Create a group in config:
```yaml
groups:
  '50':
    friendly_name: my_bulbs
```
2. Remove controller from default group by sending MQTT message to `zigbee2mqtt/bridge/unbind/[CONTROLLER_FRIENDLY_NAME]` payload `default_bind_group`
3. When you press a button on controller you should no get message with group 0 (if not, then do the same as in point 4 but with `[CONTROLLER_FRIENDLY_NAME]` as payload)
4. Remove bulbs from all groups by sending MQTT to `zigbee2mqtt/bridge/group/remove_all` with payload `[BULB_FRIENDLY_NAME]`
5. Add controller to group `zigbee2mqtt/bridge/group/my_bulbs/add` with payload `[CONTROLLER_FRIENDLY_NAME]`
6. Add bulb to group `zigbee2mqtt/bridge/group/my_bulbs/add` with payload `[BULB_FRIENDLY_NAME]`

Remember to replace all things in `[]` with your names.
That's how it worked on my side.

If it still doesn't work, try the following:
1. Open the back of the controller
2. Get the controller close to the bulb and press "link" button on the back of the controller
3. After a few seconds the bulb should flash softly
4. All functions of the controller should work now, but it won't be properly configured in zigbee2mqtt
5. Repeat the configuration steps from the first part of this comment

[148 more replies]

closed 2021-03-13 by github-actions[bot]