Koenkk/zigbee2mqtt #1234: Unable to pair SwannOne Key Fob (SWO-KEF1PA)

qosmio, 2019-03-11:
Hi,

I'm having trouble setting up a [SwannOne Key Fob (SWO-KEF1PA)](https://www.swann.com/au/swo-kef1pa) ([Quickstart Guide](https://www.swann.com/au/downloads/dl/file/id/2049/product/844/smart_home_key_fob_quick_start_guide.pdf)). It's a 4-button remote (panic, arrive, leave, nightmode) that registers as an **IAS Ancillary Control Equipment**. This device doesn't seem to behave well with z2m. After experimenting with creating a new device I was able to pair and configure it but none of the key presses seem to register any incoming messages. I do however see the key presses in the 'cc-znp' output. 

I'm sort of shooting in the dark here since I don't have the hub that goes with this remote to do any packet capture. But I was able to breakdown the key presses based on reading an issue from https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1085 as well as seeing what someone did with [making a device handler in ST](https://github.com/jruss56880/keypad/blob/80c0e8b58ab54e524a4ace59ebfcac774e5a7449/devicetypes/RudiP/Swann-One-Key-Fob.src/Swann-One-Key-Fob.groovy).

I'm still trying to learn the zigbee protocol but confused how else to proceed. I'm not sure if this will need an extension or would a device handler suffice. 

## devices.js
```javascript
    {
        zigbeeModel: ['SWO-KEF1PA'],
        model: 'SWO-KEF1PA',
        vendor: 'SwannONe',
        description: 'Key Fob Remote',
        supports: 'panic, home, away, night',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];
            execute(device, actions, callback, 350);
        },
    },
```

## Logs

I've highlighted a few of the key messages from the debug log below. You can view the full log here: https://pastebin.com/raw/NXFQjnnf

**fromZigbee type warnings**

```actionscript
 warn: No converter available for 'SWO-KEF1PA' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"hwVersion":1,"dateCode":"20160606        "}}'
 warn: No converter available for 'SWO-KEF1PA' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"SwannONe","modelId":"SWO-KEF1PA","powerSource":3}}'
 warn: No converter available for 'SWO-KEF1PA' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"identifyTime":0}}'
 warn: No converter available for 'SWO-KEF1PA' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}'
 warn: No converter available for 'SWO-KEF1PA' with cid 'genPowerCfg', type 'devChange' and data '{"cid":"genPowerCfg","data":{"batteryAlarmState":0}}'
 warn: No converter available for 'SWO-KEF1PA' with cid 'ssIasZone', type 'devChange' and data '{"cid":"ssIasZone","data":{"zoneState":1,"zoneType":0,"zoneStatus":16,"iasCieAddr":"0x00124b001936639c","zoneId":23}}'
```
**cc-znp**

```actionscript
  cc-znp { sof: 254,
  cc-znp   len: 33,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 1281,
  cc-znp      srcaddr: 25344,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 11,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 34,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 8812311,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 13,
  cc-znp      data: <Buffer 11 4d 00 00 00 00 00 00 00 00 00 00 17> },
  cc-znp   fcs: 42,
  cc-znp   csum: 42 } +12s
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +16s
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +12s
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +17ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 7 +15s
[... 234 lines cut ...]
                        }
                    }
                },
                "ssIasWd": {
                    "dir": {
                        "value": 2
                    }
                },
                "genIdentify": {
                    "dir": {
                        "value": 3
                    },
                    "acls": {
                        "identifyTime": "RW"
                    },
                    "attrs": {
                        "identifyTime": 0
                    }
                }
            }
        }
    },
    "_id": "THxBpO8t64WDLCDg"
}
```

Koenkk, 2019-03-11:
Can you provide the `DEBUG=* npm start` logging when running zigbee2mqtt 1.2.1 and pressing some buttons? (this adds some extra information that I need).

qosmio, 2019-03-11:
It should be in the pastebin I linked. The key presses are on the bottom. 

... However

I think I have it working I needed to update ```zigbee-shepherd/lib/components/af.js``` and add ```'arm', 'panic'``` to the list of command ids 

```javascript
const cmdIDs = ['on', 'offWithEffect', 'step', 'stop', 'hueNotification', //Philips-RWL020-RWL021
                'off', 'stepColorTemp', 'moveWithOnOff', 'move', 'moveHue', 'moveToSaturation', //OSRAM-Switch 4x EU-LIGHTIFY
                'stopWithOnOff', 'moveToLevelWithOnOff', 'toggle', 'tradfriArrowSingle', 'tradfriArrowHold', 'tradfriArrowRelease',
                'stepWithOnOff', 'moveToColorTemp', 'moveToColor', 'onWithTimedOff', 'recall', 'arm', 'panic' //SwannOne Key Fob SWO-KEF1PA
               ];
``` 

And with the following ```devices.js```

```javascript
{
        zigbeeModel: ['SWO-KEF1PA'],
        model: 'SWO-KEF1PA',
        vendor: 'SwannONe',
        description: 'Key Fob Remote',
        supports: 'panic, home, away, night',
        fromZigbee: [fz.KEF1PA_arm, fz.KEF1PA_panic, fz.generic_battery],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.bind('ssIasZone', coordinator, cb),
                 // This ended up making all the messages come from the cie app, so ended up getting ignored in Z2M
                //(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            ];

            execute(device, actions, callback, 250);
        },
    },
```
and the adding to ```fromZigbee.js```

```javascript
    KEF1PA_arm: {
        cid: 'ssIasAce',
        type: 'cmdArm',
        convert: (model, msg, publish, options) => {
            const value = msg.data.data['armmode'];
            const modeLookup = {
                0: 'home',
                2: 'night',
                3: 'away',
            };
            return { armmode : modeLookup[value] };
        },
    },
    KEF1PA_panic: {
        cid: 'ssIasAce',
        type: 'cmdPanic',
        convert: (model, msg, publish, options) => {
            return { armmode : 'panic' };
        },
    },
```

Then I could see it properly showing up in the logs. I even have it working in Home Assistant.

```actionscript
Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":0}}' of device 'SWO-KEF1PA'
MQTT publish: topic 'zigbee2mqtt/keyfob/Key Fob 2', payload '{"armmode":"home","linkquality":23}'
Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":2}}' of device 'SWO-KEF1PA'
MQTT publish: topic 'zigbee2mqtt/keyfob/Key Fob 2', payload '{"armmode":"night","linkquality":23}'
Received zigbee message of type 'cmdArm' with data '{"cid":"ssIasAce","data":{"armmode":3}}' of device 'SWO-KEF1PA'
MQTT publish: topic 'zigbee2mqtt/keyfob/Key Fob 2', payload '{"armmode":"away","linkquality":23}'
Received zigbee message of type 'cmdPanic' with data '{"cid":"ssIasAce","data":{}}' of device 'SWO-KEF1PA'
MQTT publish: topic 'zigbee2mqtt/keyfob/Key Fob 2', payload '{"armmode":"panic","linkquality":25}'
```

There's still a few issues I'm trying to figure out, mostly related to the "configuration" portion I based it mostly off https://github.com/Koenkk/zigbee2mqtt/issues/801 where I wasn't writing the ```iasCieAddr``` to ```ssIasZone``` instead just binding. I'm not sure exactly what that does tbh. But for the most part it seems to be working. Anything else you think I should try?

Koenkk, 2019-03-12:
If everything works, the `ssIasZone` stuff isn't needed (I also don't expect that to be necessary for this kind of device). Could you make a pr?

qosmio, 2019-03-13:
I tried to remove the ```ssIasZone``` portion out but it doesn't seem to be behaving well. Still need to do more tests. Will open a PR shortly. Thanks!

jedisha, 2019-08-08:
Hi there,

just wondering if you had got this to work? I just bought a Swann One Keyfob and have managed to pair it to my Hass.io and zigbee2mqtt but it only appears to register 1 entity: sensor.0x00124b00062839c2_action

I'm pretty much a noob but have got this far - just wondering if you had any pointers?

Thanks!
P

fred-c1, 2019-10-23:
@qosmio    maybe you can help here:
It seems that I have the same problem than @jedisha and that you had ("none of the key presses seem to register any incoming messages"):

- pairing worked 
- when I press a button on the remote I can see the green LED on the remote
- when I press a button on the remote I don't receive any message from the MQTT broker

I am running the latest version of zigbee2MQTT (1.6.0) on hassio
I am wondering if this device has really been integrated in zigbee2MQTT.
Thanks

Koenkk, 2019-10-25:
@fred-c1 that's an interesting finding, first, can you make sure you are on the latest CC2530 firmware?

If that doesn't fix it, could you sniff the traffic while pressing the button https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

qosmio, 2019-10-25:
@fed-c1 I've noticed in the past that the key fob had issues pairing through a router (if your setup has any). Try pairing the key fob directly next to the  coordinator.

fred-c1, 2019-11-07:
@Koenkk   and   @qosmio 
My problem has been solved by upgrading the firmware of my coordinator from CC2530_CC2591ZNP-Prod_20181123
to the latest F/W = CC2530_CC2591_DEFAULT_20190608

A few details:
- power down the raspberry pi 3B+
- upgrade CC2530 F/W
- power up the raspberry pi 3B+
- SWO-KEF1PA works fine with zigbee2MQTT

- a few hours later SWO-KEF1PA is no more in my network
- this could be due to OSRAM Smart+ plug (AB3257001NJ) (has router capability)
which seems to be causing many problems
- pairing of SWO-KEF1PA
- SWO-KEF1PA works fine
- this has now been working for about 10 days

Thanks for the help!

closed 2020-01-13 by stale[bot]