Koenkk/zigbee2mqtt #1236: sending device information via mqtt

directman66, 2019-03-11:
I integrate your project into the smart home management system majordomo  https://mjdm.ru/  (https://github.com/directman66/majordomo-zigbee2mqtt/ ). Link to russian community https://mjdm.ru/forum/viewtopic.php?f=5&t=6011.  When integrating via mashing, I will encounter the need to parse a database file. I need the database.db database file to get the modelId, which is missing in the message on the server. Is it possible to make the information from the database duplicated in the messages to the server?

I would like to receive in the message mqtt the following data:  type":"Router","ieeeAddr":"0x00158d00022bc3ab","nwkAddr":42799,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.ctrl_neutral2" etc

Koenkk, 2019-03-11:
This can be achieved by sending an empty payload to `zigbee2mqtt/bridge/config/devices` (see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgelog)

directman66, 2019-03-14:
Thanks, all the data on the main received at home. It turns out in the answer there is no information about the IEEE address of the coordinator. Is there another request for this?

Koenkk, 2019-03-14:
Implemented!

closed 2019-03-14 by Koenkk