Koenkk/zigbee2mqtt #1237: new installation is buggy

arteck, 2019-03-11:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.79-v7+
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.2.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.2.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

Koenkk, 2019-03-11:
Try upgrading npm to the latest version

arteck, 2019-03-11:
npm@6.4.1

aXionDevelopment, 2019-03-11:
It looks like I have the same issues, currently I can't get Zigbee2mqtt to work anymore

0 info it worked if it ends with ok
1 verbose cli [ '/home/pi/.nvm/versions/node/v10.15.3/bin/node',
1 verbose cli   '/home/pi/.nvm/versions/node/v10.15.3/bin/npm',
1 verbose cli   'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/home/pi/.nvm/versions/node/v10.15.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/home/pi/.nvm/versions/node/v10.15.3/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.98-v7+
17 verbose argv "/home/pi/.nvm/versions/node/v10.15.3/bin/node" "/home/pi/.nvm/versions/node/v10.15.3/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.2.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.2.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

The output after an attempt to start Zigbee2mqtt from de command prompt:

`pi@hassbian:/opt/zigbee2mqtt $ debug=* npm start

> zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/util/logger.js:10
const level = settings.get().advanced.log_level;
                                     ^

TypeError: Cannot read property 'log_level' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:10:38)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/mqtt.js:2:16)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-03-11T14_26_47_609Z-debug.log
`

Koenkk, 2019-03-11:
current npm is 6.9, update with `sudo npm install -g npm`

aXionDevelopment, 2019-03-12:
@Koenkk, for me, the npm version didn't make any difference. Still the same error as with version 6.4.1.  I will look into it more, when I have time.

Koenkk, 2019-03-12:
Can you share your configuration.yaml?

aXionDevelopment, 2019-03-12:
Sure:

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d00027210e3':
    friendly_name: z_sens1
    retain: false
  '0x00158d00029a9fe5':
    friendly_name: z_pir1
    retain: false
  '0x00158d00029a9fd1':
    friendly_name: z_pir2
    retain: false
  '0x00158d00027aeeda':
    friendly_name: z_sens2
    retain: false
  '0x00158d000272b3ad':
    friendly_name: z_sens3
    retain: false
  '0x00158d0002b56c71':
    friendly_name: z_sens4
    retain: false
  '0x00158d0002ad43fb':
    friendly_name: z_smoke_1
    retain: false
  '0x90fd9ffffe1ee424':
    friendly_name: z_tradfri_1
    retain: false
  '0x000b57fffeb944bb':
    friendly_name: z_tradfri_2
    retain: false
  '0x000b57fffe139166':
    friendly_name: z_dimmer_1
    retain: false
  '0x00158d0002b56888':
    friendly_name: z_door_1
    retain: false
  '0x00158d0002c35228':
    friendly_name: z_door_2
    retain: false
  '0x000b57fffe96db4a':
    friendly_name: z_tradfri_4
    retain: false
  '0x000b57fffe96c995':
    friendly_name: z_tradfri_3
    retain: false
  '0x000b57fffe8fcdf8':
    friendly_name: z_tradfri_5
    retain: false
  '0x00158d0002b48bdd':
    friendly_name: z_switch_2
    retain: false
  '0x00158d000254ce3a':
    friendly_name: z_switch_1
    retain: false
  '0x00158d0002251b8e':
    friendly_name: z_socket_1
    retain: false
  '0x00158d000224a28d':
    friendly_name: z_socket_2
    retain: false
  '0x00158d0002753937':
    friendly_name: z_sens5
    retain: false
advanced:
  # availability_timeout: 60
  # pan_id: 0x1a63
```

Koenkk, 2019-03-12:
Fixed in the dev branch, if you want to stay on 1.2.1, use the following `configuration.yaml`.

```
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d00027210e3':
    friendly_name: z_sens1
    retain: false
  '0x00158d00029a9fe5':
    friendly_name: z_pir1
    retain: false
  '0x00158d00029a9fd1':
    friendly_name: z_pir2
    retain: false
  '0x00158d00027aeeda':
    friendly_name: z_sens2
    retain: false
  '0x00158d000272b3ad':
    friendly_name: z_sens3
    retain: false
  '0x00158d0002b56c71':
    friendly_name: z_sens4
    retain: false
  '0x00158d0002ad43fb':
    friendly_name: z_smoke_1
    retain: false
  '0x90fd9ffffe1ee424':
    friendly_name: z_tradfri_1
    retain: false
  '0x000b57fffeb944bb':
    friendly_name: z_tradfri_2
    retain: false
  '0x000b57fffe139166':
    friendly_name: z_dimmer_1
    retain: false
  '0x00158d0002b56888':
    friendly_name: z_door_1
    retain: false
  '0x00158d0002c35228':
    friendly_name: z_door_2
    retain: false
  '0x000b57fffe96db4a':
    friendly_name: z_tradfri_4
    retain: false
  '0x000b57fffe96c995':
    friendly_name: z_tradfri_3
    retain: false
  '0x000b57fffe8fcdf8':
    friendly_name: z_tradfri_5
    retain: false
  '0x00158d0002b48bdd':
    friendly_name: z_switch_2
    retain: false
  '0x00158d000254ce3a':
    friendly_name: z_switch_1
    retain: false
  '0x00158d0002251b8e':
    friendly_name: z_socket_1
    retain: false
  '0x00158d000224a28d':
    friendly_name: z_socket_2
    retain: false
  '0x00158d0002753937':
    friendly_name: z_sens5
    retain: false
```

closed 2019-03-12 by Koenkk