Koenkk/zigbee2mqtt #1239: Again mistakes!

salopette, 2019-03-11:
Hello, I have constantly crashes and mistakes.
What could be the problem?

https://paste.ubuntu.com/p/dfjpxZzHk5/

Fabiancrg, 2019-03-11:
Hi

I had the some issues today after upgrading to zigbee2mqtt 1.2.1 and firmware 20190223, I can see some errors in your log I had too (like "failed to ping device" and "AF data request fails, status code: 205. No network route."
I had also the following warnings:
3/11/2019, 2:00:58 PM - warn: No converter available for 'ZNLDP12LM' with cid 'msTemperatureMeasurement', type 'attReport' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2028}}'
3/11/2019, 2:00:58 PM - warn: No converter available for 'ZNLDP12LM' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2028}}'
3/11/2019, 2:00:58 PM - warn: No converter available for 'ZNLDP12LM' with cid 'msRelativeHumidity', type 'attReport' and data '{"cid":"msRelativeHumidity","data":{"measuredValue":4562}}'
3/11/2019, 2:00:58 PM - warn: No converter available for 'ZNLDP12LM' with cid 'msPressureMeasurement', type 'attReport' and data '{"cid":"msPressureMeasurement","data":{"16":10094,"20":-1,"measuredValue":1009}}'
3/11/2019, 2:00:58 PM - warn: No converter available for 'ZNLDP12LM' with cid 'msRelativeHumidity', type 'devChange' and data '{"cid":"msRelativeHumidity","data":{"measuredValue":4562}}'
3/11/2019, 2:00:58 PM - warn: No converter available for 'ZNLDP12LM' with cid 'msPressureMeasurement', type 'devChange' and data '{"cid":"msPressureMeasurement","data":{"16":10094,"20":-1,"measuredValue":1009}}'

Very starnge as the ZNLDP12LM are LED bulbs (routers).

It seems to be OK now, what I did is the following:
1- Remove power from LED bulb
2- Remove device using zigbee2mqtt/bridge/config/remove
3- Stop zigbee2mqtt
4- Reset the LED bulb
5- Restart zigbee2mqtt

I did that with all router devices, all other end devices were back without repairing them (some Xiaomi Mija needed to push on the button once to be back).

Fabiancrg, 2019-03-11:
Forget to say, that after restarting zigbee2mqtt, I re-paired all routers of course.

Fabiancrg, 2019-03-12:
After a few hours, all messages are back so it seems doing all that was just for nothing.
But except the warning above, everything seems working OK.

Koenkk, 2019-03-12:
@salopette could you try to change https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/zigbeeQueue.js#L1 to `const maxSimultaneouslyRunning = 2;`

salopette, 2019-03-13:
@Koenkk 
Thanks, I changed it this morning and now seems to be running without error! I keep watching!

closed 2019-05-19 by stale[bot]