Koenkk/zigbee2mqtt #1240: topics send 2 times

Bastiencc, 2019-03-11:
I update to the last version.
eveything seems ok (but I need to re-paire all my device... lot of Hue dimmer switch).

I have a issue with the xiaomi aqara bulb, when I toggle it on HA, topics seems to be sent 2 times : 
`Mar 11 20:31:15 raspberrypi npm[3935]:   zigbee2mqtt:info 3/11/2019, 8:31:15 PM Zigbee publish to device '0x00158d0002d6f569', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 11 20:31:15 raspberrypi npm[3935]:   zigbee2mqtt:info 3/11/2019, 8:31:15 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":76,"brightness":33}'
Mar 11 20:31:15 raspberrypi npm[3935]:   zigbee2mqtt:info 3/11/2019, 8:31:15 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":73,"brightness":33}'
Mar 11 20:31:18 raspberrypi npm[3935]:   zigbee2mqtt:info 3/11/2019, 8:31:18 PM Zigbee publish to device '0x00158d0002d6f569', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 11 20:31:18 raspberrypi npm[3935]:   zigbee2mqtt:info 3/11/2019, 8:31:18 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"OFF","linkquality":73,"brightness":33}'
Mar 11 20:31:18 raspberrypi npm[3935]:   zigbee2mqtt:info 3/11/2019, 8:31:18 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"OFF","linkquality":70,"brightness":33}'
`

Bastiencc, 2019-03-12:
Sometimes it's publish twice, but not with the same state (still from HA)

```
Mar 12 10:09:13 raspberrypi npm[28976]:   zigbee2mqtt:info 3/12/2019, 10:09:13 AM Zigbee publish to device '0x00158d0002d6f569', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 12 10:09:13 raspberrypi npm[28976]:   zigbee2mqtt:info 3/12/2019, 10:09:13 AM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":65,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'
Mar 12 10:09:13 raspberrypi npm[28976]:   zigbee2mqtt:info 3/12/2019, 10:09:13 AM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":65,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'

Mar 12 10:09:37 raspberrypi npm[28976]:   zigbee2mqtt:error 3/12/2019, 10:09:37 AM Failed to ping 0x00158d0002d6f569
Mar 12 10:09:41 raspberrypi npm[28976]:   zigbee2mqtt:info 3/12/2019, 10:09:41 AM Zigbee publish to device '0x00158d0002d6f569', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 12 10:09:41 raspberrypi npm[28976]:   zigbee2mqtt:info 3/12/2019, 10:09:41 AM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"OFF","linkquality":65,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'
Mar 12 10:09:41 raspberrypi npm[28976]:   zigbee2mqtt:info 3/12/2019, 10:09:41 AM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":65,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'

```

GForce2010, 2019-03-12:
This is normal behaviour in the latest version see #1212

Koenkk, 2019-03-12:
Duplicate messages are expected, however the following is not: *Sometimes it's publish more than twice, and not with the same state ! (still from HA)*

Could you publish the debug log of that?

Bastiencc, 2019-03-12:
Light is off. I toggle it from HA : 
```
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:debug 3/12/2019, 9:21:59 PM Received MQTT message on 'zigbee2mqtt/aqara_bulb_1/set' with data '{"state": "ON"}'
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:info 3/12/2019, 9:21:59 PM Zigbee publish to device '0x00158d0002d6f569', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:info 3/12/2019, 9:21:59 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":63,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:debug 3/12/2019, 9:21:59 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":1,"onTime":0,"offWaitTime":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:info 3/12/2019, 9:21:59 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":63,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:debug 3/12/2019, 9:21:59 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Mar 12 21:21:59 raspberrypi npm[32048]:   zigbee2mqtt:debug 3/12/2019, 9:21:59 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Mar 12 21:22:15 raspberrypi npm[32048]:   zigbee2mqtt:error 3/12/2019, 9:22:15 PM Cannot get the Node Descriptor of the Device: 0x001788010408cd64 (Error: Timed out after 10000 ms)
Mar 12 21:22:20 raspberrypi npm[32048]:   zigbee2mqtt:debug 3/12/2019, 9:22:20 PM Received MQTT message on 'zigbee2mqtt/aqara_bulb_1/set' with data '{"state": "OFF"}'
Mar 12 21:22:20 raspberrypi npm[32048]:   zigbee2mqtt:info 3/12/2019, 9:22:20 PM Zigbee publish to device '0x00158d0002d6f569', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 12 21:22:20 raspberrypi npm[32048]:   zigbee2mqtt:info 3/12/2019, 9:22:20 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"OFF","linkquality":63,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'
Mar 12 21:22:20 raspberrypi npm[32048]:   zigbee2mqtt:debug 3/12/2019, 9:22:20 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":1,"onTime":0,"offWaitTime":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Mar 12 21:22:20 raspberrypi npm[32048]:   zigbee2mqtt:info 3/12/2019, 9:22:20 PM MQTT publish: topic 'zigbee2mqtt/aqara_bulb_1', payload '{"state":"ON","linkquality":63,"brightness":33,"color_temp":338,"color":{"x":0.333,"y":0.343}}'
```
The light is off, but still "on" on HA...

Koenkk, 2019-03-13:
Should be fixed in the dev branch (edge for hassio).

Bastiencc, 2019-03-13:
Thanks, I'll try, could you tell me how to update to dev branch?

Koenkk, 2019-03-13:
http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Bastiencc, 2019-03-13:
It seems to be fixed, thanks!

Koenkk, 2019-03-14:
Great!

closed 2019-03-14 by Koenkk