Koenkk/zigbee2mqtt #1241: next.func is not a function don't starting

llandryV, 2019-03-11:
It's fresh install.

`zigbee2mqtt:debug 3/11/2019, 9:35:52 PM Mounted the cieApp (epId 11)
/opt/zigbee2mqtt/lib/util/zigbeeQueue.js:63
            next.func((error) => this.handleJobComplete(next, error));
                 ^

TypeError: next.func is not a function
    at ZigbeeQueue.executeNext (/opt/zigbee2mqtt/lib/util/zigbeeQueue.js:63:18)
    at Timeout.timer.setInterval [as _onTimeout] (/opt/zigbee2mqtt/lib/util/zigbeeQueue.js:33:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)`

npm version and node version it's ok.

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/o$
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:$
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.98-v7+
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.9.0
`

Can you help me ?

llandryV, 2019-03-11:
When I'm delete `experimental:
  livolo: true
`
for configuration.yaml is starting.

Koenkk, 2019-03-12:
Should be fixed in the dev branch.

closed 2019-05-12 by Koenkk