Koenkk/zigbee2mqtt #1246: Flashing CC2530 as Router with Arduino + CCLib

GerSant, 2019-03-13:
I try to flash my new CC2530 using my Arduino UNO loaded with the CCLib Proxy and i obtain this meesage:

INFO: Found a CC2530 chip on COM5

Chip information:
Chip ID : 0xa524
Flash size : 256 Kb
Page size : 2 Kb
SRAM size : 8 Kb
USB : No
Traceback (most recent call last):
File "C:\Users\T15412\Downloads\CCLib-master\Python\cc_write_flash.py", line 48, in 
hexFile.load()
File "C:\Users\T15412\Downloads\CCLib-master\Python\cclib\cchex.py", line 128, in load
self._loadHex()
File "C:\Users\T15412\Downloads\CCLib-master\Python\cclib\cchex.py", line 367, in _loadHex
raise IOError("Line %i: Unknown record type %02x" % (i, bType))
IOError: Line 12527: Unknown record type 05

What can be wrong?

Thanks

sesceu, 2019-03-13:
I ran into the exact same problem. After reading https://github.com/wavesoft/CCLib/issues/16 I removed line 12527 (the second to last line of the hex file). I'm still flashing, but at least I'm not getting this error anymore.

GerSant, 2019-03-14:
Thank you very much! worked for me for flahsing the CC2530 as Router, now i still flashing the CC2531 as coordinator

sesceu, 2019-03-14:
You're welcome. I'm glad I could help.

GerSant, 2019-03-15:
This issue can be closed :)

closed 2019-05-21 by stale[bot]