Koenkk/zigbee2mqtt #1247: zigbee service don't work well

medalle23, 2019-03-13:
Hello,
everytime when i restart the hardware(z83), the zigbee2mqtt don't work well.
first step: 
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since mer. 2019-03-13 14:53:37 CET; 11min ago
 Main PID: 913 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─ 913 npm
           ├─1004 sh -c node index.js
           └─1005 node index.js

mars 13 14:54:43 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:54:43 wallsingleswitch1 (0x00158d0002c608c1): QBKG11LM - Xiaomi Aqara single key wired wall switch (Ro
mars 13 14:54:43 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:54:43 Zigbee: disabling joining new devices.
mars 13 14:54:43 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:54:43 Connecting to MQTT server at mqtt://localhost
mars 13 14:54:43 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:54:43 zigbee-shepherd ready
mars 13 14:54:53 igenyx npm[913]:   zigbee2mqtt:error 2019-3-13 14:54:53 Not connected to MQTT server!
mars 13 14:54:55 igenyx npm[913]:   zigbee2mqtt:error 2019-3-13 14:54:55
mars 13 14:54:55 igenyx npm[913]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/l
mars 13 14:55:01 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:55:01 Connected to MQTT server
mars 13 14:55:01 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:55:01 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
mars 13 14:55:01 igenyx npm[913]:   zigbee2mqtt:info 2019-3-13 14:55:01 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":fals
lines 1-19/19 (END)

As  you see, the service start, but the usb led if off (no light)

second step: i need to restart the service again 

● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since mer. 2019-03-13 15:08:29 CET; 6s ago
 Main PID: 2978 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─2978 npm
           ├─2991 sh -c node index.js
           └─2992 node index.js

mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 zigbee-shepherd ready
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 Connected to MQTT server
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/doorsensor1', payload '{"contact":true,"linkquality":144,"batt
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/motionsensor1', payload '{"illuminance":27,"linkquality":131,"
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/humiditysensor1', payload '{"temperature":34.13,"linkquality":
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/wirelessdoubleswitch1', payload '{"battery":100,"voltage":3155
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/wallsocketplug1', payload '{"state":"ON","linkquality":170,"po
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/wallsingleswitch1', payload '{"state":"OFF","linkquality":147,
mars 13 15:08:34 igenyx npm[2978]:   zigbee2mqtt:info 2019-3-13 15:08:34 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":fal
lines 1-19/19 (END)

when i restart, it works, the usb led is on (green led)..

I do it everytime..restart again the service manually

i make all change with service config file, it didn't work..

zigbee2mqtt.service :

[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Type=simple
User=zakabox
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
ExecStart=/usr/bin/npm start
#ExecStop=/bin/kill -SIGKILL $MAINPID
Restart=always
#ExecReload=/usr/bin/npm restart
TimeoutStartSec=120
#KillMode=process

[Install]
WantedBy=multi-user.target

Someone help!!!

medalle23, 2019-03-13:
sudo journalctl -u zigbee2mqtt.service -f
-- Logs begin at mer. 2019-03-13 17:45:06 CET. --
mars 13 18:02:49 igenyx npm[3740]: npm ERR!     /home/zakabox/.npm/_logs/2019-03-13T17_02_49_891Z-debug.log
mars 13 18:02:49 igenyx systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
mars 13 18:02:49 igenyx systemd[1]: zigbee2mqtt.service: Unit entered failed state.
mars 13 18:02:49 igenyx systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
mars 13 18:02:50 igenyx systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
mars 13 18:02:50 igenyx systemd[1]: Stopped zigbee2mqtt.
mars 13 18:02:50 igenyx systemd[1]: Started zigbee2mqtt.
mars 13 18:02:51 igenyx npm[3836]: > zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
mars 13 18:02:51 igenyx npm[3836]: > node index.js
mars 13 18:02:51 igenyx npm[3836]:   zigbee2mqtt:info 2019-3-13 18:02:51 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-13.18-02-51'
mars 13 18:02:52 igenyx npm[3836]:   zigbee2mqtt:info 2019-3-13 18:02:52 Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
mars 13 18:02:52 igenyx npm[3836]:   zigbee2mqtt:info 2019-3-13 18:02:52 Starting zigbee-shepherd
mars 13 18:02:54 igenyx npm[3836]:   zigbee2mqtt:info 2019-3-13 18:02:54 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
mars 13 18:03:54 igenyx npm[3836]:   zigbee2mqtt:info 2019-3-13 18:03:54 Starting zigbee-shepherd
mars 13 18:03:55 igenyx npm[3836]:   zigbee2mqtt:error 2019-3-13 18:03:55 Error while starting zigbee-shepherd!
mars 13 18:03:55 igenyx npm[3836]:   zigbee2mqtt:error 2019-3-13 18:03:55 Press the reset button on the stick (the one closest to the USB) and start again
mars 13 18:03:55 igenyx npm[3836]:   zigbee2mqtt:error 2019-3-13 18:03:55 Failed to start
mars 13 18:03:55 igenyx npm[3836]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
mars 13 18:03:55 igenyx npm[3836]:   zigbee2mqtt:error 2019-3-13 18:03:55 Exiting...
mars 13 18:03:55 igenyx npm[3836]: npm ERR! code ELIFECYCLE
mars 13 18:03:55 igenyx npm[3836]: npm ERR! errno 1
mars 13 18:03:55 igenyx npm[3836]: npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
mars 13 18:03:55 igenyx npm[3836]: npm ERR! Exit status 1
mars 13 18:03:55 igenyx npm[3836]: npm ERR!
mars 13 18:03:55 igenyx npm[3836]: npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
mars 13 18:03:55 igenyx npm[3836]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
mars 13 18:03:55 igenyx npm[3836]: npm ERR! A complete log of this run can be found in:
mars 13 18:03:55 igenyx npm[3836]: npm ERR!     /home/zakabox/.npm/_logs/2019-03-13T17_03_55_885Z-debug.log
mars 13 18:03:55 igenyx systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
mars 13 18:03:55 igenyx systemd[1]: zigbee2mqtt.service: Unit entered failed state.
mars 13 18:03:55 igenyx systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
mars 13 18:03:56 igenyx systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
mars 13 18:03:56 igenyx systemd[1]: Stopped zigbee2mqtt.
mars 13 18:03:56 igenyx systemd[1]: Started zigbee2mqtt.
mars 13 18:03:57 igenyx npm[3932]: > zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
mars 13 18:03:57 igenyx npm[3932]: > node index.js
mars 13 18:03:57 igenyx npm[3932]:   zigbee2mqtt:info 2019-3-13 18:03:57 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-13.18-03-57'
mars 13 18:03:58 igenyx npm[3932]:   zigbee2mqtt:info 2019-3-13 18:03:58 Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
mars 13 18:03:58 igenyx npm[3932]:   zigbee2mqtt:info 2019-3-13 18:03:58 Starting zigbee-shepherd
mars 13 18:04:00 igenyx npm[3932]:   zigbee2mqtt:info 2019-3-13 18:04:00 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
mars 13 18:05:00 igenyx npm[3932]:   zigbee2mqtt:info 2019-3-13 18:05:00 Starting zigbee-shepherd
mars 13 18:05:01 igenyx npm[3932]:   zigbee2mqtt:error 2019-3-13 18:05:01 Error while starting zigbee-shepherd!
mars 13 18:05:01 igenyx npm[3932]:   zigbee2mqtt:error 2019-3-13 18:05:01 Press the reset button on the stick (the one closest to the USB) and start again
mars 13 18:05:01 igenyx npm[3932]:   zigbee2mqtt:error 2019-3-13 18:05:01 Failed to start
mars 13 18:05:01 igenyx npm[3932]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}

Koenkk, 2019-03-14:
Reading your service defintion, this should also restart zigbee2mqtt when it fails to starts.

What is the difference between this and the manual restart?

medalle23, 2019-03-14:
it can't start automatically... everytime i need to restart again via ssh command.

medalle23, 2019-03-14:
the restarting (service) don't work well... i don't understand why... i maked different change on service config file... problem is the same...

this error come everytime

mars 13 18:03:55 igenyx npm[3836]: {"message":"request timeout","stack":"Error: request timeout\n at CcZnp. (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n at Object.onceWrapper (events.js:277:13)\n at CcZnp.emit (events.js:189:13)\n at Timeout. (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n at ontimeout (timers.js:436:11)\n at tryOnTimeout (timers.js:300:5)\n at listOnTimeout (timers.js:263:5)\n at Timer.processTimers (timers.js:223:10)"}

Koenkk, 2019-03-15:
This is a typical communication problem, what is the output of `ls /dev`?

medalle23, 2019-03-15:
zakabox@igenyx:~$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 mars  15 12:00 usb-Arduino__www.arduino.cc__0042_95635333431351405141-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 mars  15 12:00 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018DF3499-if00 -> ../../ttyACM1

Koenkk, 2019-03-15:
What is in your `configuration.yaml`?

medalle23, 2019-03-15:
homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
  availability_timeout: 1

medalle23, 2019-03-15:
I added the timeout believing that it was going to solve this problem (several hours after the zigbee2mqtt is no longer functional)

Koenkk, 2019-03-15:
@medalle23 so first it works but after some time not? could you provide the log of this?

medalle23, 2019-03-18:
Hello @Koenkk 
after 24 hour, it keept working...
So after 2 days, that no longer works
here is the log:

Last login: Sat Mar 16 16:04:15 2019 from 192.168.137.4
zakabox@igenyx:~$ sudo journalctl -u zigbee2mqtt.service -f
[sudo] Mot de passe de zakabox :
-- Logs begin at lun. 2019-03-18 09:25:04 CET. --
mars 18 15:21:32 igenyx systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
mars 18 15:21:32 igenyx systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
mars 18 15:21:32 igenyx systemd[1]: Stopped zigbee2mqtt.
mars 18 15:21:32 igenyx systemd[1]: Started zigbee2mqtt.
mars 18 15:21:33 igenyx npm[16301]: > zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
mars 18 15:21:33 igenyx npm[16301]: > node index.js
mars 18 15:21:34 igenyx npm[16301]:   zigbee2mqtt:info 2019-3-18 15:21:34 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-18.15-21-34'
mars 18 15:21:34 igenyx npm[16301]:   zigbee2mqtt:info 2019-3-18 15:21:34 Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
mars 18 15:21:34 igenyx npm[16301]:   zigbee2mqtt:info 2019-3-18 15:21:34 Starting zigbee-shepherd
mars 18 15:21:36 igenyx npm[16301]:   zigbee2mqtt:info 2019-3-18 15:21:36 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
mars 18 15:22:36 igenyx npm[16301]:   zigbee2mqtt:info 2019-3-18 15:22:36 Starting zigbee-shepherd
mars 18 15:22:38 igenyx npm[16301]:   zigbee2mqtt:error 2019-3-18 15:22:38 Error while starting zigbee-shepherd!
mars 18 15:22:38 igenyx npm[16301]:   zigbee2mqtt:error 2019-3-18 15:22:38 Press the reset button on the stick (the one closest to the USB) and start again
mars 18 15:22:38 igenyx npm[16301]:   zigbee2mqtt:error 2019-3-18 15:22:38 Failed to start
mars 18 15:22:38 igenyx npm[16301]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
mars 18 15:22:38 igenyx npm[16301]:   zigbee2mqtt:error 2019-3-18 15:22:38 Exiting...
mars 18 15:22:38 igenyx npm[16301]: npm ERR! code ELIFECYCLE
mars 18 15:22:38 igenyx npm[16301]: npm ERR! errno 1
mars 18 15:22:38 igenyx npm[16301]: npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
mars 18 15:22:38 igenyx npm[16301]: npm ERR! Exit status 1
mars 18 15:22:38 igenyx npm[16301]: npm ERR!
mars 18 15:22:38 igenyx npm[16301]: npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
mars 18 15:22:38 igenyx npm[16301]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
mars 18 15:22:38 igenyx npm[16301]: npm ERR! A complete log of this run can be found in:
mars 18 15:22:38 igenyx npm[16301]: npm ERR!     /home/zakabox/.npm/_logs/2019-03-18T14_22_38_068Z-debug.log
mars 18 15:22:38 igenyx systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
mars 18 15:22:38 igenyx systemd[1]: zigbee2mqtt.service: Unit entered failed state.
mars 18 15:22:38 igenyx systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
mars 18 15:22:38 igenyx systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
mars 18 15:22:38 igenyx systemd[1]: Stopped zigbee2mqtt.
mars 18 15:22:38 igenyx systemd[1]: Started zigbee2mqtt.
mars 18 15:22:39 igenyx npm[16464]: > zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
mars 18 15:22:39 igenyx npm[16464]: > node index.js
mars 18 15:22:40 igenyx npm[16464]:   zigbee2mqtt:info 2019-3-18 15:22:40 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-18.15-22-40'
mars 18 15:22:40 igenyx npm[16464]:   zigbee2mqtt:info 2019-3-18 15:22:40 Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
mars 18 15:22:40 igenyx npm[16464]:   zigbee2mqtt:info 2019-3-18 15:22:40 Starting zigbee-shepherd
mars 18 15:22:42 igenyx npm[16464]:   zigbee2mqtt:info 2019-3-18 15:22:42 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
mars 18 15:23:42 igenyx npm[16464]:   zigbee2mqtt:info 2019-3-18 15:23:42 Starting zigbee-shepherd
mars 18 15:23:43 igenyx npm[16464]:   zigbee2mqtt:error 2019-3-18 15:23:43 Error while starting zigbee-shepherd!
mars 18 15:23:43 igenyx npm[16464]:   zigbee2mqtt:error 2019-3-18 15:23:43 Press the reset button on the stick (the one closest to the USB) and start again
mars 18 15:23:43 igenyx npm[16464]:   zigbee2mqtt:error 2019-3-18 15:23:43 Failed to start
mars 18 15:23:43 igenyx npm[16464]:         {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
mars 18 15:23:43 igenyx npm[16464]:   zigbee2mqtt:error 2019-3-18 15:23:43 Exiting...
mars 18 15:23:44 igenyx npm[16464]: npm ERR! code ELIFECYCLE
mars 18 15:23:44 igenyx npm[16464]: npm ERR! errno 1
mars 18 15:23:44 igenyx npm[16464]: npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
[... cut ...]
mars 18 15:23:44 igenyx npm[16464]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
mars 18 15:23:44 igenyx npm[16464]: npm ERR! A complete log of this run can be found in:
mars 18 15:23:44 igenyx npm[16464]: npm ERR!     /home/zakabox/.npm/_logs/2019-03-18T14_23_44_029Z-debug.log
mars 18 15:23:44 igenyx systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
mars 18 15:23:44 igenyx systemd[1]: zigbee2mqtt.service: Unit entered failed state.
mars 18 15:23:44 igenyx systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

medalle23, 2019-03-18:
sudo nano /home/zakabox/.npm/_logs/2019-03-18T14_39_08_620Z-debug.log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin$
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.0-46-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.2
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.2.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.2.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

medalle23, 2019-03-18:
I do not know if there was a power cut yesterday.
More than I installed somewhere else.
If that's the case, then the problem comes from reboot (so you have to reboot manually, but I did it, but it does not work, hence the obligation to reboot the machine).

about reboot, i even use crontab, so that it can run the command alone (sudo systemctl restart zigbee2mqtt) after a few seconds. it does not seem to work.
I do not know if it's because of the password. a password (ubuntu) I am asked each first when I turn on the machine.

Koenkk, 2019-03-19:
What is the status of the led on the CC2531 when it doesn't start?

medalle23, 2019-04-30:
Hi @Koenkk ..
sorry for my late answer.
I was traveling, so I did not have the devices nearby.
But I had to install on a NUC and it works.
I also reinstalled on the z83, it works normally.

on the list of compatible devices.
which sensors (door, motion) can be installed outdoor.
I search in vain for outdoors sensors.
also i'm looking for occupancy detector (to turn on the light), presence detector, no motion sensor

closed 2019-07-06 by stale[bot]