Koenkk/zigbee2mqtt #1249: Custom device not being auto discovered after adding to homeassistant.js

indubwestep, 2019-03-13:
I have added a few custom devices to devices.js as well as fromZigbee.js and added the device to mapping in homeassistant.js however the device is not being autodiscovered. The homeassistant.js file is located in data/lib/extension. zigbee2mqtt is running in a docker container. Other devices are being auto discovered, just not the custom devices I have added. Is my path wrong? Does anything jump out? Homeassistant set to true in z2m config and discovery with birth/will configured in homeassistant config. Addis the devices to my homeassistant configuration.yaml manually works however I would prefer auto discovery.

Actions

Copy Copy selected linesUnselect

zigbee2mqtt:debug 3/13/2019, 7:22:10 PM Received zigbee message of type 'devChange' with data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:debug 3/13/2019, 7:22:49 PM Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"localTemp":2134}}' of device 'PCT503' (0x00606efffec0e26c)

zigbee2mqtt:info 3/13/2019, 7:22:49 PM MQTT publish: topic 'zigbee2mqtt/thermostat', payload '{"local_temperature":21.34,"linkquality":23}'

zigbee2mqtt:debug 3/13/2019, 7:22:49 PM Received zigbee message of type 'devChange' with data '{"cid":"hvacThermostat","data":{"localTemp":2134}}' of device 'PCT503' (0x00606efffec0e26c)

zigbee2mqtt:info 3/13/2019, 7:22:49 PM MQTT publish: topic 'zigbee2mqtt/thermostat', payload '{"local_temperature":21.34,"linkquality":23}'

zigbee2mqtt:debug 3/13/2019, 7:23:08 PM Received MQTT message on 'zigbee2mqtt/bedroom_switch/set' with data 'ON'

zigbee2mqtt:info 3/13/2019, 7:23:08 PM Zigbee publish to device '0x000d6f000d8155c1', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null

zigbee2mqtt:info 3/13/2019, 7:23:08 PM MQTT publish: topic 'zigbee2mqtt/bedroom_switch', payload '{"state":"ON","linkquality":21,"power":0}'

zigbee2mqtt:debug 3/13/2019, 7:23:08 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:debug 3/13/2019, 7:23:08 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d000132f66a"' of device 'ZHA-CCTLight-GLS0104' (0x00158d000132f66a)

zigbee2mqtt:debug 3/13/2019, 7:23:08 PM Received MQTT message on 'zigbee2mqtt/bedroom_switch/set' with data 'OFF'

zigbee2mqtt:info 3/13/2019, 7:23:09 PM Zigbee publish to device '0x000d6f000d8155c1', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null

zigbee2mqtt:info 3/13/2019, 7:23:09 PM MQTT publish: topic 'zigbee2mqtt/bedroom_switch', payload '{"state":"OFF","linkquality":21,"power":0}'

zigbee2mqtt:debug 3/13/2019, 7:23:09 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:debug 3/13/2019, 7:23:09 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:debug 3/13/2019, 7:23:09 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:error 3/13/2019, 7:23:29 PM Cannot get the Node Descriptor of the Device: 0xb0ce181403221210 (Error: Timed out after 10000 ms)

zigbee2mqtt:error 3/13/2019, 7:23:49 PM Cannot get the Node Descriptor of the Device: 0xb0ce181403221210 (Error: Timed out after 10000 ms)

zigbee2mqtt:debug 3/13/2019, 7:23:50 PM Received zigbee message of type 'attReport' with data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:info 3/13/2019, 7:23:50 PM MQTT publish: topic 'zigbee2mqtt/bedroom_switch', payload '{"state":"OFF","linkquality":13,"power":0}'

zigbee2mqtt:error 3/13/2019, 7:24:09 PM Cannot get the Node Descriptor of the Device: 0x00158d000132f66a (Error: Timed out after 10000 ms)

zigbee2mqtt:debug 3/13/2019, 7:25:30 PM Received zigbee message of type 'attReport' with data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:info 3/13/2019, 7:25:30 PM MQTT publish: topic 'zigbee2mqtt/bedroom_switch', payload '{"state":"OFF","linkquality":13,"power":0}'

zigbee2mqtt:debug 3/13/2019, 7:26:07 PM Saving state to file /app/data/state.json

zigbee2mqtt:debug 3/13/2019, 7:27:10 PM Received zigbee message of type 'attReport' with data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:info 3/13/2019, 7:27:10 PM MQTT publish: topic 'zigbee2mqtt/bedroom_switch', payload '{"state":"OFF","linkquality":13,"power":0}'

zigbee2mqtt:debug 3/13/2019, 7:27:50 PM Received zigbee message of type 'attReport' with data '{"cid":"hvacThermostat","data":{"localTemp":2132}}' of device 'PCT503' (0x00606efffec0e26c)

zigbee2mqtt:debug 3/13/2019, 7:27:50 PM Received zigbee message of type 'devChange' with data '{"cid":"hvacThermostat","data":{"localTemp":2132}}' of device 'PCT503' (0x00606efffec0e26c)

zigbee2mqtt:info 3/13/2019, 7:27:50 PM MQTT publish: topic 'zigbee2mqtt/thermostat', payload '{"local_temperature":21.32,"linkquality":34}'

zigbee2mqtt:info 3/13/2019, 7:27:50 PM MQTT publish: topic 'zigbee2mqtt/thermostat', payload '{"local_temperature":21.32,"linkquality":34}'

zigbee2mqtt:debug 3/13/2019, 7:28:49 PM Received zigbee message of type 'attReport' with data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d8155c1)

zigbee2mqtt:info 3/13/2019, 7:28:49 PM MQTT publish: topic 'zigbee2mqtt/bedroom_switch', payload '{"state":"OFF","linkquality":13,"power":0}'

[... 70 lines cut ...]
zigbee2mqtt:debug 3/13/2019, 7:34:04 PM Received zigbee message of type 'readRsp' with data '{"cid":"seMetering","data":{"summaFormatting":0,"demandFormatting":0,"meteringDeviceType":0,"instantaneousDemand":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d7f8717)

zigbee2mqtt:debug 3/13/2019, 7:34:04 PM Received zigbee message of type 'devInterview' with data '"0x000d6f000d7f8717"'

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"manufacturerName":"OWON","modelId":"SLC605-LEVELCTR","dateCode":"20170526","powerSource":1}}' of device 'SLC605-LEVELCTR' (0x000d6f000d7f8717)

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'devInterview' with data '"0x000d6f000d7f8717"'

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"identifyTime":0}}' of device 'SLC605-LEVELCTR' (0x000d6f000d7f8717)

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'devInterview' with data '"0x000d6f000d7f8717"'

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'devInterview' with data '"0x000d6f000d7f8717"'

zigbee2mqtt:info 3/13/2019, 7:34:05 PM Successfully configured 0x000d6f000d7f8717 (0x000d6f000d7f8717)

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'devInterview' with data '"0x000d6f000d7f8717"'

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'devIncoming' with data '"0x000d6f000d7f8717"' of device 'SLC605-LEVELCTR' (0x000d6f000d7f8717)

zigbee2mqtt:info 3/13/2019, 7:34:05 PM Device incoming...

zigbee2mqtt:info 3/13/2019, 7:34:05 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'

zigbee2mqtt:debug 3/13/2019, 7:34:05 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'SLC605-LEVELCTR' (0x000d6f000d7f8717)

snjoetw, 2019-03-13:
You'll need to explicitly map the volume with homeassistant.js:

```
-v $(pwd)/data/lib/extension/homeassistant.js:/app/node_modules/lib/extension/homeassistant.js
```

see https://www.zigbee2mqtt.io/information/docker.html

indubwestep, 2019-03-14:
> You'll need to explicitly map the volume with homeassistant.js:
> 
> ```
> -v $(pwd)/data/lib/extension/homeassistant.js:/app/node_modules/lib/extension/homeassistant.js
> ```
> see https://www.zigbee2mqtt.io/information/docker.html

Thank you for your reply. That didn't seem to do the trick. Here is what my docker-compose.yaml looks like:

  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev # x86_64/amd64
    restart: always
    volumes:
      - /opt/zigbee2mqtt/data:/app/data
      - /opt/zigbee2mqtt/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters
      - /opt/zigbee2mqtt/data/lib/extension/homeassistant.js:/app/node_modules/lib/extension/homeassistant.js
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0:rwm
    restart: always
    depends_on:
      mosquitto:
        condition: service_started

Koenkk, 2019-03-14:
Should be:

```yaml
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev # x86_64/amd64
    restart: always
    volumes:
      - /opt/zigbee2mqtt/data:/app/data
      - /opt/zigbee2mqtt/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters
      - /opt/zigbee2mqtt/data/lib/extension/homeassistant.js:/app/lib/extension/homeassistant.js
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/ttyACM0:/dev/ttyACM0:rwm
    restart: always
    depends_on:
      mosquitto:
        condition: service_started
```

indubwestep, 2019-03-14:
That did it! Thank you very much.

closed 2019-03-14 by indubwestep