Koenkk/zigbee2mqtt #1251: Adjusting Hue Brightness/Temperature Generates Excessive Traceback Errors in Home Assistant

RaptahJezus, 2019-03-13:
Greetings,

First off, great work on this project. Been slowly migrating all my Xiaomi devices over without a hitch, and loving how everything is going so far. Currently though, I'm running into an issue with my Hue's.

I've noticed that when adjusting my Hue bulb's brightness or temperature using Home Assistant's sliders or direct service calls, HA's log gets hit with tons of traceback errors. I don't see this behavior when toggling the light on or off.

Here's HA's error:
`2019-03-13 19:17:55 ERROR (MainThread) [homeassistant.components.mqtt.light.schema_json] Exception in state_received when handling msg on 'zigbee2mqtt/living_room_foyer_light': '{"brightness":70,"linkquality":26}'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/homeassistant/components/mqtt/light/schema_json.py", line 208, in state_received
    if values['state'] == 'ON':
KeyError: 'state'`

I did some digging, and found that Zigbee2MQTT was sending multiple messages on temperature/brightness state changes (which I believe is expected behavior), but is only including {"state":"ON"} or {"state":"OFF"} in the first one. Here's an example of the MQTT messages during a brightness adjustment:

`{"state":"ON","brightness":70}
{"brightness":70,"linkquality":39}
{"brightness":70}
{"brightness":70}
{"brightness":70,"linkquality":39}
{"state":"ON","linkquality":28}
{"brightness":70,"linkquality":26}`

The fact that state:ON or state:OFF is only included in some of these messages seems to make Home Assistant very upset, as I believe it's expecting state data for every message sent.

While not a breaking issue, this generates quite a bit of noise in the log files.

Thanks for your time!

Koenkk, 2019-03-14:
Are you running zigbee2mqtt 1.2.1?

RaptahJezus, 2019-03-14:
That's correct. 

`npm start`
`> zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt`

Koenkk, 2019-03-14:
Could you provide the debug log of these messages without `state`? (`log_level: debug`, http://www.zigbee2mqtt.io/configuration/configuration.html)

RaptahJezus, 2019-03-14:
Certainly. Here's the output when I move the brightness slider in HASS:

>   zigbee2mqtt:debug 3/14/2019, 6:02:17 PM Received MQTT message on 'zigbee2mqtt/hallway_light/set' with data '{"state": "ON", "brightness": 66}'
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM Zigbee publish to device '0x00178801028a6917', genLevelCtrl - moveToLevelWithOnOff - {"level":66,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM MQTT publish: topic 'zigbee2mqtt/hallway_light', payload '{"state":"ON","brightness":66}'
  zigbee2mqtt:debug 3/14/2019, 6:02:17 PM Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"currentLevel":66,"remainingTime":0}}' of device 'LWB014' (0x00178801028a6917)
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM MQTT publish: topic 'zigbee2mqtt/hallway_light', payload '{"brightness":66,"linkquality":18}'
  zigbee2mqtt:debug 3/14/2019, 6:02:17 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":66}}' of device 'LWB014' (0x00178801028a6917)
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM MQTT publish: topic 'zigbee2mqtt/hallway_light', payload '{"brightness":66}'
  zigbee2mqtt:debug 3/14/2019, 6:02:17 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":66}}' of device 'LWB014' (0x00178801028a6917)
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM MQTT publish: topic 'zigbee2mqtt/hallway_light', payload '{"brightness":66}'
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM Zigbee publish to device '0x00178801028a6917', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:debug 3/14/2019, 6:02:17 PM Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"currentLevel":66}}' of device 'LWB014' (0x00178801028a6917)
  zigbee2mqtt:info 3/14/2019, 6:02:17 PM MQTT publish: topic 'zigbee2mqtt/hallway_light', payload '{"brightness":66,"linkquality":18}'

Koenkk, 2019-03-15:
Can you post your `configuration.yaml`?

RaptahJezus, 2019-03-15:
Sure. I don't use MQTT auto-discovery, and my configuration files are split up. Here's `hallway_light.yaml`, which contains all configuration for the bulb in question:

```
homeassistant:
  customize:
    light.hallway_light:
      friendly_name: Hallway Light
    sensor.hallway_light_linkquality:
      friendly_name: Hallway Light Link Quality
      

light:
  - platform: "mqtt"
    name: hallway_light
    state_topic: "zigbee2mqtt/hallway_light"
    availability_topic: "zigbee2mqtt/bridge/state"
    brightness: true
    schema: "json"
    command_topic: "zigbee2mqtt/hallway_light/set"

sensor:
  - platform: "mqtt"
    name: hallway_light_linkquality
    state_topic: "zigbee2mqtt/hallway_light"
    availability_topic: "zigbee2mqtt/bridge/state"
    unit_of_measurement: "-"
    value_template: "{{ value_json.linkquality }}"
```

Looking at Home Assistant's code, it's requiring a the state value to exist for every message from MQTT lights. I don't see anything like this for other MQTT sensors. Here's the relevant part (line 208) of `/usr/local/lib/python3.6/dist-packages/homeassistant/components/mqtt/light/schema_json.py`:

```
    async def _subscribe_topics(self):
        """(Re)Subscribe to topics."""
        last_state = await self.async_get_last_state()

        @callback
        def state_received(topic, payload, qos):
            """Handle new MQTT messages."""
            values = json.loads(payload)

            if values['state'] == 'ON':
                self._state = True
            elif values['state'] == 'OFF':
                self._state = False
```

I just updated to Home Assistant 0.89.2, and that's still occurring.

Koenkk, 2019-03-15:
Sorry, i meant your zigbee2mqtt configuration.yaml

RaptahJezus, 2019-03-15:
Ah sorry about that. Here you go:

```
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://<redacted>'
  user: <redacted>
  password: <redacted>
serial:
  port: /dev/ttyACM1
advanced:
  cache_state: false
  availability_timeout: 30
devices:
  '0x00158d0001e56217':
    friendly_name: lb_motion_03
    retain: false
```
Then the device list continues from there.

By the way, modifying `schema_json.py` to
```
    async def _subscribe_topics(self):
        """(Re)Subscribe to topics."""
        last_state = await self.async_get_last_state()

        @callback
        def state_received(topic, payload, qos):
            """Handle new MQTT messages."""
            values = json.loads(payload)

            if 'state' in values:
                if values['state'] == 'ON':
                    self._state = True
                elif values['state'] == 'OFF':
                    self._state = False
```

Fixes the traceback errors, but will get blown out by Home Assistant during every update.

Koenkk, 2019-03-15:
You should set `cache_state: true` (otherwise the cached values won't be send leading into incomplete payloads).

RaptahJezus, 2019-03-15:
Well that fixed it, definitely feel quite foolish now.

Thanks for all your help!

closed 2019-03-15 by RaptahJezus