Koenkk/zigbee2mqtt #1253: [question] two way switching with only zigbee2mqtt

dh-harald, 2019-03-14:
Hi,

I've have a two way switch in my corridor, eg I've 1 bulb and 2 switches...
I've implemented this on that way, that I've 1 wall switch (QBKG11LM) and 1 wireless remote (WXKG03LM), and I made an automatization, that If I press the remote, it toggles the wall switch... It's working, but I've problems with it:
1. slow... sometimes it takes seconds between the click and the toggle
2. it stops working when the HA dies, or HA thinks the wall switch is offline (another problem)

So can I handle somehow the switch itself in zigbee2mqtt?
I'm thinking about on that way, what, I'm groupping the switch and the remote inside zigbee2mqtt?
Any other ideas?

nldroid, 2019-03-14:
I know that you can bind a switch to the light itself. Maybe that helps:

https://www.zigbee2mqtt.io/information/binding.html

dh-harald, 2019-03-15:
@nldroid thanks for the tip, but unfortunately it's not working... I could bind the wireless switch to the wall switch, but it's not controlling it... Same result, if I put them into the same group :(

nldroid, 2019-03-15:
I would have tried binding both of them to the light..

Koenkk, 2019-03-15:
1. Can you find where the slowness comes from? I don't expect it to come from HA.

FYI: Implementing an automation/rule engine is currently not within the scope of zigbee2mqtt.

dh-harald, 2019-03-15:
> I would have tried binding both of them to the light..

My light is not a smart one... I've smart switches only :)

closed 2019-03-18 by dh-harald