Koenkk/zigbee2mqtt #1254: Crashing CC2530 / Zigbee2MQTT when starting

blacky1985, 2019-03-14:
i installed the zigbee2mqtt and i am using a cc2530 as coordinator....but when i am starting it, i am getting a error.

attached the log file...hoping somebody can help me :-) 

`
> zigbee2mqtt@1.2.1 start /root/xiaomi-zb2mqtt
> node index.js

  zigbee2mqtt:info 2019-3-14 19:43:22 Logging to directory: '/root/xiaomi-zb2mqtt/data/log/2019-03-14.19-43-22'
  zigbee2mqtt:debug 2019-3-14 19:43:22 Removing old log directory '/root/xiaomi-zb2mqtt/data/log/2019-01-07.08-20-34'
  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:debug 2019-3-14 19:43:23 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/root/xiaomi-zb2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-3-14 19:43:23 Loaded state from file /root/xiaomi-zb2mqtt/data/state.json
  zigbee2mqtt:debug 2019-3-14 19:43:23 Saving state to file /root/xiaomi-zb2mqtt/data/state.json
  zigbee2mqtt:info 2019-3-14 19:43:23 Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
  zigbee2mqtt:info 2019-3-14 19:43:23 Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyUSB3 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +9ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyUSB3 +16ms
  cc-znp The serialport /dev/ttyUSB3 is opened. +29ms
  serialport/stream _read reading +4ms
  serialport/binding-abstract read +27ms
  serialport/bindings/unixRead Starting read +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +0ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +0ms
  serialport/stream _write 8 bytes of data +33ms
  serialport/binding-abstract write 8 bytes +33ms
  serialport/bindings/unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +39ms
  serialport/bindings/poller Polling for "readable" +48ms
  serialport/bindings/unixWrite write returned null 8 +7ms
  serialport/bindings/unixWrite wrote 8 bytes +1ms
  serialport/bindings/unixWrite Finished writing 8 bytes +0ms
  serialport/stream binding.write write finished +13ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +0ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +2s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/root/xiaomi-zb2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)
    at Object.onceWrapper (events.js:277:13)
    at CcZnp.emit (events.js:189:13)
    at Timeout.<anonymous> (/root/xiaomi-zb2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10) +0ms
  zigbee2mqtt:info 2019-3-14 19:43:24 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  serialport/stream #flush +2s
  serialport/binding-abstract flush +2s
  serialport/stream binding.flush finished +2ms
  serialport/stream #close +1ms
  serialport/binding-abstract close +3ms
  serialport/bindings/poller Stopping poller +2s
  serialport/bindings/poller Destroying poller +1ms
  serialport/stream binding.read error { Error: Canceled
    at Poller.emitCanceled (/root/xiaomi-zb2mqtt/node_modules/@serialport/bindings/lib/poller.js:101:17)
    at Poller.stop (/root/xiaomi-zb2mqtt/node_modules/@serialport/bindings/lib/poller.js:91:10)
    at close.then (/root/xiaomi-zb2mqtt/node_modules/@serialport/bindings/lib/linux.js:48:19) canceled: true } +4ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/stream binding.close finished +31s
  cc-znp The serialport /dev/ttyUSB3 is closed. +32s
  zigbee2mqtt:info 2019-3-14 19:44:25 Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +1m
  cc-znp Initialize, opening serial port +30s
  serialport/stream opening path: /dev/ttyUSB3 +30s
  serialport/binding-abstract open +1m
  serialport/stream _read queueing _read for after open +1ms
  serialport/bindings/poller Creating poller +1m
  serialport/stream opened path: /dev/ttyUSB3 +13ms
  cc-znp The serialport /dev/ttyUSB3 is opened. +16ms
  serialport/stream _read reading +2ms
  serialport/binding-abstract read +17ms
  serialport/bindings/unixRead Starting read +1m
  zigbee-shepherd:request REQ --> SYS:osalNvRead +1m
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +1m
  serialport/stream _write 8 bytes of data +8ms
  serialport/binding-abstract write 8 bytes +7ms
  serialport/bindings/unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +1m
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +10ms
  serialport/bindings/poller Polling for "readable" +14ms
  serialport/bindings/unixWrite write returned null 8 +2ms
  serialport/bindings/unixWrite wrote 8 bytes +1ms
  serialport/bindings/unixWrite Finished writing 8 bytes +0ms
  serialport/stream binding.write write finished +6ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +1m
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +2s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/root/xiaomi-zb2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)
    at Object.onceWrapper (events.js:277:13)
    at CcZnp.emit (events.js:189:13)
    at Timeout.<anonymous> (/root/xiaomi-zb2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10) +1m
  zigbee2mqtt:error 2019-3-14 19:44:26 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2019-3-14 19:44:26 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2019-3-14 19:44:26 Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/root/xiaomi-zb2mqtt/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:277:13)\n    at CcZnp.emit (events.js:189:13)\n    at Timeout.<anonymous> (/root/xiaomi-zb2mqtt/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 2019-3-14 19:44:26 Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-14T18_44_57_145Z-debug.log

`

nldroid, 2019-03-14:
What version of npm are you using? Upgrade to the latest and rebuild modules again.

blacky1985, 2019-03-15:
using npm@6.4.1
using node@v10.15.3

Have rebuild modules again but nothing changed...

Koenkk, 2019-03-15:
The problem is the communication with the CC2530 (probably wiring). I don't have experience on connecting this, but plenty of people do (search in the github issues for CC2530).

An alternative would be to use the CC2531 as a coordinator and the CC2530 as router. As the CC2531 has an USB interface, it's much more easy to connect. This is the setup I'm also using (and recommending)

closed 2019-05-21 by stale[bot]