Koenkk/zigbee2mqtt #1255: Domoticz error after update

salopette, 2019-03-14:
Hello, after the last update is something broken?

2019-03-14 21:04:23.908 (Zigbee Stick) Zigbee2mqtt bridge is online
2019-03-14 21:04:23.908 (Zigbee Stick) Joining new devices is enabled on the zigbee bridge
2019-03-14 21:04:23.960 (Zigbee Stick) Received available devices list from bridge
2019-03-14 21:04:23.960 Error: (Zigbee Stick) 'onMessage' failed 'KeyError'.
2019-03-14 21:04:23.960 Error: (Zigbee Stick) ----> Line 193 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2019-03-14 21:04:23.960 Error: (Zigbee Stick) ----> Line 106 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMessage
2019-03-14 21:04:23.960 Error: (Zigbee Stick) ----> Line 167 in '/home/pi/domoticz/plugins/zigbee2mqtt/mqtt.py', function onMessage
2019-03-14 21:04:23.960 Error: (Zigbee Stick) ----> Line 146 in '/home/pi/domoticz/plugins/zigbee2mqtt/plugin.py', function onMQTTPublish
2019-03-14 21:04:23.960 Error: (Zigbee Stick) ----> Line 47 in '/home/pi/domoticz/plugins/zigbee2mqtt/device_storage.py', function update
2019-03-14 21:04:24.010 (Zigbee Stick) Joining new devices is enabled on the zigbee bridge

Koenkk, 2019-03-15:
@stas-demydiuk can you help?

HJKLMN, 2019-03-15:
Experiencing the same... How to revert to an older version? (coud restore a backup though)

salopette, 2019-03-16:
@stas-demydiuk THX for Repair!!!

mikkel75, 2019-11-20:
I have the same problem now on latest sources... @salopette what fixed it for you?

salopette, 2019-11-20:
hi, I do not use it anymore, I switched to Deconz!

gizmocuz, 2023-11-01:
@Koenkk , I think we can close this issue. If there is still a issue its better to open a issue on the Domoticz git repo