Koenkk/zigbee2mqtt #1258: Unsupported device: innr RS120

hawaltie, 2019-03-15:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Koenkk, 2019-03-15:
Please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

hawaltie, 2019-03-15:
I tried to add the Innr RS 122 device to the custom devices.js, which seems almost similiar to the Innr RS 125, using the following config.

`
    {
        zigbeeModel: ['RS 122'],
        model: 'RS 122',
        vendor: 'Innr',
        description: 'GU10 Spot',
        extend: generic.light_onoff_brightness,
    }
`
The RS122 is recognised, but shortly after that, the process crashes and restarts. I am using the Hass.io add-on with the zigbee_shepherd_devices option to use the custom devices.js perhaps that has something to do with it.

The device was not visible in homeassistant itself, probably because I didn't adjust the homeassistant.js

Koenkk, 2019-03-15:
Can you provide the log of the crash?

hawaltie, 2019-03-15:
I'm sorry, I didn't capture the log, and the log files do not contain this info. I just reverted everything. I'll try to reproduce the log later.

This is the message that proofs that the RS122 is recognised:

0x00158d0000cf0278 (0x00158d0000cf0278): RS 122 - Innr GU10 Spot (Router)

hawaltie, 2019-03-18:
I found this error log:

> 
npm
 ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
 
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-03-18T22_10_46_307Z-debug.log
2019-03-18T23:10:51: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-03-18T23:10:51: PM2 log: App [npm:0] starting in -fork mode-
2019-03-18T23:10:51: PM2 log: App [npm:0] online
> 

I think there will be morge info in /root/.npm/_logs/2019-03-18T22_10_46_307Z-debug.log but I don't know how to get in case of using hass.io. Can you help

hawaltie, 2019-03-18:
Already figured out how to get there. This is the total log:

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node',
1 verbose cli   '/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js',
1 verbose cli   'start' ]
2 info using npm@6.4.1
3 info using node@v10.14.2
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/zigbee2mqtt-1.2.1/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /zigbee2mqtt-1.2.1
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:182:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:182:13)
13 verbose stack     at maybeClose (internal/child_process.js:962:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /zigbee2mqtt-1.2.1
16 verbose Linux 4.15.0-46-generic
17 verbose argv "/usr/bin/node" "/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js" "start"
18 verbose node v10.14.2    
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.2.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.2.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

Koenkk, 2019-03-19:
Should be supported in the dev branch now, can you check? http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

hawaltie, 2019-03-19:
Thanx. I'll try it later this week.

hawaltie, 2019-03-21:
Tried it yesterday. Used the edge version of the hassio addon. I had lot of troubles (also caused by myself) moving from the stable to the edge version, which resulted in the re-pairing of all my devices(!). But for now zigbee2mqtt edge is up and running.
For some reason the inner rs 122 will not pair. When I try to pair it, zigbee2mqtt reports an incoming device but then nothing happens. I will try to move the bulb closer tot the zigbee transmitter. Perhaps that helps.
Moving to the edge environment has one positive side effect. I was finally able to connect a hue bulb to the system.

One question: is the rs 122 added to the edge version? If I understand correctly, it is.

hawaltie, 2019-03-21:
The innr RS 122 has been paired, but it is not recognised:
`
Device with modelID 'RS 122\00\00\00\00\00\00\00\00\00\00\00\00\00\00' is not supported.
`

 This has probably something to do with [https://github.com/danielwelch/hassio-zigbee2mqtt/issues/136](url) (Zigbee2mqtt edge not updated).

Koenkk, 2019-03-21:
Just fixed that issue, new edge is available now.

hawaltie, 2019-03-21:
Tried it. The RS 122 is recognised:
`
zigbee2mqtt:info 3/21/2019, 8:39:43 PM MQTT publish: topic 'homeassistant/light/0x00158d0000cf0278/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0x00158d0000cf0278/set","state_topic":"zigbee2mqtt/0x00158d0000cf0278","name":"0x00158d0000cf0278_light","unique_id":"0x00158d0000cf0278_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0000cf0278","name":"0x00158d0000cf0278","sw_version":"Zigbee2mqtt 1.2.1","model":"GU10 Spot (RS 122)","manufacturer":"Innr"},"availability_topic":"zigbee2mqtt/bridge/state"}'
`
But when sending an on or off command an error is returned:
`
zigbee2mqtt:error 3/21/2019, 8:40:52 PM Zigbee publish to device '0x00158d0000cf0278', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
`

Koenkk, 2019-03-21:
Can you re-pair the device?

hawaltie, 2019-03-22:
It works! The Zigbee range of the bulb is small, so now waitig for the stuff to make a zigbee router

Koenkk, 2019-03-22:
Good!

closed 2019-03-22 by Koenkk