Koenkk/zigbee2mqtt #1259: CC2530 + CC2591 to Raspberry Pi 2 Model B not working

Wamor, 2019-03-15:
Hello,

I have flashed a CC2530 + CC2591 board with the coordinator-firmware.
Than I have connected a CC2530 + CC2591 board to a Raspberry Pi 2 Model B.
Connected it as follows:
CC2530+CC2591 board pin <> Raspberry Pi board pin
P0_2 <> pin 8
P0_3 <> pin 10
GND <> pin 6
GND <> pin 9
VCC <> pin 1
VCC <> pin 17

I have changed the port in the configuration.yaml-file to: ttyAMA0
In the zigbee2mqtt I see that an error the ZigBee-shepherd cannot start.
Reversing P0_2 and P0_3 pins gives no solution.
Is there a possibility to check if the communication between the Raspberry Pi and CC2530 board is ok?
I read on this site that you also should short some pins of the CC2530 board to GND. Is this still needed?
Do I need to set some options on the Raspberry Pi board to enable the on board serial port?
Should set some extra options in the configuration.yaml-file like: rtscts: false?
I have also connected a CC2531 board through USB to the Raspberry Pi board and this is working ok.
Any help would be greatly appreciated!

Regards,

Wamor

HJKLMN, 2019-03-15:
Will check pin-layout later, I have same setup. My configuration.yaml:

serial:
  port: /dev/ttyAMA0
advanced:
  baudrate: 115200
  rtscts: false

Wamor, 2019-03-15:
Whow, thanks for the quick reply.
I have added you configuration.yaml suggestions and checked my pin configuration.
The are the same as yours. Still no luck in making it work.
From where did you got you coordinator-firmware? I downloaded mine from this web-site.

Regards,

Wamor

HJKLMN, 2019-03-15:
You did not forget the indents? See now I forgot them,
serial:
          port: /dev/ttyAMA0
advanced:
          baudrate: 115200
          rtscts: false

Wamor, 2019-03-16:
Hi HJKLMN,

I did not forget the indents.
I re-flashed the CC2530 board with the coordinator firmware from this website but still no succes.
Is there a way to check if the serial-port is working correctly? Could I communicate to the CC2530 module directly through a terminal program to check if the programming went ok.?

Regards,

Wamor

Wamor, 2019-03-16:
Hello,

I have solved my problem.
I ran raspi-config and disable "login shell to be accessible over serial" and enable "serial port hardware".
See: https://www.instructables.com/id/Read-and-write-from-serial-port-with-Raspberry-Pi/
Now the CC2530+CC2591 board is detecting my ZigBee-devices.
@HJKLMN for helping me.

Regards,

Wamor

closed 2019-03-16 by Wamor