Koenkk/zigbee2mqtt #1260: Apply transition time to simple on/off state change

uncle-fed, 2019-03-15:
It looks like at present the Hue bulbs transition time is only applied when parameters such as brightness or color xy are changed. It is not possible to turn off Hue bulb with non-default transition time, like it is possible with native hue bridge.

For instance, sending {"state":"OFF", "transition":0}. has no effect on how long it takes to turn a bulb off.

It should be possible to:

- use "transition" time when turning lights OFF (it's most likely transitioning to brightness = 0, so it should be made possible to control the speed of that transition, just like with the native hue bridge)

- use "transition" time when turning lights ON without specifying extra light parameters, such as brightness or color (the default settings for brightness should be used)

- control "transition" time in 100s of ms instead of s, i.e. same way it is done in the native Hue bridge, where value of 4 means 0.4s (and it is the default value for turning lights on and off, BTW).

I am trying to get rid of the hue bridge but things like that making it harder to give it up.

Thank you for listening.

uncle-fed, 2019-03-15:
Thanks for taking interest in this! 
I wasn't aware that fractions are supported, should be no problem then with the 3rd point.

uncle-fed, 2019-03-19:
Another question / possible request in line with the above (i.e. going away from Hue bridge). Philips Hue API supports setting brightness / hue / saturation / ct / xy via a delta value, i.e. relative +/- value as in bri_inc / hue_inc, etc.
Do you think this would be possible with zigbee2mqtt ? Or is it already possible and I did not find this information?

Koenkk, 2019-03-20:
This is not possible yet.

Koenkk, 2019-03-22:
Transition for ON and OFF should be implemented now in the dev branch, can you check this?

uncle-fed, 2019-03-23:
I tried to check the dev branch using the following commands on my Rasberry Pi:

```
cd /opt/zigbee2mqtt
git checkout dev
git pull
systemctl restart zigbee2mqtt
```

After that I tried to send payload such as this but it had no effect (still looked like it takes ~3 seconds instead of 10):

```
mosquitto_pub -t 'zigbee2mqtt/HueBulb/set' -m '{"state":"ON","transition":10}'
mosquitto_pub -t 'zigbee2mqtt/HueBulb/set' -m '{"state":"OFF","transition":10}'
```

Koenkk, 2019-03-23:
After checking out dev, you need to do a `rm -rf node_modules && npm install`.

uncle-fed, 2019-03-23:
Thank you for the hint, I've updated the code as you suggested.
And than you once more because the code performs as intended now when running the above test commands!

Is this normal that a single mosquitto_pub command above results in the following zigbee2mqtt log entries:

```
5:24:17 PM Zigbee publish to device '0x0017880102ca5bc8', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":100} - {"manufSpec":0,"disDefaultRsp":0} - null
5:24:17 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":255,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
5:24:17 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":5,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
5:24:17 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":5,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
5:24:17 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":5,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
5:24:27 PM Zigbee publish to device '0x0017880102ca5bc8', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
5:24:27 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
5:24:27 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
5:24:27 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
```

I sort of understand why there is "brigtness: 5" and "brigtness: 254" there (to make proper transition from lowest value to maximum). But...

- why these messages repeat 3 times?
- why is there extra "brightness: 255" at the beginning and later it is only "254"?
- why does it start with 5 and not 1 or 2 (I found that brightness = 1 is the same as OFF, at least visually).

Koenkk, 2019-03-23:
Can you provide the `log_level: debug` log? (http://www.zigbee2mqtt.io/configuration/configuration.html)

uncle-fed, 2019-03-23:
```
Mar 23 19:21:46 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:46 PM Received MQTT message on 'zigbee2mqtt/HueBulb/set' with data '{"state":"ON","transition":10}'
Mar 23 19:21:46 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:46 PM Zigbee publish to device '0x0017880102ca5bc8', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":100} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 23 19:21:46 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:46 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":78,"brightness":255,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
Mar 23 19:21:47 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:47 PM Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"16384":254,"currentLevel":5,"remainingTime":98}}' of device 'LCT010' (0x0017880102ca5bc8)
Mar 23 19:21:47 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:47 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":81,"brightness":5,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
Mar 23 19:21:47 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:47 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":5}}' of device 'LCT010' (0x0017880102ca5bc8)
Mar 23 19:21:47 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:47 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":81,"brightness":5,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
Mar 23 19:21:47 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:47 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":5}}' of device 'LCT010' (0x0017880102ca5bc8)
Mar 23 19:21:47 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:47 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":81,"brightness":5,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
Mar 23 19:21:56 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:56 PM Zigbee publish to device '0x0017880102ca5bc8', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 23 19:21:56 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:56 PM Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'LCT010' (0x0017880102ca5bc8)
Mar 23 19:21:56 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:56 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
Mar 23 19:21:56 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:56 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'LCT010' (0x0017880102ca5bc8)
Mar 23 19:21:56 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:56 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
Mar 23 19:21:57 openHABianPi npm[1905]:   zigbee2mqtt:debug 3/23/2019, 7:21:57 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":254}}' of device 'LCT010' (0x0017880102ca5bc8)
Mar 23 19:21:57 openHABianPi npm[1905]:   zigbee2mqtt:info 3/23/2019, 7:21:57 PM MQTT publish: topic 'zigbee2mqtt/HueBulb', payload '{"state":"ON","linkquality":84,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
```

Koenkk, 2019-03-25:
- Latest dev should reduce the amount of mqtt messages.
- Zigbee2mqtt sets a brightness of 255 (max according to the ZCL specification), as the command succeeds, zigbee2mqtt publishes this (assumed) value, later the bulb self reports a message that the brightness is 254
- My guess is that the transition is already started once the bulb reports this value.

uncle-fed, 2019-03-30:
Looks good! Thanks a lot for taking interest!
I guess we can close this issue now.

Should I open another one for the 'brightness / hue / saturation delta value' ?

Koenkk, 2019-03-30:
Great, yes please!

closed 2019-03-30 by Koenkk