Koenkk/zigbee2mqtt #1262: IKEA remote control error sending command

bremy, 2019-03-16:
Hello. I was updating for the the last zigbee2mqtt on tuesday, updating for the last firmware version on my CC2431

Since I can see errors when sending on/off command to the remote device. Some are working but most of the time this is generating the following error

Any idea to fix that?

```
3/15/2019, 6:57:05 PM - error: Zigbee publish to device '0x000d6ffffe9c49ef', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
3/15/2019, 6:57:05 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffe9c49ef","type":"device","friendlyName":"0x000d6ffffe9c49ef"},"message":"on"}}'

```

Koenkk, 2019-03-16:
Can you try to re-pair the device?

bremy, 2019-03-16:
Since my morning reset, I can't see any error in the zigbee2mqtt log
I will keep you informed if ever that error appears again

The last action I made before reseting my CC2531 was to pair a Xiaomi RTCGQ11LM 2 days ago

Wamor, 2019-03-16:
Hello,

Just to let you know. I have a similar issue with the Ikea Control Outlet.
Today I upgraded my system from a CC2531 to a CC2530 + CC2591. I had to repair the Control Outlet.
Switching on and off the Control Outlet works ok but most of the time I got the below error:
`'Error: AF data request fails, status code: 233. MAC no ack.', 'type': 'zigbee_publish_error', 'meta': {'message': '{"state": "ON"}', 'entity': {'friendlyName': '0xd0cf5efffecb6ed5', 'type': 'device', 'ID': '0xd0cf5efffecb6ed5'}}`
If I can do some experiments to help solving this issue just let me know.

Regards,

Wamor

Koenkk, 2019-03-16:
@Wamor does it work when re-pairing it and keeping it very close to the CC2530+CC2591?

Wamor, 2019-03-16:
Will check tomorrow. CC2530+C2591 is located in my house and Control Outlet is in my Garage about 2 meters apart separated by 2 walls.

bremy, 2019-03-17:
all working fine again today
So probably the best procedure is to be sure to reset the CC2531 each time after paing a new device?

Wamor, 2019-03-17:
Hi,

Agreed, I had to disconnect the Raspberry Pi and CC2530 in order to keep it close to the Control Outlet.
Problems are gone now. Only strange thing I see is that after sending 1 command to a device that it will report 3 identical messages in return? (see below).
```
2019-03-17 20:31:48.858 (Home) onCommand: Off, level (0) Color: 
2019-03-17 20:31:48.858 (Home) Command "Off" from device "Home - Switch2" 
2019-03-17 20:31:48.859 (Home - Switch2) Updating device from 1:'On' to have values 0:'Off'. 
2019-03-17 20:31:48.870 (Home) MqttClient::Publish zigbee2mqtt/0xd0cf5efffecb6ed5/set ({"state": "OFF"}) 
2019-03-17 20:31:49.183 (Home) MQTT message: zigbee2mqtt/0xd0cf5efffecb6ed5 {'state': 'OFF', 'linkquality': 41} 
2019-03-17 20:31:49.183 (Home - Switch2 (Link Quality)) Updating device from 41:'41' to have values 41:'41'. 
2019-03-17 20:31:49.198 (Home) MQTT message: zigbee2mqtt/0xd0cf5efffecb6ed5 {'state': 'OFF', 'linkquality': 46} 
2019-03-17 20:31:49.198 (Home - Switch2 (Link Quality)) Updating device from 41:'41' to have values 46:'46'. 
2019-03-17 20:31:49.509 (Home) MQTT message: zigbee2mqtt/0xd0cf5efffecb6ed5 {'state': 'OFF', 'linkquality': 46} 
2019-03-17 20:31:49.509 (Home - Switch2 (Link Quality)) Updating device from 46:'46' to have values 46:'46'. 
```
Regards,

Wamor

bremy, 2019-03-25:
Still all working fine for me. All my paired Xiaomi, IKEA things are working perfectly

closed 2019-05-31 by stale[bot]