Koenkk/zigbee2mqtt #1265: npm ERR! zigbee2mqtt@1.2.1 start: node index.js Start Error

djuscha, 2019-03-16:
get after the installationof zigbee2mqtt not to run that I get as error message. What am I doing wrong
[code](https://hastebin.com/deyobitima.rb)
firmware  @[https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/bin/CC2531ZNP-Prod_20190223.zip](url)

Koenkk, 2019-03-16:
The syntax in your `configuration.yaml` is invalid, fix it by using e.g. http://www.yamllint.com/

djuscha, 2019-03-16:
that's how the configuration.yaml looks like and still Errors

`permit_join: true
# MQTT settings
mqtt:
# MQTT base topic for zigbee2mqtt MQTT messages
base_topic: zigbee2mqtt
# MQTT server URL
server: 'mqtt://localhost'
# MQTT server authentication, uncomment if required:
user: openhabian
password: openhabian
# Serial settings
serial:
# Location of CC2531 USB sniffer
port: /dev/ttyACM1
`

Koenkk, 2019-03-16:
you need to put spaces in the section between `mqtt` and `serial` example: http://www.zigbee2mqtt.io/configuration/configuration.html

djuscha, 2019-03-18:
have deleted everything and made new now it . 
now everything is good :)

closed 2019-03-18 by djuscha