Koenkk/zigbee2mqtt #1268: Cannot join Hue dimmer switch anymore

f-bader, 2019-03-16:
## What happened

 * I upgraded to the coordinator firmware to `20190223`  and zigbee2mqtt dev version 1.2.1 (commit #661f79c) from `20190109` and zigbee2mqtt dev version 1.2.1 (commit #3c4fa40)
 * After the upgrade I pressed the keys on my Philips Hue dimmer switch as advised
 * They worked until I tried the new "long press" method. The switch showed a red LED after the long press and afterwards no button press was send to the coordinator anymore
 * This happened to all three of my Philips Hue dimmer switches. I only focus on one of them in this issue.

## Affected device
Device ID: 0x001788011039b157
Type: 324131092621 - Philips Hue dimmer switch

## What I did after it stopped working
 * Removed device from zigbee2mqtt via MQTT message
 * Stopped zigbee2mqtt and removed the device from the database.db file
 * Turned on debug logging and restarted zigbee2mqtt
 * Hard reset Hue Dimmer (10 second pressing the setup button on the back)
 * Error messages related to the device:
  ```
  3/16/2019, 7:31:26 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)
  3/16/2019, 7:31:46 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)

  ```
 * The Hue dimmer does not react to any button presses at all (no LED indication)
 * Soft reset of the Hue dimmer (Press all buttons at the same time)
 * Error messages related to the device:
 ```
 3/16/2019, 7:36:57 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)
 3/16/2019, 8:14:19 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)
 3/16/2019, 8:52:32 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)
 ```
 * Let the network settle for a a few hours
 * Hard reset Hue Dimmer (10 second pressing the setup button on the back)
 * Error messages related to the device:
  ```
  3/16/2019, 11:23:29 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)
  3/16/2019, 11:23:49 PM - error: Cannot get the Node Descriptor of the Device: 0x001788011039b157 (Error: Timed out after 10000 ms)
  ```
 * No more log entries regarding the switch until I posted this issue...

The dimmer switch was always very close to the coordinator while trying to joing (~ 40cm)
Before I did this with debug log on and everything I already tried with another dimmer switch so the behaviour was reproducable.
One dimmer switch is still joined and does not work, So I can only try this once more in full detail.

## Currently joined devices
```
3/16/2019, 7:29:26 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #661f79c)
3/16/2019, 7:29:27 PM - info: Coordinator firmware version: '20190223'
3/16/2019, 7:29:27 PM - info: Currently 20 devices are joined:
3/16/2019, 7:29:27 PM - info: 0x8418260xxxxxxxxx (0x8418260xxxxxxxxx): AB3257001NJ - OSRAM Smart+ plug (Router)
3/16/2019, 7:29:27 PM - info: 0x00158d0xxxxxxxxx (0x00158d0xxxxxxxxx): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x00158d0xxxxxxxxx (0x00158d0xxxxxxxxx): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x8418260xxxxxxxxx (0x8418260xxxxxxxxx): AB3257001NJ - OSRAM Smart+ plug (Router)
3/16/2019, 7:29:27 PM - info: 0x8418260xxxxxxxxx (0x8418260xxxxxxxxx): AB3257001NJ - OSRAM Smart+ plug (Router)
3/16/2019, 7:29:27 PM - info: 0x7cb03eaxxxxxxxxx (0x7cb03eaxxxxxxxxx): AB3257001NJ - OSRAM Smart+ plug (Router)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 9290011370 - Philips Hue white A60 bulb E27 (Router)
3/16/2019, 7:29:27 PM - info: 0x00158d0xxxxxxxxx (0x00158d0xxxxxxxxx): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x8418260xxxxxxxxx (0x8418260xxxxxxxxx): AB3257001NJ - OSRAM Smart+ plug (Router)
3/16/2019, 7:29:27 PM - info: 0x000b57fxxxxxxxxx (0x000b57fxxxxxxxxx): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 9290012607 - Philips Hue motion sensor (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x00158d0xxxxxxxxx (0x00158d0xxxxxxxxx): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x00158d0xxxxxxxxx (0x00158d0xxxxxxxxx): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 324131092621 - Philips Hue dimmer switch (EndDevice)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 9290011370 - Philips Hue white A60 bulb E27 (Router)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 8718696548738 - Philips Hue white ambiance E26/E27 (Router)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 9290012573A - Philips Hue white and color ambiance E26/E27/E14 (Router)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 7299355PH - Philips Hue white and color ambiance LightStrip (Router)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 7299355PH - Philips Hue white and color ambiance LightStrip (Router)
3/16/2019, 7:29:27 PM - info: 0x0017880xxxxxxxxx (0x0017880xxxxxxxxx): 9290011370 - Philips Hue white A60 bulb E27 (Router)
3/16/2019, 7:29:27 PM - warn: `permit_join` set to  `true` in configuration.yaml.
3/16/2019, 7:29:27 PM - warn: Allowing new devices to join.
```

I'm currently out of ideas what I can do. Anybody got any tipps for me?

Complete log file including multiple network maps:
https://hastebin.com/ziqekasidu.makefile

kiwinol, 2019-03-16:
I am also experiencing issues with the latest firmware and latest stable docker version of zigbee2mqtt and my philips hue dimmer switches.  When I reset them they work for a while then stop working until I reset them.  It might be due to an accidently long press if its related to your issue.

f-bader, 2019-03-17:
I did a hard reset from the last dimmer switch which I not removed from the coordinator through MQTT. It still does not work after this but the debug log is a bit more talkative.

### Affected device

Device ID: 0x00178801105fc1f2
Type: 324131092621 - Philips Hue dimmer switch

### Debuglog

```
3/17/2019, 6:33:29 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00178801105fc1f2"' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:31 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Philips","modelId":"RWL021","powerSource":3}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:31 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"manufacturerName":"Philips","modelId":"RWL021","powerSource":3}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:31 PM - warn: No converter available for '324131092621' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"Philips","modelId":"RWL021","powerSource":3}}'
3/17/2019, 6:33:31 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
3/17/2019, 6:33:32 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:32 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:32 PM - warn: No converter available for '324131092621' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1}}'
3/17/2019, 6:33:32 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
3/17/2019, 6:33:33 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:33 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"dateCode":"20160302","swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:33 PM - warn: No converter available for '324131092621' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"dateCode":"20160302","swBuildId":"5.45.1.17846"}}'
3/17/2019, 6:33:33 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
3/17/2019, 6:33:33 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:33 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:34 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:34 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:35 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:35 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:36 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:36 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:36 PM - warn: No converter available for '324131092621' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}'
3/17/2019, 6:33:36 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
3/17/2019, 6:33:37 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:37 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:33:37 PM - warn: No converter available for '324131092621' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}'
3/17/2019, 6:33:37 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
3/17/2019, 6:33:37 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:33:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":27,"batteryPercentageRemaining":142}}' of device 'RWL021' (0x00178801105fc1f2)
[... cut ...]
3/17/2019, 6:33:40 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
3/17/2019, 6:33:40 PM - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:34:26 PM - debug: Saving state to file /app/data/state.json
3/17/2019, 6:39:26 PM - debug: Saving state to file /app/data/state.json
3/17/2019, 6:40:39 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00178801105fc1f2"' of device 'RWL021' (0x00178801105fc1f2)
```

f-bader, 2019-03-17:
Just updated my docker container to `zigbee2mqtt version 1.2.1 (commit #e6044da)` and the still joined hue dimmer started to work again:

Debug log
```
3/17/2019, 6:51:09 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00178801105fc1f2"' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:11 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"Philips","modelId":"RWL021","powerSource":3}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:12 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:12 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:12 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:13 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:13 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:14 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:14 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:15 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:16 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:17 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:17 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:18 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":26,"batteryPercentageRemaining":130}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:18 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00178801105fc1f2', payload '{"battery":65,"linkquality":107}'
3/17/2019, 6:51:18 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryVoltage":26,"batteryPercentageRemaining":130}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:18 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:18 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:19 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:19 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:20 PM - debug: Received zigbee message of type 'devInterview' with data '"0x00178801105fc1f2"'
3/17/2019, 6:51:20 PM - debug: Received zigbee message of type 'devIncoming' with data '"0x00178801105fc1f2"' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:51:20 PM - info: Device incoming...
3/17/2019, 6:51:20 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
3/17/2019, 6:51:20 PM - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:55:54 PM - debug: Configuring 0x00178801105fc1f2 (0x00178801105fc1f2) ...
3/17/2019, 6:56:44 PM - warn: Failed to configure 0x00178801105fc1f2 (0x00178801105fc1f2) ('undefined') (attempt #1)
3/17/2019, 6:57:34 PM - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x00178801105fc1f2"' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:57:34 PM - debug: Configuring 0x00178801105fc1f2 (0x00178801105fc1f2) ...
3/17/2019, 6:57:37 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"manufacturerName":"Philips"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:57:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"RWL021","dateCode":"20160302","powerSource":3,"swBuildId":"5.45.1.17846"}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:57:39 PM - info: Successfully configured 0x00178801105fc1f2 (0x00178801105fc1f2)
3/17/2019, 6:59:37 PM - debug: Received zigbee message of type 'cmdHueNotification' with data '{"cid":"manuSpecificPhilips","data":{"button":4,"unknown1":3145728,"type":0,"unknown2":0,"time":0}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:59:37 PM - debug: Received zigbee message of type 'cmdOffWithEffect' with data '{"cid":"genOnOff","data":{"effectid":0,"effectvariant":0}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:59:37 PM - debug: Received zigbee message of type 'cmdHueNotification' with data '{"cid":"manuSpecificPhilips","data":{"button":4,"unknown1":3145728,"type":2,"unknown2":0,"time":2}}' of device 'RWL021' (0x001788 [... cut ...]
3/17/2019, 6:59:43 PM - debug: Received zigbee message of type 'cmdHueNotification' with data '{"cid":"manuSpecificPhilips","data":{"button":1,"unknown1":3145728,"type":2,"unknown2":0,"time":3}}' of device 'RWL021' (0x00178801105fc1f2)
3/17/2019, 6:59:43 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00178801105fc1f2', payload '{"battery":65,"linkquality":73,"duration":0,"counter":1,"action":"on-press"}'
```

The other two I still cannot join at the moment. Maybe this has to do something with the device limit but I have many routers (12) in my network so it should not be an issue.

Bastiencc, 2019-03-19:
I installed the last version (dev branch version 1.2.1 commit #4cb0c0b), I got the same issue

Bastiencc, 2019-03-29:
I am also experiencing issues with some Philips Hue Dimmer Switch : 
```
Mar 29 15:05:58 raspberrypi npm[2417]:   zigbee2mqtt:warn 3/29/2019, 3:05:58 PM Failed to remove '0x001788010408cd64', trying force remove...
Mar 29 15:05:58 raspberrypi npm[2417]:   zigbee2mqtt:info 3/29/2019, 3:05:58 PM Force removed 0x001788010408cd64
Mar 29 15:05:58 raspberrypi npm[2417]:   zigbee2mqtt:info 3/29/2019, 3:05:58 PM Successfully removed 0x001788010408cd64
Mar 29 15:05:58 raspberrypi npm[2417]:   zigbee2mqtt:info 3/29/2019, 3:05:58 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x001788010408cd64"}'
Mar 29 15:06:10 raspberrypi npm[2417]:   zigbee2mqtt:info 3/29/2019, 3:06:10 PM New device 'undefined' with address 0x001788010408cd64 connected!
Mar 29 15:06:10 raspberrypi npm[2417]:   zigbee2mqtt:info 3/29/2019, 3:06:10 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x001788010408cd64","meta":{}}'
Mar 29 15:06:10 raspberrypi npm[2417]:   zigbee2mqtt:warn 3/29/2019, 3:06:10 PM Device with modelID 'undefined' is not supported.
Mar 29 15:06:10 raspberrypi npm[2417]:   zigbee2mqtt:warn 3/29/2019, 3:06:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

and : 
```
Mar 29 15:09:34 raspberrypi npm[3727]:   zigbee2mqtt:error 3/29/2019, 3:09:34 PM Cannot get the Node Descriptor of the Device: 0x001788010408cd64 (Error: Timed out after 10000 ms)
Mar 29 15:09:54 raspberrypi npm[3727]:   zigbee2mqtt:error 3/29/2019, 3:09:54 PM Cannot get the Node Descriptor of the Device: 0x001788010408cd64 (Error: Timed out after 10000 ms)
```

f-bader, 2019-04-21:
Today my second CC2531 came and I was able to sniff the data send and received from one of the Hue dimmers which has this problem.

[Here is the PCAP file](https://drive.google.com/file/d/1CYpnrD_pRQW266Kl1NI3YalIAKN8nlhf/view?usp=sharing)

@Koenkk if you could find the time I would greatly appricate any help with analyzing this problem.
As far as I can tell it seems alright, but those endless IEEE 802.15.4 "Data Request" packets seem to be a problem because there is no response to any of them

Koenkk, 2019-04-22:
@f-bader these data request packets are send to a different device (not the coordinator it seems). What kind of device is `0x8e13`?

f-bader, 2019-04-22:
@Koenkk it's an OSRAM Smart+ plug (0x7cb03eaa00a71253 AB3257001NJ) - Maybe it's trying to use this device as router?

f-bader, 2019-04-22:
I rejoind the Smart plug and after some time tried to rejoin the Hue dimmer.
Sadly still does not work. I check the message flow and it seems a successfull rejoin. Afterwards the Cluster bind are successfull as well but then:

Package_104: Source Route Failure
Package_105: Non-tree Link Failure

This is the filter I used:
 `zbee_nwk.src == 0x0000 or zbee_nwk.dst == 0x75d7 or zbee_nwk.dst == 0x0000 or zbee_nwk.src == 0x75d7`

File: https://drive.google.com/open?id=1oS_MiKPUnRfWB9nNjdmQvktMy9fJDb_9

Koenkk, 2019-04-23:
I'm not sure why this is happening or how to debug it further.

Something you can try, in case you have a spare CC2531, is zigbee 3.0 (see #1445).

f-bader, 2019-05-27:
Latest update: After installing firmware version `CC2531_20190425` and zigbee v 1.4.0 my hue dimmer are working again. Whatever changed fixed it. Thank you!

closed 2019-05-27 by f-bader