Koenkk/zigbee2mqtt #1269: Latency

nono313, 2019-03-16:
I own 4 Philips Hue light bulbs and 2 Philips Hue Dimmer Switch. I used to have all of them connected to the Hue hub and controlled through Home Assistant (running in Docker on a Raspberry Pi 3).

A few weeks ago, I have bought a CC2531 Zigbee sniffer dongle and installed Zigbee2MQTT (and mosquitto, both on their own Docker containers) on my Raspberry Pi 3.
I have connected 2 light of my 4 light bulbs and one of my dimmer switches to Zigbee2MQTT. I have created a small Node Red flow to connect the on / off buttons to one of the light bulbs.

When I press a button on the dimmer switch, it feels like I have a small latency (about 1s or slightly less) between the press and the light changing.
When I press the color picker of the light bulb (from HA), I also feel like there is a small delay / latency.

In some cases, I ended up with several seconds of delay / latency. In some cases, a couple of commands had been queued and executed at once several seconds after the first command.

Do you think this delay / latency is expected ? I could see that the flow Zigbee2MQTT -> Mosquitto -> Node Red -> Mosquitto -> Zigbee2MQTT might cause some latency but I wouldn’t expect it to be visible for the user.
Should I be able to get the same response time and reliability as with the Philips Hue hub ?
Is anybody else having latency issues when using Zigbee2MQTT or MQTT in general ?

I see that a latency issue was raised a couple of months ago but was not fixed (#687). As I only installed all a couple of days ago I cannot say if it is linked to the upgrade of the software like in the linked issue.

pixeldoc2000, 2019-03-17:
Some Latency is normal because of the hole chain around zigbee2mqtt. Usualy the remote does not talk directly to the bulbs so every command has to pass a lot of layers.

What Hardware are you using to run zigbee2mqtt, mqtt etc. ?

Ìt may be faster with the HUE Bridge because the Remotes might talk directly talk to the bulbs via broadcast, cluster, groups etc.

It highly depends on the zigbee implementation and other factors. Also if you use a very crowed Zigbee Channel (WIFI 2.4 GHz anyone?) i noticed higher delays.

nono313, 2019-03-17:
I am running Home Assistant, Node Red, Msquitto and Zigbee2mqtt each in a
docker container on the same Raspberry Pi 3 with Raspbian lite (running
from an external Ssd drive).

I'm living in a big building. I might have to look at zigbee channel and
wifi interference to try and mitigate the latency.

pixeldoc2000, 2019-03-19:
Did you check the system load?

Any messages in HA oder zigbee2mqtt log? Maybe enable DEBUG log level?

Post you zigbee2mqtt config.

nono313, 2019-03-19:
System load seems okay. I have all 4 CPUs running below 20% even when sending commands from the HA interface.

Here is an example of logs in debug mode where I toggle a couple of times one light bulb using the HA interface (through MQTT messages sent from HA) : [https://hastebin.com/umejibezam.makefile](https://hastebin.com/umejibezam.makefile)

Here is my zigbee2mqtt config file : 
`homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
advanced:
  channel: 15
devices:
  '0x00178801040a552e':
    friendly_name: philips_hue_dimmer_livingroom1
    retain: false
  '0x0017880104e3fb4c':
    friendly_name: philips_hue_light_livingroom1
    retain: false
  '0x0017880104d1ffa8':
    friendly_name: philips_hue_light_bedroom
    retain: false
`

cvladan, 2019-03-24:
I'm also experiencing high latency, but in my case, toggling switches from Hass UI is very responsive. But when I trigger automation with some trigger (both trigger and action via mqtt and zigbee2mqtt) it is sometimes really slow, even 2 secs. That makes it unusable for lights and similar applications.

pixeldoc2000, 2019-03-26:
At least you can monitor the mqtt traffic if you subscribe to the specific topic with mqtt.fx client for example.

Bastiencc, 2019-03-27:
Same issue here, from HA toggle card everything is very responsive, but when I use Philips Hue Dimmer Switch, I'm sometimes experiencing latency (1-2sec). 
I'm using zgbee2mqtt/Mosquitto and HA for automations

aktentasche, 2019-04-18:
same here but for me it feels like this is only the case after switching to version 1.3.0

before reaction was instant

Koenkk, 2019-04-19:
In case you are having latency issues, please provide the `log_level: debug` (http://www.zigbee2mqtt.io/configuration/configuration.html) log with clear pointers with the messages we should look at.

aktentasche, 2019-04-19:
See attachment, the last 10 lines or so contain the button press of a Hue dimmer switch. The MQTT message is caught by node-red which then forwards it to another topic. This topic controls a LED strip, accessed via mysensors. When I use an "inject" node in node-red instead of the "dimmer switch topic", the LED strip switches on instantly. I selected this case instead of a "hue to hue" communication so simplify the issue. 

I don`t see anything suspicious in the log, what do you think @Koenkk?

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3097842/log.txt)

Koenkk, 2019-04-19:
I dont' see any delays, the message seems to be published instantly (assuming that these messages also appeared instantly when you pressed the button), that means the delay is probably somewhere in your nodred or LED strip accessed via mysensors.

aktentasche, 2019-04-21:
Ok, that is reassuring. However, the messages do not appear instantly when I do _journalctl -fu zigbee2mqtt_

I made a video showing the behaviour. The delay of the message appearing is pretty much the delay for the lights to switch on/off.

https://youtu.be/ZFR2N5FtKC8

If you deem this delay to be too high, I'd investigate a bit further on my installation, try to find out from where in the protocol stack the delay might be coming from.

Koenkk, 2019-04-22:
@aktentasche there is indeed quite some delay (mine responds much faster).

- How big is your network?
- How far is the button from the coordinator?
- Could there be wifi interference? (https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence)

ciB89, 2019-05-21:
I also notice a huge delay increase. Interestingly this started when I moved from ResinOS to Hassio (on an SSD installation). I also updated the firmware on my stick at the same time.

aktentasche, 2019-06-16:
Finally I have some time..

> @aktentasche there is indeed quite some delay (mine responds much faster).
> 
>     * How big is your network?
I reduced it to one Xiaomi switch WXKG03LM (started with a new database and configuration), no noticable change
>     * How far is the button from the coordinator?
Around 50cm
>     * Could there be wifi interference? (https://support.metageek.com/hc/en-us/articles/203845040-ZigBee-and-WiFi-Coexistence)
Indeed the CC2531 stick is co-housed with my router, so I moved the zigbee channel to a region in the spectrum least occupied.

The stick is connected to an OrangePi PC SBC running armbian, I played with process priority, no change. Disabled all other services, no change. Tried a powered USB hub, no change. However, when I connect the stick to my desktop PC I have little to none delay from button press to appearance in the log, so I assume it is something with the OrangePi or armbian.

aktentasche, 2019-06-16:
Ok so I changed the CPU governor to performance and minimum frequency to ~1GHz (before it was 120MHz) that helped a bit. So I assume this is a performance issue?

Koenkk, 2019-06-17:
@aktentasche I think we should start debugging with to see if there is a delay between the zigbee message send from the device, and the zigbee message being received by zigbee2mqtt.

To do this, sniff the traffic (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) and see if the moment that the message appears is that same as it appears in the log.

aktentasche, 2019-06-22:
Koen, I will order an additional CC2531 stick and try this.

rkantos, 2019-08-15:
I am experiencing the same issue when going from native OpenHAB zigbee binding to Zigbee2MQTT. On OpenHab2 the latency is ~250ms, while with Zigbee2MQTT the latency varies from around 500ms to 1500ms, and occasionally even more.. With Openhab zigbee binding the latency isn't discernible, but qith zigbee2MQTT it is very noticeable.

Openhab, zigbee2MQTT and openhab embedded server are all running on the same LXC container on Proxmox with Intel i5 2nd gen.

insensitiveclod, 2019-10-10:
I am having issues with lag too; but it's sporadic.

Sometimes it might be a few seconds before a first toggle of a lamp has it respond.
AFter that, it usually responds nicely for a while, toggling on-off-on-off-on-off ..etc.
But, typically within 10 seconds from starting to toggling it, it suddenly just stops responding.
Interestingly, all-of-a-sudden, it will respond again and perform all the delayed toggles very quickly one-after-another, as if playing a queued backlog.

As far as I could discern, the delay is not on the MQTT; it's running Mosquitto on a Raspi4 with Hassio; though I will endevour moving the Zigbee2mqtt stick to a proper PC and have Hassio connect to a mosquitto-instance there; just to rule that out.

I also just tested with a stick that has the SourceRouting version of the firmware on it.
My network is 30+ devices by now.
In the network-map (via the node-red admin-panel), I do not see a real difference in how it connects, however. It's typically a round star with an almost full mesh of connections between all the routers+coordinator. I thought the Source-Routing firmware was supposed to reduce this to 5 from the Coordinator/zigbeestick. Should I re-pair all the devices to have that work that way , or is the topology supposed to auto-settle into a new tree-based pattern ?

insensitiveclod, 2019-10-10:
Ok. I have a full resolution for the lag issue that I had and that sounds very very similar to the one that @nono313 reported; specifically the last part of this description:

'In some cases, I ended up with several seconds of delay / latency. In some cases, a couple of commands had been queued and executed at once several seconds after the first command.'

It seems to have to do with the time/effort it takes to process the 'graphviz' request to zigbee2mqtt .
In the interval between recieving the graphviz request , and it producing an output, the responsiveness of any device you are trying to control drops to almost (not quite) 0%.. Every now and then, something gets through; but overall it's entirely stalled until the graph-data is returned.

On my hass.io install, I have a node-red running with the Zigbee2mqtt dashboard. This , by default, has a 30 second update-interval for the 'network-map' node; it sends a 'graphviz' request.

Just to try things out, I changed 'graphviz' to something useless; and then suddenly all the latency/delays stopped.

Now, I had the network-map stuff installed from the start. But things only started becoming slow after adding more and more devices. I expect that this increases the time between 'graphviz' request and the response; increasing the window of delays. Currently there are 30 devices paired, with 6 of them offline; some of those remotes that simply havent been touched for a while; some of them bulbs with no power.

I suspect that the amount of unreachable/offline devices might play a factor in things too ? 
I do not know exactly how the network-map is created , but, could it be that:
a) z2m has some blocking code in the network-map segment ?
or
b) the network-mapping requires a lot of traffic ?

and perhaps
b2) the issues become worse with more devices being unresponsive/unreachable ?

I might try, later, to figure out just at what point/device the 'graphviz' call starts to take a lot of time and/or starts to cause unresponsiveness (might be it just takes long but lets other commands through fine; which'd be perfectly okay behaviour).

closed 2019-12-16 by stale[bot]