Koenkk/zigbee2mqtt #1270: Migrate from Hass.io to Docker - Best Method?

hgelpke, 2019-03-17:
I'm currently running Zigbee2Mqtt but would like to move over to Docker (via Unraid)

I have HA, Z2M, and MQTT dockers installed. 

What do I need to do do bring the paired devices over and not have to re-pair everything?

jasimancas, 2019-03-18:
Copy the folder "data", turn on the docker with z2m and voila, the devices are there.

hgelpke, 2019-03-18:
Seeming that worked but HA is not pulling in the lights I had linked

It seems that auto discovery isn't recognizing them.

Karevski, 2019-03-18:
Hi. I am looking to migrate the same way. How did you go about passing the usb to the docker?

hgelpke, 2019-03-18:
In Unraid:

Added a device in the "additional options" at the bottom.

The only really tricky part is if you already have it committed to a VM and
want to get Docker to control it. It take a bit of mounting/unmounting from
the VM

hgelpke, 2019-03-19:
> Hi. I am looking to migrate the same way. How did you go about passing the usb to the docker?

Here is a screenshot of my Docker:

https://i.imgur.com/wMhsO85.png

dh-harald, 2019-03-20:
try docker-compose, something like this (I'm on arm64 box, so maybe you should chose different manifests):
```
version: '3'

services:
  hass:
    depends_on:
      - mosquitto
    image: homeassistant/aarch64-homeassistant:0.89.2
    network_mode: "host"
    restart: always
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /containers/hass/config:/config
  mosquitto:
    image: eclipse-mosquitto:latest
    network_mode: "host"
    restart: always
    volumes:
      - /containers/mosquitto/config:/mosquitto/config
      - /containers/mosquitto/data:/mosquitto/data
      - /containers/mosquitto/log:/mosquitto/log
  zigbee2mqtt:
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
    depends_on:
      - mosquitto
    image: koenkk/zigbee2mqtt:arm64v8
    network_mode: "host"
    restart: always
    volumes:
      - /containers/zigbee2mqtt/data:/app/data
```

closed 2019-05-26 by stale[bot]