Koenkk/zigbee2mqtt #1271: Help: Adding Honyar 3-gang in-wall switch

hhalibo, 2019-03-17:
I have a new 3-gang HONYAR in-wall switch, I add the following information into the node_modules/zigbee-shepherd-converters/devices.js

{
    zigbeeModel: ['00500c35'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'U86K31ND6', // Vendor model number, look on the device for a model number
    vendor: 'Honyar', // Vendor of the device (only used for documentation and startup logging)
    description: 'Hongyan 3gang switch ', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},

then I push the button of the switch, no reply. The database.db is here
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/2975110/database.zip)

Koenkk, 2019-03-19:
Hello!

Please try with:
```js
{
    zigbeeModel: ['00500c35'],
    model: 'U86K31ND6',
    vendor: 'Honyar',
    description: '3 gang switch ',
    supports: 'on/off',
    fromZigbee: [],
    toZigbee: [tz.on_off],
    ep: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
        const ep1 = shepherd.find(ieeeAddr, 1);
        const ep2 = shepherd.find(ieeeAddr, 2);
        const ep3 = shepherd.find(ieeeAddr, 3);
        const actions = [
            (cb) => ep1.bind('genOnOff', coordinator, cb),
            (cb) => ep1.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            (cb) => ep2.bind('genOnOff', coordinator, cb),
            (cb) => ep2.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            (cb) => ep3.bind('genOnOff', coordinator, cb),
            (cb) => ep3.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
         ];

        execute(device, actions, callback);
    },
},
```

You should be able to control the device through `zigbee2mqtt/[DEVICE_FRIENDLY_NAME]/left/set` with payload `{"state": "ON"}`

hhalibo, 2019-03-19:
THX! I test it immediately！

hhalibo, 2019-03-19:
I use a docker running zigbee2mqtt server, when I copied and pasted your code, restart zigbee2mqtt container, it's dead...

so I write a code like this 
###     {
        zigbeeModel: ['00500c35'],
        model: 'U86K31ND6',
        vendor: 'HONYAR',
        description: 'Smart light switch - 3 gang',
        supports: 'on/off',
        fromZigbee: [fz.generic_state_multi_ep, fz.ignore_onoff_change],
        toZigbee: [tz.on_off],
        ep: (device) => {
            return {'left': 1, 'center': 2, 'right': 3};
        },
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const ep1 = shepherd.find(ieeeAddr, 1);
            execute(ep1, [(cb) => ep1.bind('genOnOff', coordinator, cb)], () => {
                const ep2 = shepherd.find(ieeeAddr, 2);
                execute(ep2, [(cb) => ep2.bind('genOnOff', coordinator, cb)], () => {
                    const ep3 = shepherd.find(ieeeAddr, 3);
                    execute(ep3, [(cb) => ep3.bind('genOnOff', coordinator, cb)], callback);
                });
            });
        },
    },

It's working so well. I am newbie here and my English is poor, ha...

Thx again for your amazing project !

Koenkk, 2019-03-19:
Is the device working now?

hhalibo, 2019-03-19:
Yes, it's working very flawless.

closed 2019-03-21 by hhalibo

Koenkk, 2019-03-21:
thanks, will be supported in next release.