Koenkk/zigbee2mqtt #1273: Hi,

MenoRohde, 2019-03-17:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

MenoRohde, 2019-03-17:
Hi,
to install zigbee2mqtt on a Raspberry Pi 3B+ I followed the instructions
in https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
Starting npm delivers the following errors:

pi@raspberrypi:/opt/zigbee2mqtt $ npm start
> zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
> node index.js
  zigbee2mqtt:info 2019-3-17 20:34:11 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-17.20-34-11'
  zigbee2mqtt:info 2019-3-17 20:34:13 Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
  zigbee2mqtt:info 2019-3-17 20:34:13 Starting zigbee-shepherd
Segmentation fault
npm ERR! code ELIFECYCLE
npm ERR! errno 139
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 139
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-03-17T19_34_14_003Z-debug.log

The log file is:
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gy$
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 139  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 139
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js$
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spaw$
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.98-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 139
22 error zigbee2mqtt@1.2.1 start: `node index.js`
22 error Exit status 139
23 error Failed at the zigbee2mqtt@1.2.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 139, true ]

Node version is 10.15.3
npm version is 6.4.1

What can I try next?

Jan

nickrout, 2019-03-17:
DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Mific78, 2019-03-18:
[<script src="https://pastebin.com/embed_js/W07RdjEj"></script>](url)

jakenl, 2019-03-18:
Unfortunately, by following the exact instructions, I end up with the same error message, about 'segmtation fault' and exiting with error 139
https://pastebin.com/pkNgsHts

nickrout, 2019-03-19:
Provide a log with log_level set to debug, see https://www.zigbee2mqtt.io/configuration/configuration.html

Mific78, 2019-03-19:
I have exactly the same error :
- See [https://pastebin.com/0jw0XXzp](url) for the log (with log_level set to debug).
- See [https://pastebin.com/CZ6UBzRe](url) for the console traces.
Thanks for your help !

Mific78, 2019-03-19:
I have exactly the same error :
    See https://pastebin.com/0jw0XXzp for the log (with log_level set to debug).
    See https://pastebin.com/CZ6UBzRe for the console traces.
 Thanks for your help !

jakenl, 2019-03-19:
> Provide a log with log_level set to debug, see https://www.zigbee2mqtt.io/configuration/configuration.html

Thanks for the suggestion. However, this is already with loglevel: debug enabled.

btw, this is the 'normal log (i added some enters on line 2 to show the whole line):

3/19/2019, 7:10:11 PM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-19.19-10-11'
3/19/2019, 7:10:12 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,
"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},
"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
3/19/2019, 7:10:12 PM - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
3/19/2019, 7:10:12 PM - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
3/19/2019, 7:10:12 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #ad835e0)

nickrout, 2019-03-19:
@Mific78 your console trace shows the failure seems to be on zigbee-shepherd starting. (But you probably already figured that out).

@Koenkk any ideas?

jakenl, 2019-03-19:
> @Mific78 your console trace shows the failure seems to be on zigbee-shepherd starting. (But you probably already figured that out).
> 
My console is the same, I just don't know what zigbee-shepeherd is executing. Does it talk with the USB-stick, or does it use the created database? could it be that the database.db is not defined correctly? Is the firmware for the USB stick not correct, or the image of the Texas Instrument flashing software showing wrong settings for flashing the image? Is the cause a configuration.yaml file without any devices?

Mific78, 2019-03-20:
With the same USB stick connected to my Rasberry Pi, everything is OK. The problem occurs when the USB stick is connected to my Ubuntu desktop. So the problem is not the USB stick or the firmware.

nickrout, 2019-03-20:
Does the user running zigbee2mqtt have access to the device? ls -l /dev/ttyUSB0. On ubuntu it is usually in the dialout group, if so add your user to the dialout group.

Mific78, 2019-03-20:
The result of  : `ls -l /dev/ttyACM0` is =>
`crw-rw---- 1 root dialout 166, 0 mars  20 21:39 /dev/ttyACM0`

The result of : `groups choupi` ("choupi" is the user accessing to the device) =>
`choupi : choupi adm tty dialout cdrom sudo dip plugdev lpadmin sambashare wireshark`

So the user running zigbee2mqtt has access to the device.

jakenl, 2019-03-20:
Same here, user pi is member of group dial out.

nickrout, 2019-03-21:
OK that blows that theory.

jakenl, 2019-03-21:
Sweet, how can I install 1 1.1?
Do I need to remove 1.2.1? How?

Joerg633, 2019-03-21:
Remove: manually remove /opt/zigbee2mqtt (includes all config data...!!) for a "clean" restart.
Copy folder ziggbee2mqtt-1.1.1 from downloaded and unpacked v1.1.1 [archive](https://github.com/Koenkk/zigbee2mqtt/archive/1.1.1.tar.gz) to /opt/zigbee2mqtt
Then follow the instructions on ["Running zigbee2mqtt..."](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html), starting with
_sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm install..._

maybe before you start, check autostart service respectively disable autostart with
_sudo systemctl disable zigbee2mqtt.service_
boot raspi and then start with reinstall mentioned above...
Problem resolved (for me)

jakenl, 2019-03-21:
I tried the same, but the npm install failed.
Googling on the error report, I came across a known issue in 1.1.1 where was pointed towards the dev branch. I wiped everything and followed the instructions. Results: it works for me as  well, but then running the dev branch of 1.2.1

https://github.com/Koenkk/zigbee2mqtt/issues/1028#issuecomment-460828365

burgano84, 2019-03-21:
Thanks @jakenl !!! it works :-)

I've been looking for a solution during almost 3 weeks! Thanks a lot

Joerg633, 2019-03-22:
> Results: it works for me as well, but then running the dev branch of 1.2.1

@jakenl & @burgano84 
Firmware 20190109 or 20190223 ?

jakenl, 2019-03-22:
The latest, as of last Monday (that is when I started this project)

Mific78, 2019-03-22:
It works for me as well, but with the dev branch of 1.2.1.

So for the samle USB stick and the same firmware :
- OK on my Rasberry Pi with 1.2.1 stable
- KO on Ubuntu with 1.2.1 stable
- OK on Ubuntu with 1.2.1 dev

burgano84, 2019-03-23:
Hi @Joerg633, In 20190223. Regards,

closed 2019-05-29 by stale[bot]