Koenkk/zigbee2mqtt #1275: Ikea E1525 motion detector not sending any message

Fresher14, 2019-03-17:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

Fresher14, 2019-03-17:
Hello,
I'm quite new to zigbee2mqtt. First trial is with an Ikea Motion Detector.
Pairing looks good but not any motion detected. Nothing happens.
Domoticz is installed. But no way to add this device neither.
Well, one issue at a time. What can prevent E1525 from sending any message related to motion detection ?

  zigbee2mqtt:info 3/17/2019, 11:24:49 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-17.23-24-49'
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM zigbee-shepherd started
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM Coordinator firmware version: '20190109'
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM Currently 1 devices are joined:
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM PIR_Ikea_01 (0xd0cf5efffef28e45): E1525 - IKEA TRADFRI motion sensor (EndDevice)
  zigbee2mqtt:warn 3/17/2019, 11:24:50 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 3/17/2019, 11:24:50 PM Allowing new devices to join.
  zigbee2mqtt:warn 3/17/2019, 11:24:50 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 3/17/2019, 11:24:50 PM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 3/17/2019, 11:24:51 PM zigbee-shepherd ready
  zigbee2mqtt:info 3/17/2019, 11:24:51 PM Connected to MQTT server
  zigbee2mqtt:info 3/17/2019, 11:24:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 3/17/2019, 11:24:51 PM MQTT publish: topic 'zigbee2mqtt/PIR_Ikea_01', payload '{"battery":74,"linkquality":31}'
  zigbee2mqtt:info 3/17/2019, 11:24:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
  zigbee2mqtt:warn 3/17/2019, 11:25:41 PM Failed to configure PIR_Ikea_01 (0xd0cf5efffef28e45) ('Error: Timed out after 10000 ms') (attempt #1)
  zigbee2mqtt:warn 3/17/2019, 11:25:41 PM This can be ignored if the device is working properly

Kryzek, 2019-03-18:
I believe this is caused by the fact (same as with Ikea Remote control (E1524)) that ```20190223``` firmware is required (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.2.0). 

Log is hinting you are using ```20190109```.

Koenkk, 2019-03-18:
@Kryzek is right, updating to, in order to use this device you need at least `20190223`.

closed 2019-03-18 by Koenkk

Fresher14, 2019-03-18:
Ok that's clear. I need a programmer now (I bought a pre-programmed key to get it faster).
Thanks for quick support.

vikdb, 2019-03-26:
I got 20190223 and nothing happens.. Do i need to press something? does it need to be in night or day mode?

Fresher14, 2019-03-27:
Latest FW installed. E1525 needed to be reset. Now motion detection is processed by zigbee2mqtt. So far so good. Next step is to understand why Domoticz does not see the device. But probably more a mis-configuration in Domoticz.

Fresher14, 2019-03-27:
This time it works. Plugin in Domoticz needed to be updated.

jsponz, 2020-05-05:
I am facing the same issue. The IKEA module was updated via zigbee2mqtt, but even after that I am not receiving anything. 

It is paired... 

I get this:

{"update_available":false,"battery":100,"linkquality":0}

after powering... 

Any help?

EricPalmquist, 2022-12-02:
@jsponz i am having the same issue.  Were you able to fix it?

jsponz, 2022-12-02:
@EricPalmquist the device is back in the box :-(

I will try to test it again during the next days.