Koenkk/zigbee2mqtt #1278: Smartthings Centralite outlet reports power multiplied by 10

jeromelaban, 2019-03-17:
Version 1.2.1

Smartthings Centralite outlet is recognized as `4256251-RZHAC` and reports instant power multiplied by 10.

Device description:
```
{"id":61,
"type":"Router",
"ieeeAddr":"xxxx",
"nwkAddr":xxx,
"manufId":49887,
"manufName":"CentraLite",
"powerSource":"Mains (single phase)",
"modelId":"4257050-RZHAC",
"epList":[1],"status":"online","joinTime":1552860889,"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2820,2821],
"outClusterList":[25],"clusters":{}}},"_id":"xxxxx"}
```

jeromelaban, 2019-03-19:
Otherwise works great!

Koenkk, 2019-03-19:
Please change this to: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L1590 `power: msg.data.data['activePower'] / 10,`.

If that works, can you make a PR?

jeromelaban, 2019-03-20:
Thanks! It most probably will work, though my question is more about why it was setup that way originally, and whether it could be that the original device (not the Smartthings one) was behaving differently.

I'll make a PR.

Koenkk, 2019-03-20:
@eagleadam I see that you added support for this device initially, do you recognize the issue mentioned here?

EagleAdam, 2019-03-22:
Hi, indeed I do remember something like that but have since then replaced the device with another one so I couldn't check... I actually have some memory of the power readings being erratic (sometimes 10x but not always.) Sorry not to be of more help...

closed 2019-05-28 by stale[bot]