Koenkk/zigbee2mqtt #1279: How long the log file to be kept

bongsam89, 2019-03-18:
Hi.

I just have a quick question.
Every time the raspberry pi is turned on, a log file is created at /share/zigbee2mqtt/log/. I found from previous issue that it keep the last 10 files, which means that the pi keeps the log file and records until the memory runs out if there are less than 10 files?

Koenkk, 2019-03-19:
- Max 10 directories
- Per directory 3 files
- Per file max 10 MB

bongsam89, 2019-03-20:
Thanks for your answer.
I have two more questions for that, if the first file exceeds 10 mb, it automatically creates a new file and saves continuously or it stops? 
Can I control the maximum size of the file?

Koenkk, 2019-03-20:
it automatically creates a new file. This can not be controlled without the changing the source code.

closed 2019-05-19 by Koenkk