Koenkk/zigbee2mqtt #1281: No converter for KS-SM001

Erelen-Laiquendi, 2019-03-18:
Hass.io zigbee2mqtt 1.2.1, сoordinator firmware 20190223.

I have two KS-SM001, and every every availability_timeout (300 seconds) i see 4 messages in log:

```
3/18/2019, 12:40:19 PM - warn: No converter available for 'KS-SM001' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":0}}'
3/18/2019, 12:40:19 PM - warn: No converter available for 'KS-SM001' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":0}}'
3/18/2019, 12:40:31 PM - warn: No converter available for 'KS-SM001' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":0}}'
3/18/2019, 12:40:31 PM - warn: No converter available for 'KS-SM001' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":0}}'
```

sapkra, 2019-03-18:
I have the same problem and I think the reason is that this line is set to `devChange` and should be `attReport` instead but maybe it will break the implementation of other devices.

https://github.com/Koenkk/zigbee-shepherd-converters/blob/962666e5523b418bc43dd0adb0de929cd8e5e6d6/converters/fromZigbee.js#L840

Koenkk, 2019-03-19:
Should be fixed with the latest dev branch.

sapkra, 2019-03-25:
It's working fine - thank you for your fast support.

```
payload '{"state":"ON","linkquality":76}'
```

closed 2019-03-25 by Koenkk