Koenkk/zigbee2mqtt #1282: Quirky POTLK-WH02 - ZHA Smart Plug

Mdleal, 2019-03-18:
Device does not show as not supported in the logs and is showing as an outlet but no mqtt message is being sent for the device.

Koenkk, 2019-03-19:
Please follow http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Mdleal, 2019-03-19:
Will do. I'm moving my HASS.IO install to ubuntu to make things easier.

Mdleal, 2019-03-22:
Logs:
`
3/22/2019, 8:11:37 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":7,"stackVersion":2,"hwVersion":1,"manufacturerName":"Quirky"}}' of device 'ZHA Smart $
3/22/2019, 8:11:37 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"ZHA Smart Plug","dateCode":"20141217","powerSource":1,"locationDesc":"","physicalEnv":1}}' of device '$
3/22/2019, 8:11:37 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"deviceEnabled":1,"alarmMask":0}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
3/22/2019, 8:11:37 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"identifyTime":0}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"identifyTime":0}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
3/22/2019, 8:11:38 PM - debug: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '1553285498153'
3/22/2019, 8:11:38 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00158d0001e550a6","type":"EndDevice","model":"RTCGQ11LM","friendly_name":"kitchen"},{"ieeeAddr$
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genGroups","data":{"nameSupport":128}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":128}}' of device 'ZHA Smart Plug' (0x7c$
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"lastCfgBy":"0xffffffffffffffff"}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
3/22/2019, 8:11:38 PM - debug: Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'`

Koenkk, 2019-03-22:
Please try with the following configuration:

```js
{
        zigbeeModel: ['ZHA Smart Plug'],
        model: 'POTLK-WH02',
        description: 'White smart wall outlet',
        supports: 'on/off',
        vendor: 'Quirky',
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
    },
```

Mdleal, 2019-03-22:
That works. It has power reading as well but that's not a big deal. Any ideas on how to get it discovered by home assistant. I'm guessing adding the following to homeassistant.js? 

'POTLK-WH02': [configurations.switch],

Koenkk, 2019-03-22:
For home assistant that should be enough,

Can you share you `database.db`?

Mdleal, 2019-03-22:
`{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d00022516ec","nwkAddr":7650,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"6WycgQ8jkAWW9291"}
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0001e550a6","nwkAddr":27123,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion.aq2","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,1,3,1024,1030,1280,65535],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"msIlluminanceMeasurement":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":1},"attrs":{}},"ssIasZone":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"INN35mub1seMxJAh"}
{"id":5,"type":"Router","ieeeAddr":"0x7ce524000004a8d6","nwkAddr":41831,"manufId":4316,"manufName":"GE_Appliances","powerSource":"Mains (single phase)","modelId":"ZLL Light","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":4,"stackVersion":2,"hwVersion":1,"manufacturerName":"GE_Appliances","modelId":"ZLL Light","dateCode":"20140812","powerSource":1,"deviceEnabled":1,"swBuildId":"1.3.4.0"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"remainingTime":0,"onOffTransitionTime":0,"onLevel":255,"onTransitionTime":0,"offTransitionTime":0,"defaultMoveRate":255}},"genOta":{"dir":{"value":2},"attrs":{"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":16974848,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"downloadedZigbeeStackVersion":65535,"imageUpgradeStatus":0,"manufacturerId":4316,"imageTypeId":65535,"minimumBlockReqDelay":0}},"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"WlK6lZgnfCN48WE7"}
{"id":7,"type":"EndDevice","ieeeAddr":"0x00158d000187297c","nwkAddr":33558,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"Z2kgIwiP2z8BuGKN"}
{"id":6,"type":"Router","ieeeAddr":"0x7c25e400000cc40a","nwkAddr":28678,"manufId":48042,"manufName":"Quirky","powerSource":"Mains (single phase)","modelId":"ZHA Smart Plug","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":2,"inClusterList":[0,3,4,5,6,1794,64544],"outClusterList":[25],"clusters":{}}},"_id":"hJ4KyDt2yzaUCfeh"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b00014513d3","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1553286887,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[0,1281,3],"outClusterList":[1280,1282,3],"clusters":{"genBasic":{"dir":{"value":1},"acls":{"zclVersion":"R","hwVersion":"R","manufacturerName":"R","modelId":"R","dateCode":"R","powerSource":"R","locationDesc":"RW","physicalEnv":"RW","deviceEnabled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"g [... cut ...] bled":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"sBk2X39VxlF1cGgf"}`

Koenkk, 2019-03-22:
Does metering work with: 

```js
    {
        zigbeeModel: ['ZHA Smart Plug'],
        model: 'POTLK-WH02',
        description: 'White smart wall outlet',
        supports: 'on/off',
        vendor: 'Quirky',
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            execute(device, [(cb) => device.report('seMetering', 'instantaneousDemand', 0, 3, 0, cb)], callback);
        },
    },
```

Mdleal, 2019-03-22:
zigbee2mqtt:warn 3/22/2019, 10:10:46 PM No converter available for 'POTLK-WH02' with cid 'seMetering', type 'attReport' and data '{"cid":"seMetering","data":{"instantaneousDemand":0}}'

Koenkk, 2019-03-22:
Does the value change when you draw power from it? (e.g. connect a bulb)

Mdleal, 2019-03-22:
I added 'POTLK-WH02': [configurations.switch, configurations.sensor_power], in homeassistant.js and do not see it updating.  I am also following the logs and see nothing. Am I looking in the wrong place?

Koenkk, 2019-03-22:
But do you still see zigbee2mqtt:warn 3/22/2019, 10:10:46 PM No converter available for 'POTLK-WH02' with cid 'seMetering', type 'attReport' and data '{"cid":"seMetering","data":{"instantaneousDemand":0}}' ?

Mdleal, 2019-03-22:
Correct.

Koenkk, 2019-03-23:
Can you try with

```js
    {
        zigbeeModel: ['ZHA Smart Plug'],
        model: 'POTLK-WH02',
        description: 'White smart wall outlet',
        supports: 'on/off',
        vendor: 'Quirky',
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            execute(device, [(cb) => device.report('seMetering', 'currentSummDelivered', 0, 3, 0, cb)], callback);
        },
    },
```

Mdleal, 2019-03-23:
Same. 
  zigbee2mqtt:warn 3/23/2019, 2:46:25 PM No converter available for 'POTLK-WH02' with cid 'seMetering', type 'attReport' and data '{"cid":"seMetering","data":{"instantaneousDemand":0}}'

Koenkk, 2019-03-23:
Can you re-pair the device and see if you still get the same?

Mdleal, 2019-03-23:
Sure. Do you want me to upload my log file somewhere?

Mdleal, 2019-03-23:
Here is the pairing part of the log.

`ZHA Smart Plug' (0x7c25e400000cc40a)
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"modelId":"ZHA Smart Plug","dateCode":"20141217","powerSource":1,"locationDesc":"","physicalEnv":1}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"deviceEnabled":1,"alarmMask":0}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"identifyTime":0}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'readRsp' with data '{"cid":"genGroups","data":{"nameSupport":128}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)
  zigbee2mqtt:debug 3/23/2019, 6:12:50 PM Received zigbee message of type 'devInterview' with data '"0x7c25e400000cc40a"'
  zigbee2mqtt:debug 3/23/2019, 6:12:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"genScenes","data":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":128}}' of device 'ZHA Smart Plug' (0x7c25e400000cc40a)`

Koenkk, 2019-03-23:
do you get any no converter available messages now?

Mdleal, 2019-03-24:
No I am not

Koenkk, 2019-03-25:
Can you try with

```js
{
    zigbeeModel: ['ZHA Smart Plug'],
    model: 'POTLK-WH02',
    description: 'White smart wall outlet',
    supports: 'on/off',
    vendor: 'Quirky',
    fromZigbee: [fz.ignore_onoff_change, fz.state],
    toZigbee: [tz.on_off],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const device = shepherd.find(ieeeAddr, 1);
        const cfg = {
            direction: 0, attrId: 0, dataType: 37, minRepIntval: 0, maxRepIntval: 3, repChange: 0,
        };

        const actions = [
            (cb) => device.foundation('seMetering', 'configReport', [cfg], foundationCfg, cb),
        ];

        execute(device, actions, callback);
    },
},
```

Mdleal, 2019-03-25:
/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:269
            throw e;
            ^

Error: The value for UINT48/BITMAP48/DATA48 must be orgnized in an 2-element number array.
    at getDataTypeBuf (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:482:23)
    at FoundPayload._getBuf (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:148:46)
    at /opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:112:22
    at Array.forEach (<anonymous>)
    at FoundPayload.frame (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:111:21)
    at Object.zcl.frame (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/zcl.js:66:78)
    at Object.af.zclFoundation (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:263:25)
    at ZShepherd._foundation (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:695:20)
    at Endpoint.ep.foundation (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:583:25)
    at Timeout.actions [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js:153:20)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/mdleal/.npm/_logs/2019-03-25T18_12_55_163Z-debug.log

Koenkk, 2019-03-26:
Can you provide the `DEBUG=* npm start` log?

Mdleal, 2019-03-29:
https://pastebin.com/yZFU3RMX

Koenkk, 2019-03-29:
I don't see the *Error: The value for UINT48/BITMAP48/DATA48 must be orgnized in an 2-element number array.* in that log.

Mdleal, 2019-03-29:
let's try this one. https://pastebin.com/VN543pze

The other one was some weird timeout errors. I had to reset my stick.

Koenkk, 2019-03-30:
Can you open `/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js` and add before https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L263 this code:

```js
console.log(frameCntl, manufCode, seqNum, cmd, zclData)
```

Mdleal, 2019-03-30:
https://pastebin.com/ziFdTYE5

Koenkk, 2019-04-01:
Can you open `/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js` and replace https://github.com/Koenkk/zcl-packet/blob/master/lib/foundation.js#L148 with 

```
dataBuf = dataBuf.buffer(getDataTypeBuf('int16', arg.repChange));
```

Mdleal, 2019-04-06:
https://pastebin.com/kPsq8qPN

Koenkk, 2019-04-08:
Can you revert the changes you made in zigbee-shepherd?

Perhaps do a fresh install with:

```
rm -rf node_modules
npm install
```

and apply the change from https://github.com/Koenkk/zigbee2mqtt/issues/1282#issuecomment-478689975 again.

closed 2019-06-14 by stale[bot]

knkaiser, 2020-04-02:
I recently moved to z2m and have a Quirky switch that I try to use. I followed the notes in this case but get an error just joining as an on/off switch.

{
        zigbeeModel: ['ZHA Smart Plug'],
        model: 'POTLK-WH02',
        description: 'White smart wall outlet',
        supports: 'on/off',
        vendor: 'Quirky',
        fromZigbee: [fz.ignore_onoff_change, fz.state],
        toZigbee: [tz.on_off],
    },

as far as I see @Mdleal reported this to work. Here is my DEBUG=* npm start log section

 serialport/stream binding.read finished { bytesRead: 18 } +332ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,27,24,27,24,42,195,12,0,0,228,37,124,142,95] +338ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,27,24,27,24,42,195,12,0,0,228,37,124,142,95] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [27,24,27,24,42,195,12,0,0,228,37,124,142] - 95 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":6171,"nwkaddr":6171,"ieeeaddr":"0x7c25e400000cc32a","capabilities":142} +343ms
  zigbee-herdsman:controller:log Device announce '0x7c25e400000cc32a' +338ms
zigbee2mqtt:debug 2020-04-01 21:24:50: Device '0x7c25e400000cc32a' announced itself
winston:file written true false
zigbee2mqtt:info  2020-04-01 21:24:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x7c25e400000cc32a"}}'
winston:file written true false
  mqtt-packet:writeToStream generate called +24s
  mqtt-packet:writeToStream generate: packet.cmd: publish +0ms
  mqtt-packet:writeToStream publish: packet: { cmd: 'publish', topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x7c25e400000cc32a"}}', qos: 0, retain: false, messageId: 33315, dup: false } +1ms
  mqtt-packet:writeToStream writeVarByteInt: writing to stream: <Buffer 76> +1ms
  mqtt-packet:writeToStream writeNumberCached: number: 22 +0ms
  mqtt-packet:writeToStream writeNumberCached: <Buffer 00 16> +0ms
  mqtt-packet:writeToStream publish: payload: '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x7c25e400000cc32a"}}' +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +16ms
winston:file logged 30858 debug 2020-04-01 21:24:50: Device '0x7c25e400000cc32a' announced itself

winston:file logged 31036 info  2020-04-01 21:24:50: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x7c25e400000cc32a"}}'

  serialport/stream _read reading { start: 675, toRead: 64861 } +20ms
  serialport/binding-abstract read +351ms
  serialport/bindings/unixRead Starting read +22ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -11,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +27ms
  serialport/bindings/poller received "readable" +606ms
  serialport/bindings/unixRead Starting read +608ms
  serialport/bindings/unixRead Finished read 37 bytes +1ms
  serialport/stream binding.read finished { bytesRead: 37 } +613ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,27,24,1,13,0,0,0,124,75,128,0,0,12,1,4,1,0,136,84,2,81,3,0,2,1,27,24,29,221] +616ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,27,24,1,13,0,0,0,124,75,128,0,0,12,1,4,1,0,136,84,2,81,3,0,2,1,27,24,29,221] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [0,0,25,0,27,24,1,13,0,0,0,124,75,128,0,0,12,1,4,1,0,136,84,2,81,3,0,2,1,27,24,29] - 221 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":6171,"srcendpoint":1,"dstendpoint":13,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":8407932,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,4,1,0,136,84,2,81,3,0,2,1]}} +633ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":21640,"imageType":20738,"fileVersion":16908291}},"address":6171,"endpoint":1,"linkquality":0,"groupID":0}' +638ms
zigbee2mqtt:debug 2020-04-01 21:24:50: Received Zigbee message from '0x7c25e400000cc32a', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":21640,"imageType":20738,"fileVersion":16908291}' from endpoint 1 with groupID 0
winston:file written true false
zigbee2mqtt:error 2020-04-01 21:24:50: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /srv/zigbee2mqtt/lib/extension/deviceReceive.js:106:42
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/srv/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
    at Controller.callExtensionMethod (/srv/zigbee2mqtt/lib/controller.js:346:44)
    at Controller.onZigbeeEvent (/srv/zigbee2mqtt/lib/controller.js:231:14)
    at Zigbee.emit (events.js:311:20)
    at Controller.<anonymous> (/srv/zigbee2mqtt/lib/zigbee.js:57:52)
    at Controller.emit (events.js:311:20)
[... cut ...]
  mqtt-packet:parser parse: packet.length: 0, buffer list length: 0 +1ms
  mqtt-packet:parser _newPacket +1ms
  mqtt-packet:parser _newPacket: parser emit packet: packet.cmd: pingresp, packet.payload: null, packet.length: 0 +1ms
  mqtt-packet:parser _newPacket: new packet +0ms
  mqtt-packet:parser parse: state complete. _stateCounter is now: 4 +1ms
  mqtt-packet:parser parse: packet.length: -1, buffer list length: 0 +1ms
  mqtt-packet:parser parse: exited while loop. packet: -1, buffer list length: 0 +0ms

Koenkk, 2020-04-02:
The converter name has change, try with

```js
{
zigbeeModel: ['ZHA Smart Plug'],
model: 'POTLK-WH02',
description: 'White smart wall outlet',
supports: 'on/off',
vendor: 'Quirky',
fromZigbee: [fz.on_off],
toZigbee: [tz.on_off],
},
```