Koenkk/zigbee2mqtt #1283: Zigbee CC2531 4dBm Wireless Transceiver E18-2G4U04B ?

tradenet, 2019-03-18:
Has anyone tried this device?
[https://www.aliexpress.com/i/32973371661.html?spm=2114.12057483.0.0.3c7451efLVrfMq](url)

User manual here:
[http://www.ebyte.com/en/downpdf.aspx?id=523](url)

I haven't seen any feedback on this device that I can find.
200 m operating distance is an interesting spec.

h4nc, 2019-03-19:
I mean both links here:

Both of your links go here https://github.com/Koenkk/zigbee2mqtt/issues/url

james-fry, 2019-03-19:
It seems to have no programming header

james-fry, 2019-03-19:
OK it does have a header, but you'd need to solder or use pogo pins to program it from the cc debugger.
IMO go for the old style green pcb ones unless you are ok with the programming connections.

tradenet, 2019-03-19:
> I mean both links here:
> 
> Both of your links go here https://github.com/Koenkk/zigbee2mqtt/issues/url

Yeah, I think I fixed that. It's githubs weird URL formatting.

tradenet, 2019-03-19:
> OK it does have a header, but you'd need to solder or use pogo pins to program it from the cc debugger.
> IMO go for the old style green pcb ones unless you are ok with the programming connections.

Well, what's interesting to me is it's 200m operating distance. Which the green pcb's seem lacking.

matejzero, 2019-03-19:
The antenna looks the same (is the same), so I doubt it will have a better range then the green PCB. But I could be mistaken.

h4nc, 2019-03-19:
Yeah may the CC2531 also has a 200m range with perfect conditions and without walls. 

Antenna looks the same as @matejzero said

fred-c1, 2019-03-19:
"200m" or even "200 meters" means nothing (beside "we want your money")
No standard "CC debugger" connector
PCB/board antenna layout seems poor (/ not as good as Texas Instruments reference design)
Conclusion = avoid E18-2G4U04B

closed 2019-05-25 by stale[bot]

mabusdogma, 2021-01-01:
Actually is as easy to flash as "normal" CC2531 ones
You need a CC debugger and 5 DuPont male-to-female cables

Method is described here:
https://community.home-assistant.io/t/flash-zigbee-e18-2g4u04b-without-soldering-zigbee2mqtt/253953