Koenkk/zigbee2mqtt #1284: Not starting CC2531 after crashing on switching too many lights off

jlo88, 2019-03-18:
Having a bunch of ikea tradfri bulbs, xiaomi door sensors, a xiaomi motions sensor and a tradfri motion sensor. 

I build the network and paired all my devices. When I switch off all the lights at the same time the stick seems to crash and it's very difficult and unpredictable for me to get it working again. I think the initial crash might be related to 1207 [#1207](https://github.com/Koenkk/zigbee2mqtt/issues/1027). 

When I try to start, I get this log: [logfile](https://pastebin.com/CRj5eeBS)

I'm running the dev branch of zigbee2mqtt and the latest firmware on the stick. 

My system is Ubuntu 18.10 and hassio 0.89.2

Things I tried to get the stick online again: 

1. Restart zigbee2mqtt, keeps coming up with the same error.
2. Press the reset button and restart zigbee2mqtt
3. Stop zigbee2mqtt, unplug and plug the stick back in, start zigbee2mqtt. I get further but get stuck on the following log: [log](https://pastebin.com/gzS7wrNv) where I see the error `Failed to mount the cieApp`

4. Stop zigbee2mqtt, turn the routers towards the device that causes the error off, flash the stick with a different machine, plug it back in and start again. Turning lights on one by one again Success! [log](https://pastebin.com/aqnYLbHS)

Why do I have to turn off devices to get it started again? It seems like one device is to blame for this (a xiaomi window sensor). 

Configuration: 
```
advanced:
  channel: 11
  elapsed: true
  last_seen: enable
  log_level: debug
  soft_reset_time: 60
devices: devices.yaml
groups: groups.yaml
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  password: <my_pw>
  server: mqtt://homeassistant
  user: <username>
permit_join: false
serial:
  port: /dev/ttyACM0
```

Devices:

```
3/17/2019, 9:54:54 PM - info: light_couch_left (0x000b57fffec492cb): LED1536G5 - IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (Router)
3/17/2019, 9:54:54 PM - info: light_couch_right (0x90fd9ffffed533c7): LED1546G12 - IKEA TRADFRI LED bulb E26/E27 950 lumen, dimmable, white spectrum, clear (Router)
3/17/2019, 9:54:54 PM - info: light_tv_right (0x000b57fffeea5bc3): LED1536G5 - IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (Router)
3/17/2019, 9:54:54 PM - info: light_couch_main (0x000b57fffe9de21d): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
3/17/2019, 9:54:54 PM - info: light_tv_left (0x000b57fffeaf7f74): LED1545G12 - IKEA TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white (Router)
3/17/2019, 9:54:54 PM - info: lightremote_livingroom_front (0x000b57fffeb20320): E1524 - IKEA TRADFRI remote control (EndDevice)
3/17/2019, 9:54:54 PM - info: light_livingroom_backside (0x000b57fffe9c29db): LED1546G12 - IKEA TRADFRI LED bulb E26/E27 950 lumen, dimmable, white spectrum, clear (Router)
3/17/2019, 9:54:54 PM - info: light_remote_diningtable (0x000b57fffec060f5): E1524 - IKEA TRADFRI remote control (EndDevice)
3/17/2019, 9:54:54 PM - info: window_couch (0x00158d0002c144bd): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: light_kitchen_panel (0xd0cf5efffe1d23cc): L1528 - IKEA FLOALT LED light panel, dimmable, white spectrum (30x90 cm) (Router)
3/17/2019, 9:54:54 PM - info: light_panel_bathroom (0x90fd9ffffef2ebb2): L1527 - IKEA FLOALT LED light panel, dimmable, white spectrum (30x30 cm) (Router)
3/17/2019, 9:54:54 PM - info: light_bedroom_main (0x000b57fffe9de304): LED1624G9 - IKEA TRADFRI LED bulb E27/E26 600 lumen, dimmable, color, opal white (Router)
3/17/2019, 9:54:54 PM - info: light_spot_bedroom_hallway (0x000b57fffeb27565): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
3/17/2019, 9:54:54 PM - info: light_spot_bedroom_window (0x000b57fffeb1cf5c): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
3/17/2019, 9:54:54 PM - info: light_spot_bedroom_center (0x000b57fffeb87597): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
3/17/2019, 9:54:54 PM - info: light_spot_hallway_center (0x000b57fffe9c20f1): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
3/17/2019, 9:54:54 PM - info: light_spot_hallwal_frontdoor (0x000b57fffeb0c052): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
3/17/2019, 9:54:54 PM - info: light_spot_hallway_toilet (0x000b57fffeb1e2f9): LED1537R6 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum (Router)
3/17/2019, 9:54:54 PM - info: light_hallway_across_toilet (0x000b57fffeeb7833): LED1536G5 - IKEA TRADFRI LED bulb E12/E14 400 lumen, dimmable, white spectrum, opal white (Router)
3/17/2019, 9:54:54 PM - info: light_remote_hallway (0x000b57fffe374468): E1524 - IKEA TRADFRI remote control (EndDevice)
3/17/2019, 9:54:54 PM - info: lightremote_bedroom (0x90fd9ffffee3e25f): E1524 - IKEA TRADFRI remote control (EndDevice)
3/17/2019, 9:54:54 PM - info: lightremote_kitchen (0x000b57fffeb26951): E1524 - IKEA TRADFRI remote control (EndDevice)
3/17/2019, 9:54:54 PM - info: light_kitchen_spicerack_bottom (0x000b57fffe9d2ff0): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
3/17/2019, 9:54:54 PM - info: light_kitchen_spicerack_top (0x000b57fffebac0d0): ICPSHC24-10EU-IL-1 - IKEA TRADFRI driver for wireless control (10 watt) (Router)
3/17/2019, 9:54:54 PM - info: lightremote_bathroom (0x90fd9ffffeff522e): E1524 - IKEA TRADFRI remote control (EndDevice)
3/17/2019, 9:54:54 PM - info: motion_bathroom (0xd0cf5efffebdf0b8): E1525 - IKEA TRADFRI motion sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: window_dining (0x00158d0002582d78): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: window_bathroom (0x00158d0002c1016f): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: door_front (0x00158d00025ec0ab): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: window_washingroom (0x00158d0002ec4f0f): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: motion_kitchen (0x00158d0002eda7b8): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: window_kitchen (0x00158d0002582cd5): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
3/17/2019, 9:54:54 PM - info: window_bedroom (0x00158d0002d6e5ad): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
```

Koenkk, 2019-03-19:
Please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5

Relative to the `20190223` firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5).

jlo88, 2019-03-20:
Okay, I'll try that tonight! Thanks!

Because I have a lot of routers, this should not be an issue for me right? Suppose that you would have only end devices this firmware would only allow 5 devices to connect at the same time? 

Another question, I have used the grouping functionality and added my lights as MQTT lights to home-assistant. I notice that the light status doesn't always bet updated of the individual bulbs when I switch the groups. What is the best method of checking the actual state of the lights?

jlo88, 2019-03-20:
I just flashed the stick with the provided firmware and I get this log: [log](https://pastebin.com/i4a8TfZF)
I still see a bunch of errors 16 and 17 such as: 
```
3/20/2019, 3:55:55 PM - error: Failed to bind for reporting for 0x90fd9ffffef2ebb2 - genOnOff - 1 - (Error: request unsuccess: 16)
3/20/2019, 3:55:55 PM - debug: Setup reporting for 0x90fd9ffffef2ebb2 - genLevelCtrl - 1
3/20/2019, 3:55:56 PM - error: Failed to bind for reporting for 0x90fd9ffffef2ebb2 - genLevelCtrl - 1 - (Error: request unsuccess: 17)
3/20/2019, 3:55:56 PM - debug: Setup reporting for 0x90fd9ffffef2ebb2 - lightingColorCtrl - 3
3/20/2019, 3:55:56 PM - error: Failed to bind for reporting for 0x90fd9ffffef2ebb2 - lightingColorCtrl - 3 - (Error: request unsuccess: 17)
3/20/2019, 3:55:56 PM - debug: Setup reporting for 0x000b57fffeb1cf5c - genOnOff - 1
3/20/2019, 3:55:56 PM - error: Failed to bind for reporting for 0x000b57fffeb27565 - genOnOff - 1 - (Error: Timed out after 10000 ms)
3/20/2019, 3:55:56 PM - error: Failed to bind for reporting for 0x000b57fffeb1cf5c - genOnOff - 1 - (Error: request unsuccess: 17)
```
However, everything is functioning. The network map also looks good: [network map](https://ibb.co/fHpjqcf). 

As a final test I tried switching all the lights off at the same time via groups. This also seems to work! So for now I think I'm very happy with this :+1: Thanks for helping me out! 

Just out of curiosity, as I understood an error 17 means that the buffer of the stick is full right? How come I still see those errors but stuff works this time?

BadEendt, 2019-03-20:
This happened to me as well when the stick crashed running, firmware from 20190109 and 20190218. I had to reflash the stick to get it working again. I don't know if the issue is still present in 20190223 because it has not crashed since.

I don't think i have the log files anymore, but if anyone is interested i can check

Koenkk, 2019-03-21:
@jlo88 don't worry about the `17` error. This error indeed means that the buffer is full. However, this error is caught by zigbee2mqtt and will be tried again.

jlo88, 2019-03-29:
@Koenkk I just came back from holiday and noticed that the network map changed dramatically: [network map](https://imgur.com/WL7gwNv), the old one: [network map old](https://ibb.co/fHpjqcf). 

I also notice that I cannot use one of my Xiaomi Aqara motion sensors anymore. When I try repairing it works but no status updates come by. The rest devices on the network function fine. 

I tried restarting zigbee2mqtt and switching from dev to the master branch. I'm considering now to reflash the stick. 

The firmware on the stick is responsible for creating the mesh etc right? That would explain why restarting zigbee2mqtt does not have any effect. 

Do you have any thoughts on this why the mesh changes so much? I was also not at home for a week if that information helps any. 

I'm also not sure if the mesh change is connected to the not functioning motion sensor, those things can also be unrelated and might as well be on the sensor side. I checked the motion sensor battery and it's completely full, I also only have it for a few weeks yet.

Koenkk, 2019-03-29:
@jlo88 no, the devices itself are responsible for creating the mesh. About the networkmap, see https://github.com/Koenkk/zigbee2mqtt/issues/1001#issuecomment-473227310.

rickkdotnet, 2019-05-31:
I think I've just ran into the same issue. Ofcourse just right after I've installed and paired the last of my 41 devices :((( 

Turned off a bunch of lights (yes, Ikea ones) at the same time and z2m stopped responding. Orange led on CC2531 was flashing. Tried plugging it in again, restarted z2m and the green led went off.  Restarting z2m, rebooting, powercycling doesn't work. When I start z2m the green led turns off and z2m just sits there on " Starting zigbee-shepherd". 

z2m 1.4.0 and firmware 20190223

rickkdotnet, 2019-06-01:
Had to reflash the stick. Things are back (without re-pairing) but interestingly enough they now all report linkquality:0 and the first devices have already lost their connections.

jlo88, 2019-06-03:
@rickkdotnet Sounds very frustrating and reminds me of how it used to go with me. @Koenkk recommended this firmware to me: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5 which solved my problems. See: 

> Please try with the following firmware: https://drive.google.com/open?id=1-xzI6b8umZFpki-pfaKdLgcPrUUlswe5
> 
> Relative to the `20190223` firmware, it has an increased memory heap at the cost of direct connected devices to the coordinator (15 -> 5).

Koenkk, 2019-06-03:
@rickkdotnet note that the firmware you were using was old, recommended firmwares can be found here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin

rickkdotnet, 2019-06-03:
Thanks @jlo88 and @Koenkk. 
I didn't try the special build since I'm kinda apprehensive pushing more devices off the coordinator to routers because of my issues with Tradfri bulbs as routers (https://github.com/Koenkk/zigbee2mqtt/issues/1566#issuecomment-498158711) , but 'll give it a shot if my problems persist.

I'm on the "max_stability" 20190315 now, which seems relatively stable except from my issues with Tradfri bulbs and the fact that link_quality is always reported as 0. 

The latest recommended firmware (20190425?) also reported 0 linkquality but when I started losing devices I quickly went to 20190315. 

I have ordered another stick, once it's in I'll experiment with different firmwares.

closed 2019-08-09 by stale[bot]