Koenkk/zigbee2mqtt #1285: dev Update

salopette, 2019-03-18:
Hello, I'm bing again!
What is this message when trying to make an update?

If update is finished the log remains in spite of the branch  commit #e2194b5

2019-3-18 18:36:06 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-18.18-36-06'
2019-3-18 18:36:07 - info: Starting zigbee2mqtt version 1.2.1 (commit #e2194b5)
2019-3-18 18:36:07 - info: Starting zigbee-shepherd
2019-3-18 18:36:08 - info: zigbee-shepherd started
2019-3-18 18:36:08 - info: Coordinator firmware version: '20190223'
----------------------------------------------------------------------------------------------------------

pi@raspberrypi:/opt/zigbee2mqtt $ bash update.sh
Stopping zigbee2mqtt...
Creating backup of configuration...
Updating...

*** Bitte geben Sie an, wer Sie sind.

Führen Sie

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

aus, um das als Ihre standardmäßige Identität zu setzen.
Lassen Sie die Option "--global" weg, um die Identität nur
für dieses Repository zu setzen.
fatal: empty ident name (for <(null)>) not allowed

salopette, 2019-03-19:
nobody can help me?

Koenkk, 2019-03-19:
What is the output of cat `.git/config`?

salopette, 2019-03-19:
pi@raspberrypi:~ $ git config
Verwendung: git config [<Optionen>]

Ort der Konfigurationsdatei
    --global              globale Konfigurationsdatei verwenden
    --system              systemweite Konfigurationsdatei verwenden
    --local               Konfigurationsdatei des Repositories verwenden
    -f, --file <Datei>    die angegebene Konfigurationsdatei verwenden
    --blob <Blob-Id>      Konfiguration von angegebenem Blob-Objekt lesen

Aktion
    --get                 Wert zurückgeben: Name [Wert-regex]
    --get-all             alle Werte zurückgeben: Schlüssel [Wert-regex]
    --get-regexp          Werte für den regulären Ausdruck zurückgeben: Name-regex [Wert-regex]
    --get-urlmatch        Wert spezifisch für eine URL zurückgeben: section[.var] URL
    --replace-all         alle passenden Variablen ersetzen: Name Wert [Wert-regex]
    --add                 neue Variable hinzufügen: Name Wert
    --unset               eine Variable entfernen: Name [Wert-regex]
    --unset-all           alle Übereinstimmungen entfernen: Name [Wert-regex]
    --rename-section      eine Sektion umbenennen: alter-Name neuer-Name
    --remove-section      eine Sektion entfernen: Name
    -l, --list            alles auflisten
    -e, --edit            einen Editor öffnen
    --get-color           die konfigurierte Farbe finden: Slot [Standard]
    --get-colorbool       die Farbeinstellung finden: Slot [Standard-Ausgabe-ist-Terminal]

Typ
    --bool                Wert ist "true" oder "false"
    --int                 Wert ist eine Dezimalzahl
    --bool-or-int         Wert ist --bool oder --int
    --path                Wert ist ein Pfad (Datei oder Verzeichnisname)

Sonstiges
    -z, --null            schließt Werte mit NUL-Byte ab
    --name-only           nur Variablennamen anzeigen
    --includes            beachtet "include"-Direktiven beim Nachschlagen
    --show-origin         Ursprung der Konfiguration anzeigen (Datei, Standard-Eingabe, Blob, Befehlszeile)

pi@raspberrypi:~ $

Koenkk, 2019-03-20:
I'm not sure why you get this and other dont, but executing this in the zigbee2mqtt should prevent the error:

```bash
git config user.email "you@example.com"
git config user.name "Your Name"``

closed 2019-05-19 by Koenkk