Koenkk/zigbee2mqtt #1289: Flash cc2531 with raspberry

jmichault, 2019-03-19:
Hello,
I have successfully flashed my CC2531 directly connected to my Raspberry pi3.
For those interested, software and code is available at https://github.com/jmichault/flash_cc2531.

tb-killa, 2019-03-19:
Hi nice, does this mean we could use the rpi serial for data transfer too?
@Koenkk this could help build some sort of update routine inside zigbee2mqtt.
I think about something like a mqtt topic where we could bring the device to bootloader and flash, after reset we run with the Z2M instance again.

Koenkk, 2019-03-19:
@jmichault awesome! would you mind adding this to the docs? https://github.com/Koenkk/zigbee2mqtt.io/blob/master/information/alternative_flashing_methods.md

@tb-killa this is the same as using the CC debugger (but now without the debugger), you also need to connect DC and DD pins. In order to have such a function updating should be possible without connecting this (only via USB port).

keitetran, 2019-03-21:
I tried and succeeded. No need to have hardware debuger
Please update your pin map.

VincentMasselis, 2019-03-23:
I'm trying but my CC2531 green led turns off when I call `./cc_chipid`.

[image]

I figured out that my chip led also turns off when I press the "reset" button on my CC Debugger.

Any idea ?

timdonovanuk, 2019-03-28:
Out of interest, how long does it take to flash using a Pi? Absolutely amazing work, well done :) 👍

closed 2019-06-03 by stale[bot]