Koenkk/zigbee2mqtt #1291: WeMo smart LED bulb bug

crag-r, 2019-03-19:
Hi Koen, 
I have recently upgraded to 1.2.1 from 0.2.0 and it's changed the way how my WeMo bulb is reporting using MQTT. Are there any breaking changes I have missed? I know I was very out of date.
I am using Home Assistant to turn it on and off.
In 0.2.0 turning it Off would turn the Light off and the switch would stay Off in Home Assistant, now in 1.2.1 turning it Off does turn the Light off however the switch toggles back to On instantly.
I downgraded again to double check and I have attached the two log snippets from each version. It has always sent two MQTT messages  - I presume this is by design but in 1.2.1 it sends the Off command to the Zigbee device correctly, then MQTT reports Off then On to Home assistant, not sure why?

I have flashed my CC2531 with 20190223 but it hasn't made any difference.
Any advice? 

Log files are here;
[0.2.0.txt](https://github.com/Koenkk/zigbee2mqtt/files/2985599/0.2.0.txt)
[1.2.1.txt](https://github.com/Koenkk/zigbee2mqtt/files/2985600/1.2.1.txt)

Please let me know if you need any more information.
Thank you

Koenkk, 2019-03-21:
Should be fixed in the dev branch, can you check?

crag-r, 2019-03-21:
Hi Koen, yes that has fixed it thank you.

Koenkk, 2019-03-21:
great!

closed 2019-03-21 by Koenkk