Koenkk/zigbee2mqtt #1292: Tradfri bulbs not acting as router

StimpyMGS, 2019-03-19:
As the title says my Tradfri bulbs are not acting as routers. They do however show up as routers according to the networkmap, but everything in my network only connects directly to my coordinator. I have paired the network many times but this behaviour is always the same. This results in poor response of the lights and end devices (aqara motion sensors and wireless switches). When pairing it either pairs with the coordinator or the pairing fails. I'm running the latest firmware on a cc2531 and zigbee2mqtt 1.21. System: Ubuntu running hass.io
Any suggustions? 
http://i63.tinypic.com/mblg1v.png

LucaKaufmann, 2019-03-20:
My network map usually looks quite similar, eventually the bulbs start connecting to each other but each bulb always seems to connect to the coordinator as well. Maybe this is just how Zigbee works? Routers connect to the coordinator and end devices connect to routers, but no router to router only connections.

I ended up moving some bulbs back to the Tradfri hub since I think I've hit the limit of devices that can be connected to the coordinator, that helped a bit with the performance but it's still a bit unreliable and unresponsive at times.

way2many, 2019-03-20:
I just tested and first map I got was that everything was connected to the coordinator like your case. After 5 minutes I just redone a network map without doing anything and now I have a near perfect mesh network.

[image]

I think it's still pretty much an experimental tool and might not reflect the real situation. Xiaomi will pair with the coordinator for a long time but after a while it will jump to a router (this is my case). The problem on the network might be due to WiFi interference from you or neighbors. Example: almost all my neighbors are on channel 1 for Wifi which interferes with my zigbee channel 11 (default I think). I should change the zigbee channel but I'm lazy to re-pair everything and a delay of 1s is acceptable for me and is really not often. There are some good websites that you can find quickly with google which explains it further.

StimpyMGS, 2019-03-20:
I have already changed the default channel to 14,still the same behaviour, also over a couple of days. I picked 14 because in my appartment that was the least busy channel.

way2many, 2019-03-21:
My experience with Ikea E27 and GU10 is that they are less stable than Philips and, indeed, doesn't really like other non-Tradfri end devices.

On my map I have two Xiaomi end devices connected on my Philips while they should have a better signal if connected on Ikea. Same for a Xiaomi door sensor which is still connected to the coordinator while there are two Ikea bulbs at less than 1 meter (I will try Landrash method on placing it on the bulb). I plan to add Philips bulb here and there to compensate this "behavior" from Ikea.

There was also a discussion on reddit about poor performance of Tradfri and apparently this is why they have now those routers with usb charging port. But, hey, they are 50% cheaper than Philips after all ... (excluding promo prices).

StimpyMGS, 2019-03-21:
Thx for al the info. I used landrash's method with succes: the end devices are now connected through the bulb. Insta better performance!

closed 2019-05-27 by stale[bot]