Koenkk/zigbee2mqtt #1294: Error Zigbee2mqtt after one or two hours

joseska, 2019-03-19:
Hi all. I have this error with zigbee2mqtt (https://pastebin.com/fEzqNge2). After one or two hours the software crash. I´m using the last version of  Zigbee2mqtt on Docker "koenkk/zigbee2mqtt:arm32v6" on Raspberry pi 3 (update yesterday) and the Coordinator firmware version is 20190223

Any idea?????
sorry for my English

Koenkk, 2019-03-21:
Can you try with the `MAX_STABILITY` firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

joseska, 2019-03-22:
I´m trying..... Waiting for two or three days, that is the maximum days that the zigbee2mqtt work in my house.....

joseska, 2019-03-28:
sorry, but same again with MAX_STABILITY firmware (https://pastebin.com/BDYNSevt). 
Is posible that CC2531 is broken??? or 
USB (config???) disconect the usb stick????

Koenkk, 2019-03-28:
Can you disable the soft reset mechanism? (If you dont know how to do that please share your configuration.yaml)

joseska, 2019-03-28:
> 
> 
> Can you disable the soft reset mechanism? (If you dont know how to do that please share your configuration.yaml)

I was trying all the past weekend and monday **without** soft reset and not work for me. I enable only for trying diferent options....

thanks for your work.....

joseska, 2019-03-28:
a full log of my zigbee2mqtt:
https://pastebin.com/S0mhCs6F

Koenkk, 2019-03-29:
Can you post your `database.db` file?

joseska, 2019-03-29:
my databse.db:

`{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0001a9788d","nwkAddr":28051,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_motion","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"97XTAVJbrB3qTBWl"}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d00020258f4","nwkAddr":19522,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"F8OudajsZMDpjVfV"}
{"id":5,"type":"Router","ieeeAddr":"0x00158d0002b88bd5","nwkAddr":22993,"manufId":4447,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.light.aqcn02","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,1,3,4,5,6,8,10,13,19,258,768,1026,1027,1029,1030],"outClusterList":[1,6,8,10,13,19,25,258,768,1030],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":23,"stackVersion":2,"hwVersion":1,"manufacturerName":"LUMI","modelId":"lumi.light.aqcn02","dateCode":"11-22-2018","powerSource":4,"swBuildId":"1.23\u0000"}},"genPowerCfg":{"dir":{"value":3},"attrs":{"mainsVoltage":0}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":3},"attrs":{"onOff":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":3},"attrs":{"currentLevel":255,"remainingTime":0,"onOffTransitionTime":150}},"genTime":{"dir":{"value":3},"attrs":{}},"genAnalogOutput":{"dir":{"value":3},"attrs":{}},"genMultistateOutput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"closuresWindowCovering":{"dir":{"value":3},"attrs":{}},"lightingColorCtrl":{"dir":{"value":3},"attrs":{"remainingTime":0,"currentX":20650,"currentY":21396,"colorTemperature":153,"colorMode":1,"colorTempPhysicalMin":0,"colorTempPhysicalMax":65279}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msPressureMeasurement":{"dir":{"value":1},"attrs":{}},"msRelativeHumidity":{"dir":{"value":1},"attrs":{}},"msOccupancySensing":{"dir":{"value":3},"attrs":{}}}}},"_id":"Nzdz6prInfnlAdJC"}
{"id":10,"type":"EndDevice","ieeeAddr":"0x00158d0001aab2cd","nwkAddr":41053,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{}}},"_id":"OCBBfnd4kj3fHts6"}
{"id":7,"type":"EndDevice","ieeeAddr":"0x00158d0001abdbb4","nwkAddr":54340,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_switch","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,25,65535],"outClusterList":[0,3,4,5,6,8,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":1},"attrs":{}}}}},"_id":"OTeKODkrDRhEOBCq"}
{"id":11,"type":"EndDevice","ieeeAddr":"0x00158d0001f531bf","nwkAddr":52272,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,1026,1027,1029,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"msTemperatureMeasurement":{"dir":{"value":1},"attrs":{}},"msPressureMeasurement":{"dir":{"value":1},"attrs":{}},"msRelativeHumidity":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"YPh2xLykzMQquKP1"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0019365d70","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1553851678,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"out [... cut ...] led":"RW"},"attrs":{"zclVersion":1,"hwVersion":1,"manufacturerName":"sivann inc.","modelId":"hiver0001","dateCode":"20170407","powerSource":1,"locationDesc":"    ","physicalEnv":0,"deviceEnabled":1}},"ssIasAce":{"dir":{"value":1}},"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}},"ssIasWd":{"dir":{"value":2}},"genIdentify":{"dir":{"value":3},"acls":{"identifyTime":"RW"},"attrs":{"identifyTime":0}}}}},"_id":"eQ9ShvIdgGEdsTGm"}
`

Koenkk, 2019-03-29:
The failed to find device issue should be fixed now (see https://github.com/Koenkk/zigbee2mqtt/issues/1334#issuecomment-478100009).

Can you check again with the latest dev branch?

joseska, 2019-04-02:
Hi, I´m trying with new power plug.... It´s posible a power issue. Now we have two days without errors

closed 2019-06-08 by stale[bot]