Koenkk/zigbee2mqtt #1295: dependency issue with cc-znp

webster42, 2019-03-19:
@Koenkk 
While trying to build a nix package for zigbee2mqtt (using node2nix) I stumbled across an issue with cc-znp.
When last updating the serialport version used by cc-znp you forgot to update npm-shrinkwrap.json, see here: https://github.com/Koenkk/cc-znp/commit/7a57508a2a462808e34f2c9d6fc7b7a98040781d
It seems this leads to node2nix using the 2.0.5 version of the serial port bindings.
Would be nice if you could fix this.

Thanks

Koenkk, 2019-03-21:
thanks, done!

closed 2019-03-21 by Koenkk

webster42, 2019-03-21:
@Koenkk 
Sorry for bugging again, but I just noticed, that the npm-shrinkwrap of the zigbee2mqtt repo is affected, too.
It also references the old serial port bindings.

Koenkk, 2019-03-21:
I'm not sure how to fix that/what is wrong there, the npm-shrinkwrap.json was regenerated afterwards.

webster42, 2019-03-21:
I just realized it's a bit more complicated.
The problem is, the zigbee2mqtt package.json references a specific commit of zigbee-shepherd, which itself references a specific commit of cc-znp.
So the solution is to first update the commit hash of cc-znp in the zigbee-shepherd package.json and update the shrinkwrap file there.
The second step would be to update the zigbee-shepherd version in the zigbee2mqtt package.json with the new commit hash and then update the shrinkwrap file there.

webster42, 2019-03-21:
Just saw that you already updated the commit hash of cc-znp in the package.json of zigbee-shepherd.
So the only thing left to do is update the zigbee-shepherd version in the zigbee2mqtt repo.

Koenkk, 2019-03-21:
I already did that in: https://github.com/Koenkk/zigbee2mqtt/commit/24cf72f166f089eb8c8165e8d26e0f377e722cf8

webster42, 2019-03-21:
Thanks, I looked at the wrong branch. 👍

Koenkk, 2019-03-22:
good! Is everything ok now?

webster42, 2019-03-22:
Yes, no more segfaults.