Koenkk/zigbee2mqtt #1299: Hassio doesn’t connect to zigbee2mqtt

michaeldbr, 2019-03-21:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

michaeldbr, 2019-03-21:
Also there are no error logs in Hassio or the add-on.

hgelpke, 2019-03-24:
Do you have autodiscovery on in your mqtt integration?

Do you have homeassitant: true in your Z2M configuration.yaml?

olegbuk, 2019-03-26:
Having similar issues with my setup. Any help would be greatly appreciated.

hgelpke, 2019-03-26:
Can you be more specific on the problem?

I wen through a lot of potential issues so I think I can help but need more
details on what you've tried.

Did it work before?

Are you ale to pull any info from the logs of Zigbee2MQTT or MQTT itsself?

Does HA recognize other MQTT devices?

Podds84, 2019-05-18:
This worked for me:
mqtt:
  discovery: true
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'

Should be part of documentation... took me hours to find..

closed 2019-07-24 by stale[bot]