Koenkk/zigbee2mqtt #1300: Gledopto GL-C-009 is not supported

NinoContino, 2019-03-21:
Question:

Can the gledopto GL-C-009 be supported by zigbee2mqtt? I think it is the same as the GL-C 006 Gledopto Zigbee LED controller WW/CW Dimmer (on/off, brightness, color temperature).
The seller has send me the wrong type, so otherwise this one is useless.

Koenkk, 2019-03-21:
Please follow http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

yozik04, 2019-04-10:
I see it in devices.js

https://github.com/Koenkk/zigbee-shepherd-converters/blob/029f638e984a12202d3450e2a7566ba4c95f710f/devices.js#L2116

yozik04, 2019-04-10:
Just tested with mine.  Works perfectly.
I think this issue can be closed.

```
/zigbee/living_room_cabinet_light/set {"brightness":100}
/zigbee/living_room_cabinet_light {"brightness":100,"linkquality":0,"state":"ON"}
```

closed 2019-06-09 by Koenkk