Koenkk/zigbee2mqtt #1301: No converter available for 'LED1649C5'

santik, 2019-03-21:
I'm trying to connect Ikea bulb and getting this log
`  zigbee2mqtt:info 3/21/2019, 10:39:24 PM New device 'TRADFRI bulb E14 W op/ch 400lm' with address 0x000b57fffea1d704 connected!
  zigbee2mqtt:info 3/21/2019, 10:39:24 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000b57fffea1d704","meta":{"modelID":"TRADFRI bulb E14 W op/ch 400lm"}}'
  zigbee2mqtt:info 3/21/2019, 10:39:24 PM MQTT publish: topic 'homeassistant/light/0x000b57fffea1d704/light/config', payload '{"brightness":true,"schema":"json","command_topic":"zigbee2mqtt/0x000b57fffea1d704/set","state_topic":"zigbee2mqtt/0x000b57fffea1d704","name":"0x000b57fffea1d704_light","unique_id":"0x000b57fffea1d704_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffea1d704","name":"0x000b57fffea1d704","sw_version":"Zigbee2mqtt 1.2.1","model":"TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (LED1649C5)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/21/2019, 10:39:24 PM MQTT publish: topic 'homeassistant/sensor/0x000b57fffea1d704/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x000b57fffea1d704","json_attributes_topic":"zigbee2mqtt/0x000b57fffea1d704","name":"0x000b57fffea1d704_linkquality","unique_id":"0x000b57fffea1d704_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x000b57fffea1d704","name":"0x000b57fffea1d704","sw_version":"Zigbee2mqtt 1.2.1","model":"TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal (LED1649C5)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
  zigbee2mqtt:info 3/21/2019, 10:39:27 PM MQTT publish: topic 'zigbee2mqtt/0x000b57fffea1d704', payload '{"state":"ON","linkquality":141,"brightness":254}'
  zigbee2mqtt:info 3/21/2019, 10:39:27 PM MQTT publish: topic 'zigbee2mqtt/0x000b57fffea1d704', payload '{"state":"ON","linkquality":144,"brightness":254}'
  zigbee2mqtt:warn 3/21/2019, 10:39:28 PM No converter available for 'LED1649C5' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"lastMessageLqi":232,"lastMessageRssi":-42}}'
  zigbee2mqtt:warn 3/21/2019, 10:39:28 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:info 3/21/2019, 10:39:29 PM Device incoming...
  zigbee2mqtt:info 3/21/2019, 10:39:29 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'`
And I can not control the bulb. 
Not sure if it is because of this warning or something else.

Koenkk, 2019-03-22:
It's unlikely that this issue is connected to the warning.

Can you provide the log when controlling the bulb?

johtajajake, 2019-04-13:
I have something similar ongoing. I have somewhat paired the lamp - just a normal Ikea lamp - with zigbee2mqtt. Somewhat, as I did have a lot of trouble with the 'undefined' -error (https://github.com/Koenkk/zigbee2mqtt/issues/1378). I'm not exactly quite sure how I overcame with that. Now the lamp seems paired, but when I turn it on, this happens:

```
Apr 13 13:52:30 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:30 PM No converter available for 'LED1546G12' with cid 'genIdentify', type 'devChange' and data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}'
Apr 13 13:52:30 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:30 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:30 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:30 PM No converter available for 'LED1546G12' with cid 'genScenes', type 'devChange' and data '{"cid":"genScenes","data":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}}'
Apr 13 13:52:30 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:30 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:30 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:30 PM No converter available for 'LED1546G12' with cid 'genScenes', type 'devChange' and data '{"cid":"genScenes","data":{"65533":1}}'
Apr 13 13:52:30 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:30 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:31 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:31 PM MQTT publish: topic 'zigbee2mqtt/ikea_lamppu_olohuone', payload '{"state":"ON","linkquality":99,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
Apr 13 13:52:31 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:31 PM MQTT publish: topic 'zigbee2mqtt/ikea_lamppu_olohuone', payload '{"state":"ON","linkquality":99,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
Apr 13 13:52:31 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:31 PM MQTT publish: topic 'zigbee2mqtt/ikea_lamppu_olohuone', payload '{"state":"ON","linkquality":102,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
Apr 13 13:52:31 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:31 PM MQTT publish: topic 'zigbee2mqtt/ikea_lamppu_olohuone', payload '{"state":"ON","linkquality":102,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
Apr 13 13:52:32 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:32 PM MQTT publish: topic 'zigbee2mqtt/ikea_lamppu_olohuone', payload '{"state":"ON","linkquality":99,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
Apr 13 13:52:32 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:32 PM MQTT publish: topic 'zigbee2mqtt/ikea_lamppu_olohuone', payload '{"state":"ON","linkquality":99,"brightness":254,"color_temp":370,"color_mode":2,"color":{"x":0.46,"y":0.411}}'
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM No converter available for 'LED1546G12' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0}}'
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM No converter available for 'LED1546G12' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":176,"lastMessageRssi":-56}}'
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM No converter available for 'LED1546G12' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"65533":1}}'
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM No converter available for 'LED1546G12' with cid 'lightLink', type 'devChange' and data '{"cid":"lightLink","data":{"65533":1}}'
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:warn 4/13/2019, 1:52:33 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:33 PM Device incoming...
Apr 13 13:52:33 pie3 npm[28037]:   zigbee2mqtt:info 4/13/2019, 1:52:33 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
```

I guess the most interesting part is this `PM No converter available for 'LED1546G12' with cid 'haDiagnostic',`. Obviously this LED1546G12 does exist in the devices.js, however the manufacturer is "IKEA", while the broadcasted manufacturer (e.g. in data/database.db) is "IKEA from Sweden". Any guidance further? In case logs etc would help, please be understanding and define the log folder... I'm not exactly idiot with computers, but my linux/programming capabilities are limited :)

johtajajake, 2019-04-13:
...and then another strange one. I paired another Ikea lamp (identical to another lamp already paired, though not the one in above log). I get a:
`Device with modelID 'multiv4' is not supported.`

Koenkk, 2019-04-16:
@johtajajake these messages can be ignored and will be ignored in the next zigbee2mqtt release, thanks for reporting.

closed 2019-06-22 by stale[bot]