Koenkk/zigbee2mqtt #1302: Migrated to new HA install. HA won't pickup already paired devices

hgelpke, 2019-03-22:
I believe I've successfully migrated my install from Hass.io to a Dockerized install via Unraid. 

Everything seems to be communicating but Ha wont' pick up devices I've already paired. When I paired a new device it was recognized immediately though. 

I copied my previous database over and Zigbee2MQTT seems to recognize the devices though I don't know how to tell 100%. It looks like they are showing up in the logs though. 

In theory I could re-pair everything but there has to be something simple I'm missing.

hgelpke, 2019-03-22:
Here is my log:

The only two showing up in HA are:
0x00158d0002ebfc4f
0xd0cf5efffedceca9

3/21/2019, 11:26:41 PM - info: Logging to directory: '/app/data/log/2019-03-21.23-26-41'
3/21/2019, 11:26:41 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)
3/21/2019, 11:26:41 PM - info: Starting zigbee-shepherd
3/21/2019, 11:26:42 PM - info: zigbee-shepherd started
3/21/2019, 11:26:42 PM - info: Coordinator firmware version: '20190223'
3/21/2019, 11:26:42 PM - info: Currently 13 devices are joined:
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f000aa9a3fe): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f000aa9d1de): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f00057aa6e4): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f000aa82762): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f000aa85d11): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f00057d7034): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f000aa9d24f): 53170161 - Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x00178801045a1b8c): 9290011370 - Philips Hue white A60 bulb E27 (Router)
3/21/2019, 11:26:42 PM - info: unknown (0x000d6f000b3be4b3): unkown - undefined unknown (EndDevice)
3/21/2019, 11:26:42 PM - info: unknown (0x00158d0002b5803f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
3/21/2019, 11:26:42 PM - info: 0x00158d0002ebfc4f (0x00158d0002ebfc4f): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
3/21/2019, 11:26:42 PM - info: 0xd0cf5efffedceca9 (0xd0cf5efffedceca9): E1603 - IKEA TRADFRI control outlet (Router)
3/21/2019, 11:26:42 PM - info: 0x000d6ffffe8e95c2 (0x000d6ffffe8e95c2): unkown - undefined unknown (Router)
3/21/2019, 11:26:42 PM - warn: `permit_join` set to  `true` in configuration.yaml.
3/21/2019, 11:26:42 PM - warn: Allowing new devices to join.
3/21/2019, 11:26:42 PM - warn: Set `permit_join` to `false` once you joined all devices.
3/21/2019, 11:26:42 PM - info: Zigbee: allowing new devices to join.
3/21/2019, 11:26:42 PM - info: Connecting to MQTT server at mqtt://192.168.0.198:1883
3/21/2019, 11:26:42 PM - info: zigbee-shepherd ready
3/21/2019, 11:26:42 PM - info: Connected to MQTT server
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000aa9a3fe', payload '{"state":"ON","color_temp":198}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000aa9d1de', payload '{"state":"ON","color_temp":289}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057aa6e4', payload '{"state":"ON","color_temp":372,"brightness":233}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000aa82762', payload '{"state":"ON","color_temp":372,"brightness":121,"linkquality":13,"color_mode":2,"color":{"x":0.381,"y":0.377}}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000aa85d11', payload '{"state":"ON","color_temp":198}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f00057d7034', payload '{"state":"OFF"}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000aa9d24f', payload '{"state":"OFF"}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00178801045a1b8c', payload '{"state":"OFF"}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002b5803f', payload '{"temperature":19.94,"linkquality":99,"humidity":48.79,"pressure":982,"battery":97,"voltage":2995}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ebfc4f', payload '{"illuminance":31,"linkquality":55,"occupancy":false,"battery":100,"voltage":3045}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffedceca9', payload '{"state":"ON","linkquality":110}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
3/21/2019, 11:26:42 PM - info: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0002ebfc4f/occupancy/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.occupancy }}","device_class":"motion","state_topic":"zigbee2mqtt/0x00158d0002ebfc4f","json_attributes_topic":"zigbee2mqtt/0x00158d0002ebfc4f","name":"0x00158d0002ebfc4f_occupancy","unique_id":"0x00158d0002ebfc4f_occupancy_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002ebfc4f","name":"0x00158d0002ebfc4f","sw_version":"Zigbee2mqtt 1.2.1","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...] "sw_version":"Zigbee2mqtt 1.2.1","model":"TRADFRI control outlet (E1603)","manufacturer":"IKEA"},"availability_topic":"zigbee2mqtt/bridge/state"}'
3/21/2019, 11:31:50 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ebfc4f', payload '{"illuminance":32,"linkquality":57,"occupancy":true,"battery":100,"voltage":3045}'
3/21/2019, 11:31:50 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ebfc4f', payload '{"illuminance":32,"linkquality":60,"occupancy":true,"battery":100,"voltage":3045}'

hgelpke, 2019-03-22:
And here is the MQTT data:
1553237188: mosquitto version 1.4.10 (build date 2017-04-06 15:52:09+0000) starting

1553237188: Config loaded from /config/mosquitto.conf.
1553237188: Opening ipv4 listen socket on port 1883.
1553237188: Opening ipv6 listen socket on port 1883.
1553237188: Opening websockets listen socket on port 9001.
1553237189: New connection from 172.17.0.1 on port 1883.
1553237189: New client connected from 172.17.0.1 as mqttjs_6a5052eb (c1, k60).
1553237189: Sending CONNACK to mqttjs_6a5052eb (0, 0)
1553237189: Received SUBSCRIBE from mqttjs_6a5052eb
1553237189: zigbee2mqtt/bridge/device/+/get_group_membership (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/bridge/device/+/get_group_membership
1553237189: zigbee2mqtt/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/set
1553237189: zigbee2mqtt/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/get
1553237189: zigbee2mqtt/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/set
1553237189: zigbee2mqtt/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/get
1553237189: zigbee2mqtt/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/+/set (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/+/set
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/+/get (QoS 0)
1553237189: mqttjs_6a5052eb 0 zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/+/get
1553237189: zigbee2mqtt/+/+/+/+/+/+/+/+/+/+/+/+/+/+/+/set (QoS 0)
[... 39 lines cut ...]
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6f00057aa6e4', ... (48 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6f000aa82762', ... (110 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6f000aa85d11', ... (31 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6f00057d7034', ... (15 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6f000aa9d24f', ... (15 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x00178801045a1b8c', ... (15 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x00158d0002b5803f', ... (101 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0x00158d0002ebfc4f', ... (80 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r0, m0, 'zigbee2mqtt/0xd0cf5efffedceca9', ... (32 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r1, m0, 'zigbee2mqtt/bridge/config', ... (39 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r1, m0, 'homeassistant/sensor/0x00158d0002b5803f/temperature/config', ... (569 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r1, m0, 'homeassistant/sensor/0x00158d0002b5803f/humidity/config', ... (555 bytes))
[... cut ...]
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r1, m0, 'homeassistant/sensor/0x00158d0002ebfc4f/linkquality/config', ... (538 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r1, m0, 'homeassistant/switch/0xd0cf5efffedceca9/switch/config', ... (498 bytes))
1553237189: Received PUBLISH from mqttjs_6a5052eb (d0, q0, r1, m0, 'homeassistant/sensor/0xd0cf5efffedceca9/linkquality/config', ... (506 bytes))
1553237249: Received PINGREQ from mqttjs_6a5052eb
1553237249: Sending PINGRESP to mqttjs_6a5052eb
1553237309: Received PINGREQ from mqttjs_6a5052eb
1553237309: Sending PINGRESP to mqttjs_6a5052eb

Koenkk, 2019-03-22:
Did you also copy your zigbee2mqtt `configuration.yaml`? How does that file look now? My guess is that the already paired devices aren't present in there.

hgelpke, 2019-03-22:
I didn't copy the configuration.yaml from the old config but it didn't have all the devices. Are all the devices supposed to be added to it after they are paired?

Koenkk, 2019-03-22:
If you copied the database.db, they won't get added to `configuration.yaml`, you need to copy both `database.db` and `configuration.yaml` in order to migrate.

hgelpke, 2019-03-22:
Thank you

Should the Z2M configuration.yaml show all the devices?

Koenkk, 2019-03-22:
Yes (all should be in the `devices:` section).

hgelpke, 2019-03-23:
They weren't all there but I was able to re-create how they should be based on similar entities. Everything is back up and running without re-pairing. Thank you for the help.

Koenkk, 2019-03-23:
Great!

closed 2019-03-23 by Koenkk