Koenkk/zigbee2mqtt #1304: Hue dimmer and hue bulb binding

EmmanuelLM, 2019-03-22:
Hi, I have a hue dimmer switch (the rectangular 4 buttons one) and a hue bulb (warm ambiance).
Both bind successfully to zigbee2mqtt and I can see button presses and can control the bulb fine.

My issue is when I try to bind the remote to the bulb, the logs show a successful bind (i.e. no errors) but pressing the buttons thereafter do not control the bulb (I can see the presses in the log, and the bulb can still be controlled via zigbee2mqtt).

Any suggestions?

Koenkk, 2019-03-22:
To check what's going on, can you provide the zigbee sniffing log when pressing buttons on the remote? (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

EmmanuelLM, 2019-03-23:
Here you go:

No.     Time           Source                Destination           Protocol Length Info
      1 0.000000       0x6448                Broadcast             ZigBee   65     Link Status

Frame 1: 65 bytes on wire (520 bits), 63 bytes captured (504 bits) on interface 0
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x6448
ZigBee Network Layer Command, Dst: Broadcast, Src: 0x6448

No.     Time           Source                Destination           Protocol Length Info
      2 0.045383       0x9d52                0xc069                IEEE 802.15.4 12     Data Request

Frame 2: 12 bytes on wire (96 bits), 10 bytes captured (80 bits) on interface 0
IEEE 802.15.4 Command, Dst: 0xc069, Src: 0x9d52

No.     Time           Source                Destination           Protocol Length Info
      3 0.046151                                                   IEEE 802.15.4 5      Ack

Frame 3: 5 bytes on wire (40 bits), 3 bytes captured (24 bits) on interface 0
IEEE 802.15.4 Ack, Sequence Number: 62

No.     Time           Source                Destination           Protocol Length Info
      4 1.099316       0x9d52                0xc069                IEEE 802.15.4 12     Data Request

Frame 4: 12 bytes on wire (96 bits), 10 bytes captured (80 bits) on interface 0
IEEE 802.15.4 Command, Dst: 0xc069, Src: 0x9d52

No.     Time           Source                Destination           Protocol Length Info
      5 1.100086                                                   IEEE 802.15.4 5      Ack

Frame 5: 5 bytes on wire (40 bits), 3 bytes captured (24 bits) on interface 0
IEEE 802.15.4 Ack, Sequence Number: 63

No.     Time           Source                Destination           Protocol Length Info
      6 1.842825       0x0000                Broadcast             ZigBee   65     Link Status

Frame 6: 65 bytes on wire (520 bits), 63 bytes captured (504 bits) on interface 0
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x0000
ZigBee Network Layer Command, Dst: Broadcast, Src: 0x0000

No.     Time           Source                Destination           Protocol Length Info
      7 2.153512       0x9d52                0xc069                IEEE 802.15.4 12     Data Request

Frame 7: 12 bytes on wire (96 bits), 10 bytes captured (80 bits) on interface 0
IEEE 802.15.4 Command, Dst: 0xc069, Src: 0x9d52

No.     Time           Source                Destination           Protocol Length Info
      8 2.154256                                                   IEEE 802.15.4 5      Ack

Frame 8: 5 bytes on wire (40 bits), 3 bytes captured (24 bits) on interface 0
IEEE 802.15.4 Ack, Sequence Number: 64

No.     Time           Source                Destination           Protocol Length Info
      9 3.207761       0x9d52                0xc069                IEEE 802.15.4 12     Data Request

Frame 9: 12 bytes on wire (96 bits), 10 bytes captured (80 bits) on interface 0
IEEE 802.15.4 Command, Dst: 0xc069, Src: 0x9d52

No.     Time           Source                Destination           Protocol Length Info
     10 3.208550                                                   IEEE 802.15.4 5      Ack

Frame 10: 5 bytes on wire (40 bits), 3 bytes captured (24 bits) on interface 0
IEEE 802.15.4 Ack, Sequence Number: 65

No.     Time           Source                Destination           Protocol Length Info
     11 3.554322       0x202b                Broadcast             ZigBee   65     Link Status

Frame 11: 65 bytes on wire (520 bits), 63 bytes captured (504 bits) on interface 0
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0x202b
ZigBee Network Layer Command, Dst: Broadcast, Src: 0x202b
[... 2299 lines cut ...]
IEEE 802.15.4 Data, Dst: 0x202b, Src: 0x6448
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x347b
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 2
ZigBee Cluster Library Frame, Command: Configure Reporting Response, Seq: 6

No.     Time           Source                Destination           Protocol Length Info
    334 24.609344                                                  IEEE 802.15.4 5      Ack

Frame 334: 5 bytes on wire (40 bits), 3 bytes captured (24 bits) on interface 0
IEEE 802.15.4 Ack, Sequence Number: 59

No.     Time           Source                Destination           Protocol Length Info
    335 24.624211      0x347b                0x0000                ZigBee HA 49     ZCL: Configure Reporting Response, Seq: 6

Frame 335: 49 bytes on wire (392 bits), 47 bytes captured (376 bits) on interface 0
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x202b
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x347b
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 2
ZigBee Cluster Library Frame, Command: Configure Reporting Response, Seq: 6

No.     Time           Source                Destination           Protocol Length Info
    336 24.626163                                                  IEEE 802.15.4 5      Ack

Frame 336: 5 bytes on wire (40 bits), 3 bytes captured (24 bits) on interface 0
IEEE 802.15.4 Ack, Sequence Number: 190

Koenkk, 2019-03-23:
please provide the wireshark file and the ids of the button click messages. (This is not readable for me)

closed 2019-05-29 by stale[bot]

grostim, 2019-12-14:
> Setup is Hassio + Zigbee2mqtt 1.2.1 + latest version of coordinator firmware (master branch).
> 
> Hi, I have a hue dimmer switch (the rectangular 4 buttons one) and a hue bulb (warm ambiance).
> Both bind successfully to zigbee2mqtt and I can see button presses and can control the bulb fine.
> 
> My issue is when I try to bind the remote to the bulb, the logs show a successful bind (i.e. no errors) but pressing the buttons thereafter do not control the bulb (I can see the presses in the log, and the bulb can still be controlled via zigbee2mqtt).
> 
> Any suggestions?

I am facing the same issue.
The binding process went smoothly and i can see it in the logs :

> Message 11 reçu sur zigbee2mqtt/bridge/log à 14:46 :
> {
>     "type": "device_bind",
>     "message": {
>         "from": "Hue-Dimmer-switch",
>         "to": "Hue-GU10",
>         "cluster": "genLevelCtrl"
>     }
> }
> QoS: 0 - Retain: false
> Message 10 reçu sur zigbee2mqtt/bridge/log à 14:46 :
> {
>     "type": "device_bind",
>     "message": {
>         "from": "Hue-Dimmer-switch",
>         "to": "Hue-GU10",
>         "cluster": "genOnOff"
>     }
> }
> QoS: 0 - Retain: false
> Message 9 reçu sur zigbee2mqtt/bridge/log à 14:46 :
> {
>     "type": "device_bind",
>     "message": {
>         "from": "Hue-Dimmer-switch",
>         "to": "Hue-GU10",
>         "cluster": "genScenes"
>     }
> }
> QoS: 0 - Retain: false
> 
But nothing happen when i press a button.

I can't provide sniffing logs currently as i don't have an additional cc25xx usb stick.