Koenkk/zigbee2mqtt #1305: zigbee2mqtt should broadcast to friendlyname and ieeeAddr

jjrh, 2019-03-23:
`pub zigbee2mqtt/0xb0ce..../get -m '{"state": "" }'`
What's returns is not published to the topic `zigbee2mqtt/0xb0ce....` it's only published to the FriendlyName. 

One can however publish to `zigbee2mqtt/FriendlyName/get` and receive the output. 

The primary issue is that discovering devices using something like `zigbee2mqtt/bridge/networkmap -m 'raw'` only returns ieeeAddr and not the FriendlyName. 

I would propose:
- we publish to both the FriendlyName and ieeeAddr, 
- to the type that was requested (ie `FriendlyName/get `publishes to FriendlyName and `ieeeAddr/get` publishes to ieeeAddr )
- Add friendly name to the payload of 'raw' in networkMap.js or create another option/function if other things depend on that. 

Currently I am simply doing:
```
  if (settings.get().experimental.output === 'json') {
	  this.mqtt.publish(entity.ID, JSON.stringify(messagePayload), options);
          this.mqtt.publish(entity.friendlyName, JSON.stringify(messagePayload), options);

```

in the controller.js publishEntityState function. 

If there is a better way to achieve this please advise.

Koenkk, 2019-03-23:
To retrieve a list of all devices including the friendly name, send an empty payload to `zigbee2mqtt/bridge/config/devices`. Documetation: http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqtt/bridge/log

jjrh, 2019-03-23:
While that's good to know, the issue is I would receive everything sent to `bridge/log`. I can do `if payload.type="devices" ` but that really seems like the wrong use of a topic called `bridge/log`

I'm fairly new to mqtt in general but is there a reason why we don't publish the result to `zigbee2mqtt/bridge/config/devices/result` or send the empty payload to 
`zigbee2mqtt/bridge/config/devices/get` and publish to `zigbee2mqtt/bridge/config/devices` ? 

I'm happy to make a PR if there are no objections to changes like this.

Koenkk, 2019-03-25:
@jjrh I agree with you, as you mentioned, the correct behaviour would be to publish an empty payload to `zigbee2mqtt/bridge/config/devices/get`, resulting in zigbee2mqtt publishing to `zigbee2mqtt/bridge/config/devices`.

Could you make a PR? (please don't remove the old topic, but mark it as deprecated to avoid a breaking change).

Koenkk, 2019-03-26:
Merged, thanks!

closed 2019-03-26 by Koenkk