Koenkk/zigbee2mqtt #1306: Müllerlicht Tint remote control (MLI-404011): pairing works but then nothing

MarcSN311, 2019-03-23:
Hi,

I am trying to use the Tint remote control. Pairing was a pain. Most of the time nothing happend at all. Then the first time something happend: https://pastebin.com/3wd33K1U

After that I removed the entries from database.db and tried to pair again. Again nothig happend at all until I got this: https://pastebin.com/dCi9ctb9

For me this lookslike it paired successfully, but pressing buttons on the remote doesn't show in the output of zigbee2mqtt.

Any advice, how I can further debug this problem?

Thanks,
Marc

Koenkk, 2019-03-23:
Are you running the `20190223` firmware? (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/CC2531/bin)

MarcSN311, 2019-03-23:
yeah just figured this out as well. my firmware was so old, that it didn't even report it's version:
`zigbee2mqtt:info 3/23/2019, 2:11:23 PM Coordinator firmware version: 'undefined'`
Thank you

MarcSN311, 2019-03-23:
okay so after updating to 20190223 it worked. After restarting zigbee2mqtt it stopped working again.

closed 2019-03-23 by MarcSN311