Koenkk/zigbee2mqtt #1308: no log entry or state update if CC2531 becomes unavailable

thewilli, 2019-03-23:
I'm using the latest version (i.e. dev / e537784) of zigbee2mqtt on macOS 10.14.3 with node 10.15.0. 

When I pull the CC2531 stick while zigbee2mqtt is running, there is no error message in the log (log level: debug), nor is the `zigbee2mqtt/bridge/state` topic updated.

IMHO there should be some (preferrably mqtt-based) notification. This might happen if e.g. one was using a USB extension with a loose connection.

peterforeman, 2019-03-23:
The normal response would be to exit the application, sending the LWT to the mqtt broker.

thewilli, 2019-03-23:
> The normal response would be to exit the application, sending the LWT to the mqtt broker.

would be a good starting point, right now just *nothing* happens. Nevertheless, imho whether the application should actually exist or just set the LWT should depend on whether this would be treated as a *non-recoverable* or *recoverable* (waiting for the CC2531 to re-appear) error. I'm not deep enough into the code to say if this is possible in the first place, though.

tb-killa, 2019-03-24:
We use ``cc-znp`` who use ``SerialPort`` to communicate from zigbee shepherd with the coordinator stick.

Normally there should be a event ``close``on SerialPort to get the disconnect event triggered.
Maybe we could modify this for our need??

closed 2019-07-20 by stale[bot]