Koenkk/zigbee2mqtt #1309: proposal: add option to prevent duplicate messages

thewilli, 2019-03-23:
I'd propose to add a per-device configuration option to prevent duplicate messages from being sent: If the previous message would be equal to the current one (maybe excluding the signal strength), it would be silently discarded. This would allow to reduce the load on the mqtt broker and attached home automation system, when no actual update is required. Could even be combined with the `retain` option.

Nevertheless, it seems to be really an opt-in solution, to prevent action messages (e.g. contact closed, switch pressed), from being omitted.

antst, 2019-04-29:
I would also limit this by time.
As most of duplicated messages comes within second, then I’d say that duplicated messages within some time-frame(like 1-2 seconds) are dropped.
This will automatically work for sensors also.
And will not exclude long time-period  updates of states of the devices, even if state is the same.

MarsWarrior, 2019-04-29:
A time window would be very nice.
Some of my sensors give 1 update per hour. These updates should not be filtered.

antst, 2019-05-31:
And one more thing, unlike with debounce, we should not wait a sec, but publish and if next  message within time-frame is going to be the same (aside of link quality), we skip it.

closed 2019-08-07 by stale[bot]

dvbit, 2020-03-26:
seeing this behaviour regularly
Xiami button 
Xaiomi human body sensor

latest releaseo of zigbee2mqtt running on raspberry pi and ti 1532p

MadSnuif, 2020-11-27:
I see this also regularly. I would love the double sent items removed! It only happens for my devices not bind as a child dirrectly to the coordinator.

SkySlab, 2022-01-22:
Ditto, routers are duplicating messages