Koenkk/zigbee2mqtt #1310: proposal: add Docker healthcheck

thewilli, 2019-03-23:
The Docker container is great to run zigbee2mqtt as a "controlled, predefined environment". The only thing that I'm currently missing and would propose to add are [health checks](https://docs.docker.com/engine/reference/builder/#healthcheck).

If health checks were integreated, we could use monitoring and alerting systems to informat about the system being in an unusable state - or just restart the container.

Those *unhealthy* states might include:

* CC2531 no longer available (#1308)
* MQTT Server not reachable / MQTT authentication failed
* (maybe) certain devices marked as *important* are not available

peterforeman, 2019-03-23:
The first and second bullet should be fixed by exiting the application when these errors occur. If this happens while already running, the LWT will be sent (bridge offline). When restarting the application tries to reconnect all.

Your third bullet I'm not sure of the use and if it should be in zigbee2mqtt. Should it restart when a certain device is unavailable? Since the application is working fine, and is healthy.

thewilli, 2019-03-23:
> The first and second bullet should be fixed by exiting the application when these errors occur

@peterforeman restarting the application does not necessarily help when the MQTT Broker is down. And if it restarts too often in a short period of time, it might not come up again at all (e.g. when used as a Docker container). The healthcheck approach allows for a greater flexibility.

PedroLamas, 2019-08-09:
To be honest, what @thewilli says makes complete sense to me!

I use a mapped folder to have the latest `zigbee-shepherd-converters` and as some dependencies changed, the "node_modules" folder inside the docker container is not up-to-date.

Given that, the container launches fine and reports as started, yet looking at the logs I can see that the services didn't start and it just keeps restarting them.

A health check would have detected this issue and reported back to docker.

PedroLamas, 2019-08-09:
Thanks @Koenkk, given the opportunity I'll take a look at what we can do to have some basic health check here!

toxuin, 2020-05-29:
Kubernetes-based deployments would benefit from this as well.

I sometimes see odd error where there's an unhandled promise warning - and app did not crash, but does not send Zigbee/MQTT anymore.

Having a healthcheck would allow to eliminate whole class of problems like this.

Could this please be revisited? Thank you!

pedrolamas, 2020-11-01:
@Koenkk hope you don't mind, but I'm revisiting this topic!

Looking at the code, I was thinking we could just have a simple nodejs script that would use the websocket endpoint to perform a `health_check` ("zigbee2mqtt/bridge/request/health_check") and report the status back to docker if that was ok.

Koenkk, 2020-11-02:
@pedrolamas the websocket is only running when the frontend is enabled (and potentially requires authentication if `password` is set).

pedrolamas, 2020-11-02:
Any alternative to it? Ideally, I would like to avoid having to use MQTT as that means that the communication is going through another party outside of z2m control!

Koenkk, 2020-11-02:
@pedrolamas I currently see no other option, maybe health-check via frontend is the best one, but it can only be enabled when the user has enabled the frontend. Most easy thing would be a wget on the port where the frontend is running on (which is configurable and needs to be retrieved from `configuration.yaml`.

lubbifo, 2021-01-15:
Looking at issues like #5015 #4541 I would appreciate to have a Healthcheck implemented on the core functions of the gateway, not the UI.

pedrolamas, 2021-01-15:
@lubbifo the #5015 error will cause the container to stop, so you can check for that.

In case you have it set to "restart: unless-stopped", then you can add an inline health-check and make it go ok when the container has been running for something like 1 minute.

lubbifo, 2021-01-15:
@pedrolamas sorry cant confirm that, just had such an issue (to the extend that I could pinpoint it) and the container kept running. It did however count up the numbers the issue has been reported, see (rejection id: x) on the right end:

```
(node:17) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 9)
(node:17) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 10)
(node:17) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
```
Maybe this is not 100% what has been raised in #5015 or #4398, but it should emphasis the point, that there are situations, where the function may stop but the container does not. This has been observed using Zigbee2MQTT version 1.16.2 (commit #04c15f7). 

I now tried 1.17.0 to resolve my primary issue.

lesleypersyn, 2021-02-08:
@lubbifo, I can confirm this. Docker was running fine but zigbee2mqtt was not working anymore. I had following error (in loop)  last night (during cold night, no heating). A healthcheck (and action) would solve this.
version 1.17.1

> (node:235) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
>     at Timeout._onTimeout (/zigbee2mqtt-1.17.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
>     at listOnTimeout (internal/timers.js:554:17)
>     at processTimers (internal/timers.js:497:7)
> (node:235) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 249)

pedrolamas, 2021-02-08:
In my mind, if we could output some sort of status from zigbee2mqtt to a file, we could get Docker to check that file contents every x minutes and use that as a health check.

I've tried to find a place on the code to put a status output but was a bit overwhelmed with it, so no such luck yet...

lubbifo, 2021-02-08:
What about "/metrics", then it would be Prometheus conform?

pedrolamas, 2021-02-08:
For health check the regular address would be "/health", but I think you are assuming that there is an http endpoint running, though that is dependent on the frontend being enabled (if you disable the frontend, there is no http endpoint running).

As we can't assume the existence of such endpoint, that's why I suggested just output the state to a file and then getting the Docker health check to look into that file content

gsdevme, 2021-02-08:
I would just like to chime in that I've had events where zb2mq is running but in a crashed state whereby the node process is printing errors to a screen but from the point of view of the deployed container is healthily.

Ideas sound a little like a `pid` file, which zb2mq could create on boot and on error remove? or even it could touch every X like a heartbeat. I guess similar here could be to issue a heartbeat to MQTT and if that failures or become stale signal the container to stop for those that want to opt into this behavior?

peterforeman, 2021-02-08:
Well, first of all the "UnhandledPromiseRejectionWarning" should not be thrown at all. This simply means there is not correct error handling in place. 

You should catch the rejection and respond to it with... whatever is a suitable response. For example reinit the adapter, or simply quit the process and let it restart entirely.

Adding --unhandled-rejections=strict to the nodejs execution arguments would be a quick and perhaps dirty option at this point too: when an error occurs which is not catched properly the process terminates, hence the container will restart.

oernii, 2022-10-31:
I'd also like to request a health check. For monitoring and for kubernetes. Currently I have to check for URL/assets/index.521d5d7a.js but this changes with every npm build or release.

cnaslain, 2023-02-11:
I would love having an healtheck solution. It's important that docker/kubernetes restarts the container if something goes wrong.
Does anyone found a workaround to an app hack?

matti125, 2023-02-12:
I would also like to see a health check. I run z2m with homeassistant. I have the zigbee stick running on a remote node, and I connect to it with `ser2net`. If the node hosting the stick is not online when z2m starts up, z2m gets into a state where it is just stuck, and will not die/restart. Once the remote node is up, it is enough restart z2m, and things run smoothly. The problem is that this "grey" failure does not seem to kill the container, so that it would start automatically.

This is the state where z2m ends up:
```
[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"tcp://<ip address redacted>:20108"}}'
Zigbee2MQTT:error 2023-02-12 11:38:41: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-02-12 11:38:41: Failed to start zigbee
Zigbee2MQTT:error 2023-02-12 11:38:41: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-02-12 11:38:41: Exiting...
Zigbee2MQTT:error 2023-02-12 11:38:41: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:198:24)
    at Socket.emit (node:events:525:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
```

pkissling, 2023-02-20:
i would as well like to see this feature

daniel-wassenberg, 2023-05-21:
I turned on the 'Log symlink current' option then added the following to my docker-compose file to parse the log file for any warning or error entries.
```
healthcheck:
  test: ['CMD-SHELL', '! grep -i -E "warn|error" /app/data/log/current/log.txt']
  interval: 5m
  timeout: 10s
  retries: 3
```

peterforeman, 2023-05-26:
> I turned on the 'Log symlink current' option then added the following to my docker-compose file to parse the log file for any warning or error entries.
> 
> ```
> healthcheck:
>   test: ['CMD-SHELL', '! grep -i -E "warn|error" /app/data/log/current/log.txt']
>   interval: 5m
>   timeout: 10s
>   retries: 3
> ```

This seems to be a good solution, but if you have one unresponsive sensor (e.g. empty battery) the healthcheck fails as well.

MYeager1967, 2023-06-29:
I usually use a curl command for my health checks, but it doesn't appear it will work with Zigbee2MQTT. Is curl not part of the container?

koseduhemak, 2023-07-19:
I am using wget to check if the web ui is accessible. Hypothesis: if the web ui responds with 200, the container should be fine.

```
healthcheck:
      test: "wget --no-verbose --tries=1 --header 'Host: zigbee2mqtt.koseduhemak.de' --spider http://localhost:8080 || exit 1"
      start_period: 20s
      interval: 30s
      retries: 5
      timeout: 5s
```

MYeager1967, 2023-07-22:
Not being well versed in linux, WGET never crossed my mind. Works perfectly, thank you!!!

ingoratsdorf, 2024-03-04:
Guys, again wget and curl depend on the fact that a http endpoint, ie UI is running. I'd assume most people will have a UI running but there may be cases where people do not.
Plus, as outlined earlier, even if the UI responds, that does not mean the actual container works as expected. You'd only know that a http endpoint is running - which may be enough for some people, not enough for others.
The healthcheck needs to come from inside z2m.

peterforeman, 2024-07-04:
> Hello @peterforeman how you config your logs to get a CURRENT folder? Here the log folders are based on date only : "2024-07-03.13-50-39"

I'm currently not using this healthcheck, it was mentioned earlier by @daniel-wassenberg .

pabloaugusto, 2024-07-04:
> > Hello @peterforeman how you config your logs to get a CURRENT folder? Here the log folders are based on date only : "2024-07-03.13-50-39"
> 
> I'm currently not using this healthcheck, it was mentioned earlier by @daniel-wassenberg .

The question was about the current/log.txt functionality.
thanks

jbyers, 2024-10-12:
Adding an alternative to https://github.com/Koenkk/zigbee2mqtt/issues/1310#issuecomment-720104087 (a web endpoint that exercises the native `health_check` topic) in hopes this is helpful to others.

It seems like using the `health_check` topic is correct, but without a dependency on the UI. Natively in MQTT space this can be done out of the zigbee2mqtt container as a short ash script:

https://gist.github.com/jbyers/5f0a279db8b0022251eb699ee3db5305

This isn't ideal as it requires another Alpine package in the container (at least for the original purpose, a Docker healthcheck) but it removes other dependencies. I don't love grepping for raw JSON. Improvements welcome in the Gist.