Koenkk/zigbee2mqtt #1311: LED1623G12 (IKEA): needs repair after powerloss

peterforeman, 2019-03-23:
Running 1.2.1 with CC2531 (20190223) I bought some LED1623G12 light bulbs. Pairing was a breeze: allow pairing, turn on close to the CC2531, paired, done. Then I moved them to their definitive locations. After powering them on again I could not control them.

I found out that these bulbs (all 6 I have) need to repair when you put power on them. This means that I cut off power, put zigbee2mqtt in join mode enabled, power on the bulbs and I can see them repair again in the logs. I can also control them and they're responsive. Then I turn off the join mode in zigbee2mqtt.

When cutting power again and putting power on them again: no way to control them. I can only fix  this by cutting power again, and enable join mode and power on the bulbs. Then they're controllable again.

So every time I have a powerloss or whatever I need to do this sequence to be able to control them again. Any ideas why/how to fix? Because I do not want to permanently enable join mode...

Koenkk, 2019-03-25:
I don't experience this with similar TRADFRI bulbs, do other LED1623G12 also experience this?

davos2, 2019-04-13:
I have the same issue with a single new LED1623G12 in a CC2530 setup, 6 Other end devices (xiaomi/ikea) devices are online but this router shows up as offline. (nb I'm kinda new to this so I may be learning), re-paring fixes temporarily.
{"version":"1.3.0","commit":"da94252","coordinator":20190223,"log_level":"debug","permit_join":true}

davos2, 2019-04-18:
My issue was resolved by moving to 1.3.1. Don't know about the OP

peterforeman, 2019-04-23:
Unfortunately it did not... I'll try to reset and re-pair them again when I have some time to do so.

closed 2019-06-29 by stale[bot]