Koenkk/zigbee2mqtt #1314: Adding Innr "FL 130 C" Color LED Strip; Joined/not joined & Node Descriptor fail

NoWhere2Go19, 2019-03-24:
Hi guys,

I'm trying to add the Innr Color Flex LED Strip but until now I have had no luck.

When trying to pair, I get the following messages:
`Device with modelID 'FL 130 C' is not supported.
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Cannot get the Node Descriptor of the Device: 0x00158d000342a7cf (Error: Timed out after 10000 ms)
`
To me this sounds, that the device is not paired correctly. However looking at the database.db, the device is joined. 
`{"id":4,"type":"Router","ieeeAddr":"0x00158d000342a7cf","nwkAddr":18892,"manufId":4454,"epList":[1,2],"status":"online","joinTime":1553417965,"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{}}},"_id":"C7iJg2rlE4SOmpEF"}
{"id":4,"type":"Router","ieeeAddr":"0x00158d000342a7cf","nwkAddr":18892,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"FL 130 C","epList":[1,2],"status":"online","joinTime":1553417965,"endpoints":{"1":{"profId":49246,"epId":1,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{}},"2":{"profId":49246,"epId":2,"devId":4096,"inClusterList":[4096],"outClusterList":[],"clusters":{}}},"_id":"C7iJg2rlE4SOmpEF"}
`

But after a restart of zigbee2mqtt, the device is not listed in database.db anymore.

Moreover to support the device, I simply added the following in devices.js:
`{
        zigbeeModel: ['FL 130 C'],
        model: 'FL 130 C',
        vendor: 'Innr',
        description: 'Color Flex LED Strip',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },
`
I figured, that Innr would probably use the same logic in the LED strip as they do in their bulbs. therefore I just copy & pasted from the 'RB 185 C' bulb. However I contacted Innr which Zigbee standard they use and they said Light Link, so I think it should work with zigbee2mqtt.
Anyone any ideas, how to solve this problem?

Regards
Chris

Moreover I simply added

NoWhere2Go19, 2019-03-24:
Update: the FL 130 C seems to be paired at startup of zigbee2mqtt. However, I can't control it
`Currently 3 devices are joined:`
`0x000d6ffffe20a713 (0x000d6ffffe20a713): E1524 - IKEA TRADFRI remote control (EndDevice)`
`0x000d6ffffe240108 (0x000d6ffffe240108): E1524 - IKEA TRADFRI remote control (EndDevice)`
`0x00158d000342a7cf (0x00158d000342a7cf): FL 130 C - Innr Color Flex LED Strip (Router)`

Koenkk, 2019-03-25:
Can you provide the log when trying to control it?

NoWhere2Go19, 2019-03-25:
Current Status:
Pairing worked after some time. Holding it Close to the cc2531, resetting the LED Strip and then slowly turning it on and off seemed to make it pair.
However I can Control it now via OH2 mqtt, though it's a bit laggy and slow. Don't know wether this is an OH Thing or by zigbee2mqtt.
If nothing changes, I will close this issue and issue an update to devices.js

closed 2019-05-31 by stale[bot]